pkuehnel/TeslaSolarCharger #41: URL to REST Endpoint of smart meter

gjuecat, 2022-03-14:
whats has to be the format and data of this REST endpoint? 

I'm using a KNX counter for energy from or to the grid, so the counter itself has no IP interface (but I'm using MQTT via Gira Homeserver to get the data to IP). 
Now I'm thinking about some other tools if they are able to provide the information as REST endpoint. I'm not so familar with REST endpoint - so I need some information how I have to provide the data.

A possibilty which is already realised would be a HTTP call which delivers such data: 
{"request": {"method": "get", "key": "CO@13_3_0"}, "code": 0, "type": "call", "data": {"value": 14.0}}

The value would contain information how much energy comes or is deliverd to grid.

thx

pkuehnel, 2022-03-14:
Currently only integer values are supported. I will add an option to parse json data.

pkuehnel, 2022-03-14:
@gjuecat What does the value 14.0 represent? 14 Watt curently going to the grid? How would the json look like, if 14 Watt are coming from the grid?

gjuecat, 2022-03-14:
{"value": -1196.0} means that my house takes 1196.0 watt **from** the grid
{"value": 889.0} means that my house delivers 889.0 watt **to** the grid

gjuecat, 2022-03-19:
@pkuehnel works perfect - thank you

pkuehnel, 2022-03-19:
@gjuecat You can now switch back to `:latest` tag.

gjuecat, 2023-02-06:
Hi,
I upgraded to the latest version but now I'm not able to get power to grid with method from this issue.
Do I have to use environment variable? I din't try because on base configuration there is the possibility to enter the JSON result type $.data.value like you mentioned in this issue before.
[image]

gjuecat, 2023-02-06:
one more question: Home Battery is set to NONE but on the Overview page it is always mentioned in red:
[image]
Is this OK or is there something wrong?

pkuehnel, 2023-02-06:
This is currently an issue... Please toggle through all options (rest, modbus,MQTT) and delete every URL and so on...

gjuecat, 2023-02-07:
I changed https to http and now except battery power everthings gets correct values. Any ideas why battery power is not working? Answer from http request is the same like others and JSON pattern is also defined as "$.data.value"

pkuehnel, 2023-02-07:
Do you have self-signed certificates? If you have officially signed certificates, https should also work.

Are you sure you have not set any field (like xml path, MQTT Topix or something like that)?

gjuecat, 2023-02-07:
for Home Battery only REST and JSON Result type. XML path is empty and no MQTT used

pkuehnel, 2023-02-07:
Could you please again send me your logs but please nor the Json formatted, just the plain logs like described [here](https://github.com/pkuehnel/TeslaSolarCharger#generate-logfiles) .

gjuecat, 2023-02-07:
[teslasolarcharger.log](https://github.com/pkuehnel/TeslaSolarCharger/files/10674855/teslasolarcharger.log)

pkuehnel, 2023-02-07:
Looks like something is wrong with your URL. Maybe invisible characters, returns, or spaces or something like that?

gjuecat, 2023-02-07:
I used copy paste for all inputs and only changend one number within the URL. The answer from all of them has the same format and if I use the battery power URL in browser there‘s the correct value returned

gjuecat, 2023-02-07:
HomeBatteryPowerInversion URL is set to 0, but also 1 doesn't work

pkuehnel, 2023-02-07:
Could you please send me your baseConfig.json from your docker volume? But attention, everything you configured (including database password is in there). You can upload it [here](https://nextcloud.pkuehnel.de/s/RWxe4MWC9LDRcL2), so I am the only one who can see the file.

gjuecat, 2023-02-07:
file is online

pkuehnel, 2023-02-07:
> HomeBatteryPowerInversion URL is set to 0, but also 1 doesn't work

Ah this is your problem. I thought this was a typo... HomeBatteryPowerInversionURL needs to be set to empty... There you can insert an URL if your home battery always sends positive values but with another URL you can see if the battery is charging or discharging. Just remove the 0 there.

gjuecat, 2023-02-07:
thanks, that was the solution

closed 2023-02-07 by gjuecat