pkuehnel/TeslaSolarCharger #42: not usable for RPi ?

merlinfive, 2022-03-14:
Creating volume "pi_teslaampsetter-configs" with default driver
Pulling smartteslaampsetter (pkuehnel/smartteslaampsetter:latest)...
latest: Pulling from pkuehnel/smartteslaampsetter
ERROR: no matching manifest for linux/arm/v7 in the manifest list entries

pkuehnel, 2022-03-14:
Looks like this is a general issue with .net core https://github.com/dotnet/dotnet-docker/issues/1537

pkuehnel, 2022-03-14:
I tried a bunch of things which i reverted again. Currently I have this docker file: https://github.com/pkuehnel/SmartTeslaAmpSetter/blob/fix/rpiSupport/Plugins.SmaEnergymeter/Dockerfile

Current state is, that I get the following error on my [build action](https://github.com/pkuehnel/SmartTeslaAmpSetter/blob/fix/rpiSupport/.github/workflows/alphaRelease.yml):
```
#38 [linux/arm/v7 build 4/7] RUN dotnet restore "Plugins.SmaEnergymeter/Plugins.SmaEnergymeter.csproj"
#38 0.140 A fatal error occurred, the folder [/usr/share/dotnet/host/fxr] does not contain any version-numbered child folders
#38 ERROR: process "/dev/.buildkit_qemu_emulator /bin/sh -c dotnet restore \"Plugins.SmaEnergymeter/Plugins.SmaEnergymeter.csproj\"" did not complete successfully: exit code: 131
```

I already tried using the `6.0-bullseye-slim-amd64` (newest commit) but now I get the following error:
```
> [linux/arm64 build 4/7] RUN dotnet restore "Plugins.SmaEnergymeter/Plugins.SmaEnergymeter.csproj":
#38 0.085 .buildkit_qemu_emulator: /bin/sh: Invalid ELF image for this architecture
------
./Plugins.SmaEnergymeter/Dockerfile:11
```

@Mik4sa Could you give me a clue how I could resove this problem?

Mik4sa, 2022-03-15:
@pkuehnel Ok, got it! You need to setup qemu before you can work with buildx for .NET like this.
I read about it [here](https://github.com/lgisler/blue-whale/actions/runs/967219758/workflow) and [here](https://github.com/docker/setup-qemu-action).

Modify your workflow and add the set up qemu step. So that it looks like this:
```
name: alphaRelease

on:
  push:
    branches: [ develop ]
  pull_request:
    branches: [ develop ]

jobs:
  SmartTeslaAmpSetter:
    name: Building SmartTeslaAmpSetter image
    runs-on: ubuntu-latest

    steps:
    - name: Checkout
      uses: actions/checkout@v2

    - name: Set up QEMU
      uses: docker/setup-qemu-action@v1

    - name: Buildx
      id: buildx
      uses: docker/setup-buildx-action@v1

    - name: DockerHub login
      uses: docker/login-action@v1
      with:
        username: ${{ secrets.DOCKERHUB_USERNAME }}
        password: ${{ secrets.DOCKERHUB_PASS}}

    - name: Build and push SmartTeslaAmpSetter
      uses: docker/build-push-action@v2
      with:
        file: ./SmartTeslaAmpSetter/Server/Dockerfile
        platforms: linux/amd64,linux/arm64,linux/arm/v7
        push: true
        tags: pkuehnel/smartteslaampsetter:alpha

  SmaEnergymeterplugin:
    name: Building SMAPlugin Image
    runs-on: ubuntu-latest

    steps:
    - name: Checkout
      uses: actions/checkout@v2

    - name: Set up QEMU
      uses: docker/setup-qemu-action@v1

    - name: Buildx
      id: buildx
      uses: docker/setup-buildx-action@v1

    - name: DockerHub login
      uses: docker/login-action@v1
      with:
        username: ${{ secrets.DOCKERHUB_USERNAME }}
        password: ${{ secrets.DOCKERHUB_PASS}}

    - name: Build and push SmaPlugin
      uses: docker/build-push-action@v2
      with:
        file: ./Plugins.SmaEnergymeter/Dockerfile
        platforms: linux/amd64,linux/arm64,linux/arm/v7
        push: true
        tags: pkuehnel/smartteslaampsettersmaplugin:alpha
```

Remember too add this for each job.

pkuehnel, 2022-03-15:
@Mik4sa Thank you very much, is working fine.

@merlinfive Could you please give me short feedback, if it is working?

merlinfive, 2022-03-15:
Looks much better 😊 , will do more configuration later

 

***@***.***:~ $  docker-compose up -d

Pulling smaplugin (pkuehnel/smartteslaampsettersmaplugin:latest)...

latest: Pulling from pkuehnel/smartteslaampsettersmaplugin

59dbb2ac3829: Already exists

014a9d7851a9: Pull complete

0b0ea80e1d6f: Pull complete

b4b96de76f08: Pull complete

7d946209c911: Pull complete

2132c6eabfc0: Pull complete

4f4fb700ef54: Pull complete

cc2f1293f23a: Pull complete

Digest: sha256:53d5597cd4853b6f92778db327fb0c452bd99e211a4f124753fbfd6d67ad1686

Status: Downloaded newer image for pkuehnel/smartteslaampsettersmaplugin:latest

Pulling smartteslaampsetter (pkuehnel/smartteslaampsetter:latest)...

latest: Pulling from pkuehnel/smartteslaampsetter

59dbb2ac3829: Already exists

014a9d7851a9: Already exists

0b0ea80e1d6f: Already exists

b4b96de76f08: Already exists

7d946209c911: Already exists

179a8cf2a549: Pull complete

4f4fb700ef54: Pull complete

213616a8d146: Pull complete

Digest: sha256:6ee8e1a5d60974f94300bafbb659a45776ba0cd9f6c43a3e5951728074f18d66

Status: Downloaded newer image for pkuehnel/smartteslaampsetter:latest

pi_grafana_1 is up-to-date

pi_database_1 is up-to-date

pi_teslamate_1 is up-to-date

pi_mosquitto_1 is up-to-date

pi_teslamateapi_1 is up-to-date

Creating pi_smaplugin_1           ... done

Creating pi_smartteslaampsetter_1 ... done

 

Kind regards

closed 2022-03-15 by pkuehnel