pkuehnel/TeslaSolarCharger #77: Car not charging to minimum SoC

gjuecat, 2022-04-17:
The „min Soc + PV“ mode is active, car is charged to 10% and minimum SoC is set to 20% but the isn‘t charged.

pkuehnel, 2022-04-17:
ChangeChargeMode to PV only set yesterday as date, click save and change back to Min SOC + PV. If that does not help please get the Logfiles of the container.

pkuehnel, 2022-04-17:
Oh and I see, the second car has no values. Do you only have one car? If so change the Car Priorities Environment Variable to 1

gjuecat, 2022-04-18:
Seems to works now, thanks for the hint.

Concerning the car(s) - yes I have only one car. In the YAML File I use "- CarPriorities=1" but the second car is always displayed on the webif. Do I also have to modify the carConfig.json file if only using 1 car?

pkuehnel, 2022-04-18:
Should be fixed in yesterday's release. Just update. But caution when updating: If mosquito is also updated you need to restart all containers. So simply update only SmartTeslaAmpSetter with `docker-compose pull smartteslaampsetter` and do not update all containers. Or update all containers but then use `docker-compose down` before `docker-compose up -d`

Please let me know if the second car is gone after the update.

gjuecat, 2022-04-18:
Second car has gone after updating. Everything fine now.
Thanks

closed 2022-04-18 by gjuecat