pkuehnel/TeslaSolarCharger #91: Could not parse result 8387.36 from uri

cece46, 2022-04-23:
I have error to get my mower from EMONCMS : 
Why ?

version 1.5.0

```
teslamate-smartteslaampsetter-1  | [22:27:23.928 ERR Quartz.Core.JobRunShell] Job DEFAULT.a0173bef-ac7c-499a-9aef-647bf4d341e9 threw an unhandled Exception:
teslamate-smartteslaampsetter-1  | System.InvalidCastException: Could not parse result 8387.36 from uri http://12.18.11.3:8080/emoncms/feed/value.json?id=47&apikey=9f5cdd46xxxxxxxxxxxxxxxf8ba39 to integer
teslamate-smartteslaampsetter-1  |    at SmartTeslaAmpSetter.Server.Services.GridService.GetCurrentOverage() in /src/SmartTeslaAmpSetter/Server/Services/GridService.cs:line 46
teslamate-smartteslaampsetter-1  |    at SmartTeslaAmpSetter.Server.Services.ChargingService.SetNewChargingValues(Boolean onlyUpdateValues) in /src/SmartTeslaAmpSetter/Server/Services/ChargingService.cs:line 38
teslamate-smartteslaampsetter-1  |    at SmartTeslaAmpSetter.Server.Scheduling.Job.Execute(IJobExecutionContext context) in /src/SmartTeslaAmpSetter/Server/Scheduling/Job.cs:line 21
teslamate-smartteslaampsetter-1  |    at Quartz.Core.JobRunShell.Run(CancellationToken cancellationToken)
teslamate-smartteslaampsetter-1  | [22:27:23.929 ERR Quartz.Core.ErrorLogger] Job DEFAULT.a0173bef-ac7c-499a-9aef-647bf4d341e9 threw an exception.
teslamate-smartteslaampsetter-1  | Quartz.SchedulerException: Job threw an unhandled exception.
teslamate-smartteslaampsetter-1  |  ---> System.InvalidCastException: Could not parse result 8387.36 from uri http://12.18.11.3:8080/emoncms/feed/value.json?id=47&apikey=9f5cdd46xxxxxxxxxxxxxxxf8ba39 to integer
teslamate-smartteslaampsetter-1  |    at SmartTeslaAmpSetter.Server.Services.GridService.GetCurrentOverage() in /src/SmartTeslaAmpSetter/Server/Services/GridService.cs:line 46
teslamate-smartteslaampsetter-1  |    at SmartTeslaAmpSetter.Server.Services.ChargingService.SetNewChargingValues(Boolean onlyUpdateValues) in /src/SmartTeslaAmpSetter/Server/Services/ChargingService.cs:line 38
teslamate-smartteslaampsetter-1  |    at SmartTeslaAmpSetter.Server.Scheduling.Job.Execute(IJobExecutionContext context) in /src/SmartTeslaAmpSetter/Server/Scheduling/Job.cs:line 21
teslamate-smartteslaampsetter-1  |    at Quartz.Core.JobRunShell.Run(CancellationToken cancellationToken)
teslamate-smartteslaampsetter-1  |    --- End of inner exception stack trace --- [See nested exception: System.InvalidCastException: Could not parse result 8387.36 from uri http://12.18.11.3:8080/emoncms/feed/value.json?id=47&apikey=9f5cdd46xxxxxxxxxxxxxxxf8ba39 to integer
```

cece46, 2022-04-23:
I find the issue : the value is fload, not integer

pkuehnel, 2022-04-23:
I will add the option to parse float

cece46, 2022-04-23:
ok, cool

pkuehnel, 2022-04-24:
@cece46 Please try with tag `:gridFloat` instead of `:latest`. Let me know if it is working, if yes, I will merge the fix.

pkuehnel, 2022-04-26:
As I did not get an answer and is still working with integer values i merged it, so is available now in `:latest`

closed 2022-04-26 by pkuehnel