pkuehnel/TeslaSolarCharger #100: API_TOKEN

gjuecat, 2022-04-25:
I was using teslamateapi with EV key API_TOKEN_DISABLE=false and everything went fine until ~2 weeks and updating my docker compose container.
Was there somewhere a change? Why is it now not possible to use commands like wake up car? I get an error ins logs "failed validation".

pkuehnel, 2022-04-25:
Probably an issue with TESLAMATEAPI. TeslaMate plans to encrypt token in database. Maybe you have installed incompatible versions? I am using both with `latest` tag.

pkuehnel, 2022-04-25:
I know about an issue when mosquito is updated and restarted without restarting the other containers, so after an update I a way recreate all containers.

gjuecat, 2022-04-26:
now I started from the beginning (deleted all images and pulled them again)

In Postman I can wake up my car without authentication token.
SmartTeslaAmpSetter Website works, but is very slow. Setting of Amps doesn't work.
[image]

latest log:
`[15:03:43.192 VRB SmartTeslaAmpSetter.Server.Services.TeslamateApiService] ResumeLogging(1)
[15:03:52.355 DBG SmartTeslaAmpSetter.Server.Services.ChargingService] Relevant car ids: []
[15:03:52.355 DBG SmartTeslaAmpSetter.Server.Services.ChargingService] Irrelevant car ids: [1]
[15:03:52.355 DBG SmartTeslaAmpSetter.Server.Services.ChargingService] Resetting ChargeStart and ChargeStop for car 1
[15:03:52.355 VRB SmartTeslaAmpSetter.Server.Services.ChargingService] UpdateEarliestTimesAfterSwitch(1)
[15:03:52.358 VRB SmartTeslaAmpSetter.Server.Services.ConfigService] GetSettings()
[15:03:52.358 VRB SmartTeslaAmpSetter.Server.Services.ChargingService] SetNewChargingValues(True)
[15:03:52.358 VRB SmartTeslaAmpSetter.Server.Services.GridService] GetCurrentOverage()
[15:03:52.551 DBG SmartTeslaAmpSetter.Server.Services.GridService] Extract overage value from {"request": {"method": "get", "key": "CO@100_1_59"}, "code": 0, "type": "call", "data": {"value": -2494.0}} with $.data.value
[15:03:52.551 DBG SmartTeslaAmpSetter.Server.Services.ChargingService] Current overage is -2494 Watt.
[15:03:52.551 VRB SmartTeslaAmpSetter.Server.Services.GridService] GetCurrentInverterPower()
[15:03:52.551 DBG SmartTeslaAmpSetter.Server.Services.ChargingService] Current overage is -2494 Watt.
[15:03:52.552 DBG SmartTeslaAmpSetter.Server.Services.ChargingService] Adding powerbuffer 150
[15:03:52.552 DBG SmartTeslaAmpSetter.Server.Services.ChargingService] Relevant Geofence: Home
[15:03:52.552 WRN SmartTeslaAmpSetter.Server.Services.ChargingService] Unknown charge limit of car 1. Waking up car.
[15:03:52.552 VRB SmartTeslaAmpSetter.Server.Services.TelegramService] SendMessage(Unknown charge limit of car 1. Waking up car.)
[15:03:52.552 INF SmartTeslaAmpSetter.Server.Services.TelegramService] Can not send Telegram Message because botkey is null.
[15:03:52.552 VRB SmartTeslaAmpSetter.Server.Services.TeslamateApiService] WakeUpCar(1)
[15:03:52.552 VRB SmartTeslaAmpSetter.Server.Services.TeslamateApiService] SendPostToTeslaMate(http://192.168.36.8:8080/api/v1/cars/1/wake_up, null)
[15:03:52.763 VRB SmartTeslaAmpSetter.Server.Services.TeslamateApiService] result: {"response":{"access_type":"OWNER","api_version":40,"backseat_token":null,"backseat_token_updated_at":null,"calendar_enabled":true,"color":null,"display_name":"054545","id":1493030088600755,"id_s":"1493030088600755","in_service":false,"option_codes":"AD15,MDL3,PBSB,RENA,BT37,ID3W,RF3G,S3PB,DRLH,DV2W,W39B,APF0,COUS,BC3B,CH07,PC30,FC3P,FG31,GLFR,HL31,HM31,IL31,LTPB,MR31,FM3B,RS3H,SA3P,STCP,SC04,SU3C,T3CA,TW00,TM00,UT3P,WR00,AU3P,APH3,AF00,ZCST,MI00,CDM0","state":"online","tokens":["1b81d29d9085918c","3415ee32eca1e5f8"],"user_id":230525047806,"vehicle_id":237506787250,"vin":"xxxxxxxx054545"}}
[15:03:52.763 VRB SmartTeslaAmpSetter.Server.Services.TeslamateApiService] ResumeLogging(1)
[15:03:52.850 VRB SmartTeslaAmpSetter.Server.Services.ConfigJsonService] UpdateConfigJson()
[15:04:01.217 VRB SmartTeslaAmpSetter.Server.Services.ConfigService] GetSettings()
[15:04:01.217 VRB SmartTeslaAmpSetter.Server.Services.ChargingService] SetNewChargingValues(True)
[15:04:01.217 VRB SmartTeslaAmpSetter.Server.Services.GridService] GetCurrentOverage()
[15:04:01.391 DBG SmartTeslaAmpSetter.Server.Services.GridService] Extract overage value from {"request": {"method": "get", "key": "CO@100_1_59"}, "code": 0, "type": "call", "data": {"value": -2494.0}} with $.data.value
[15:04:01.392 DBG SmartTeslaAmpSetter.Server.Services.ChargingService] Current overage is -2494 Watt.
[15:04:01.392 VRB SmartTeslaAmpSetter.Server.Services.GridService] GetCurrentInverterPower()
[15:04:01.392 DBG SmartTeslaAmpSetter.Server.Services.ChargingService] Current overage is -2494 Watt.
[15:04:01.392 DBG SmartTeslaAmpSetter.Server.Services.ChargingService] Adding powerbuffer 150
[15:04:01.392 DBG SmartTeslaAmpSetter.Server.Services.ChargingService] Relevant Geofence: Home
[15:04:01.393 WRN SmartTeslaAmpSetter.Server.Services.ChargingService] Unknown charge limit of car 1. Waking up car.
[15:04:01.393 VRB SmartTeslaAmpSetter.Server.Services.TelegramService] SendMessage(Unknown charge limit of car 1. Waking up car.)
[15:04:01.393 INF SmartTeslaAmpSetter.Server.Services.TelegramService] Can not send Telegram Message because botkey is null.
[15:04:01.393 VRB SmartTeslaAmpSetter.Server.Services.TeslamateApiService] WakeUpCar(1)
[15:04:01.393 VRB SmartTeslaAmpSetter.Server.Services.TeslamateApiService] SendPostToTeslaMate(http://192.168.36.8:8080/api/v1/cars/1/wake_up, null)
[15:04:01.664 VRB SmartTeslaAmpSetter.Server.Services.TeslamateApiService] result: {"response":{"access_type":"OWNER","api_version":40,"backseat_token":null,"backseat_token_updated_at":null,"calendar_enabled":true,"color":null,"display_name":"054545","id":1493030088600755,"id_s":"1493030088600755","in_service":false,"option_codes":"AD15,MDL3,PBSB,RENA,BT37,ID3W,RF3G,S3PB,DRLH,DV2W,W39B,APF0,COUS,BC3B,CH07,PC30,FC3P,FG31,GLFR,HL31,HM31,IL31,LTPB,MR31,FM3B,RS3H,SA3P,STCP,SC04,SU3C,T3CA,TW00,TM00,UT3P,WR00,AU3P,APH3,AF00,ZCST,MI00,CDM0","state":"online","tokens":["1b81d29d9085918c","3415ee32eca1e5f8"],"user_id":230525047806,"vehicle_id":237506787250,"vin":"xxxxxxx04545"}}
[... cut ...]
[15:04:02.850 VRB SmartTeslaAmpSetter.Server.Services.ConfigJsonService] UpdateConfigJson()
[15:04:03.196 DBG SmartTeslaAmpSetter.Server.Services.ChargingService] Relevant car ids: []
[15:04:03.196 DBG SmartTeslaAmpSetter.Server.Services.ChargingService] Irrelevant car ids: [1]
[15:04:03.196 DBG SmartTeslaAmpSetter.Server.Services.ChargingService] Resetting ChargeStart and ChargeStop for car 1
[15:04:03.196 VRB SmartTeslaAmpSetter.Server.Services.ChargingService] UpdateEarliestTimesAfterSwitch(1)
`

pkuehnel, 2022-04-26:
How is your mosquito instance called in docker-compose.yml? From what I am seeing: SmartTeslaAmpSetter does not get the state of car, so there are two possible problems:
1. Your `mosquitto` instance is not called `mosquitto` like [here](https://docs.teslamate.org/docs/installation/docker#instructions) (last service name). If so, you can add environment variable: `MosquitoServer` with the name of your instance (default is `- MosquitoServer=mosquitto`)
2. Your car Id is not 1.

If I am wring with both: Could you please update SmartTelsaAmpSetter to `:edge` tag. I added several logging statements since last update. If it still does not work send me the logs of the first minute after startup.

gjuecat, 2022-04-26:
you were right, there was an error with mosquitto.
Seems to work now but sometimes get an error on website.

Can find this ERR in logs:
`[19:15:21.921 ERR Quartz.Core.JobRunShell] Job DEFAULT.e75da102-bef3-4248-aae6-2b60d28b3caf threw an unhandled Exception:
System.Net.Http.HttpRequestException: Response status code does not indicate success: 403 (Forbidden).
   at System.Net.Http.HttpResponseMessage.EnsureSuccessStatusCode()
   at SmartTeslaAmpSetter.Server.Services.GridService.GetCurrentOverage() in /src/SmartTeslaAmpSetter/Server/Services/GridService.cs:line 47
   at SmartTeslaAmpSetter.Server.Services.ChargingService.SetNewChargingValues(Boolean onlyUpdateValues) in /src/SmartTeslaAmpSetter/Server/Services/ChargingService.cs:line 38
   at SmartTeslaAmpSetter.Server.Scheduling.Job.Execute(IJobExecutionContext context) in /src/SmartTeslaAmpSetter/Server/Scheduling/Job.cs:line 22
   at Quartz.Core.JobRunShell.Run(CancellationToken cancellationToken)
[19:15:21.922 ERR Quartz.Core.ErrorLogger] Job DEFAULT.e75da102-bef3-4248-aae6-2b60d28b3caf threw an exception.
Quartz.SchedulerException: Job threw an unhandled exception.
 ---> System.Net.Http.HttpRequestException: Response status code does not indicate success: 403 (Forbidden).
   at System.Net.Http.HttpResponseMessage.EnsureSuccessStatusCode()
   at SmartTeslaAmpSetter.Server.Services.GridService.GetCurrentOverage() in /src/SmartTeslaAmpSetter/Server/Services/GridService.cs:line 47
   at SmartTeslaAmpSetter.Server.Services.ChargingService.SetNewChargingValues(Boolean onlyUpdateValues) in /src/SmartTeslaAmpSetter/Server/Services/ChargingService.cs:line 38
   at SmartTeslaAmpSetter.Server.Scheduling.Job.Execute(IJobExecutionContext context) in /src/SmartTeslaAmpSetter/Server/Scheduling/Job.cs:line 22
   at Quartz.Core.JobRunShell.Run(CancellationToken cancellationToken)
   --- End of inner exception stack trace --- [See nested exception: System.Net.Http.HttpRequestException: Response status code does not indicate success: 403 (Forbidden).
   at System.Net.Http.HttpResponseMessage.EnsureSuccessStatusCode()
   at SmartTeslaAmpSetter.Server.Services.GridService.GetCurrentOverage() in /src/SmartTeslaAmpSetter/Server/Services/GridService.cs:line 47
   at SmartTeslaAmpSetter.Server.Services.ChargingService.SetNewChargingValues(Boolean onlyUpdateValues) in /src/SmartTeslaAmpSetter/Server/Services/ChargingService.cs:line 38
   at SmartTeslaAmpSetter.Server.Scheduling.Job.Execute(IJobExecutionContext context) in /src/SmartTeslaAmpSetter/Server/Scheduling/Job.cs:line 22
   at Quartz.Core.JobRunShell.Run(CancellationToken cancellationToken)]
[19:15:23.787 ERR Microsoft.AspNetCore.Diagnostics.ExceptionHandlerMiddleware] An unhandled exception has occurred while executing the request.
System.Net.Http.HttpRequestException: Response status code does not indicate success: 503 (Service Unavailable).
   at System.Net.Http.HttpResponseMessage.EnsureSuccessStatusCode()
   at SmartTeslaAmpSetter.Server.Services.GridService.GetCurrentOverage() in /src/SmartTeslaAmpSetter/Server/Services/GridService.cs:line 47
   at SmartTeslaAmpSetter.Server.Services.ChargingService.SetNewChargingValues(Boolean onlyUpdateValues) in /src/SmartTeslaAmpSetter/Server/Services/ChargingService.cs:line 38
   at SmartTeslaAmpSetter.Server.Services.ConfigService.GetSettings() in /src/SmartTeslaAmpSetter/Server/Services/ConfigService.cs:line 27
   at lambda_method38(Closure , Object )
   at Microsoft.AspNetCore.Mvc.Infrastructure.ActionMethodExecutor.AwaitableObjectResultExecutor.Execute(IActionResultTypeMapper mapper, ObjectMethodExecutor executor, Object controller, Object[] arguments)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker.<InvokeActionMethodAsync>g__Awaited|12_0(ControllerActionInvoker invoker, ValueTask`1 actionResultValueTask)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker.<InvokeNextActionFilterAsync>g__Awaited|10_0(ControllerActionInvoker invoker, Task lastTask, State next, Scope scope, Object state, Boolean isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker.Rethrow(ActionExecutedContextSealed context)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker.Next(State& next, Scope& scope, Object& state, Boolean& isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker.<InvokeInnerFilterAsync>g__Awaited|13_0(ControllerActionInvoker invoker, Task lastTask, State next, Scope scope, Object state, Boolean isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeNextResourceFilter>g__Awaited|25_0(ResourceInvoker invoker, Task lastTask, State next, Scope scope, Object state, Boolean isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.Rethrow(ResourceExecutedContextSealed context)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.Next(State& next, Scope& scope, Object& state, Boolean& isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeFilterPipelineAsync>g__Awaited|20_0(ResourceInvoker invoker, Task lastTask, State next, Scope scope, Object state, Boolean isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeAsync>g__Awaited|17_0(ResourceInvoker invoker, Task task, IDisposable scope)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeAsync>g__Awaited|17_0(ResourceInvoker invoker, Task task, IDisposable scope)
[... cut ...]
   at Microsoft.AspNetCore.Routing.EndpointMiddleware.<Invoke>g__AwaitRequestTask|6_0(Endpoint endpoint, Task requestTask, ILogger logger)
   at Swashbuckle.AspNetCore.SwaggerUI.SwaggerUIMiddleware.Invoke(HttpContext httpContext)
   at Swashbuckle.AspNetCore.Swagger.SwaggerMiddleware.Invoke(HttpContext httpContext, ISwaggerProvider swaggerProvider)
   at Microsoft.AspNetCore.Diagnostics.ExceptionHandlerMiddleware.<Invoke>g__Awaited|6_0(ExceptionHandlerMiddleware middleware, HttpContext context, Task task)
`

gjuecat, 2022-04-26:
do you need the log from :edge tag?

gjuecat, 2022-04-28:
basically it seems to work, but sometimes there's an error in betweet the logs
[log.txt](https://github.com/pkuehnel/SmartTeslaAmpSetter/files/8583779/log.txt)

pkuehnel, 2022-04-28:
I am glad it is working for you.
Every error is the same: A request timeout to set the amp of your car. Could be connection issues of your car to the internet (bad wifi signal, bad mobile connection) or problems at Tesla's API at this time. You could try to get a better internet connection with your car but as the issue only occured 4 times in 2 hours I would not change anything. As I see from the logs the timeout is about ten seconds, so probably if you change the Ampere in your Tesla app every 30 seconds for two hours you probably would have the same issues 4 times in 2 hours. Just for the record: How is your car connected to the internet? At home I have a Wifi Access Point which is connected via ethernet cable, in my garage about 2 meters away from the car, so I do not know if getting timeouts is a common issue if your wifi/mobile connection is not perfect. What I could do is, that the Telegram error messages are only sent if an error occured more than e.g. five times in ten minutes.

gjuecat, 2022-05-01:
My car is connected with WLAN, but the AP isn't located behind a door. Normaly WLAN isn't a problem in the parked car.
I'm not sure if limiting error messages in telegram is the best way for this. In any case it should be only a flag that everbody has the option to disable error message limiting in telegram.

pkuehnel, 2022-05-01:
Ok, as I can not do much than, I will close the issue for now. What you can do is trying without WiFi but I am pretty sure no WiFi is worse.

closed 2022-05-01 by pkuehnel