pkuehnel/TeslaSolarCharger #109: Addon for RCT solar components

Executer1982, 2022-05-01:
RCT power converters don't have a REST API either, but there are already projects to work with that vendor-specific protocol, e.g. https://github.com/svalouch/python-rctclient

Is it possible to integrate this as an addon similar to your SMA-Addon?

pkuehnel, 2022-05-01:
Possible - yes. But as I only have SMA I can not developer such extensions. I would recommend something like home asisstant, which should have a rest API (but I have no experience with it). And as I see in your link, the repository is used by a home assistant extension.

pkuehnel, 2022-07-22:
Does RCT power have a modbus interface?

Executer1982, 2022-07-22:
As far as I know, it communicates via Modbus with the current sensor.

the protocoll itself seems to be some ModbusRTU(?), but I'm not into that :/ (see https://www.photovoltaikforum.com/thread/145688-rct-power-wechselrichter-auslesen/)

pkuehnel, 2022-07-24:
ModbusRTU means not over a Ethernet connection, this is not possible. If you could get any documentation or something like that about ModbusTCP it would be possible as the modbusPlugin released with [v2.2.0](https://github.com/pkuehnel/TeslaSolarCharger/releases/tag/v2.2.0) is generic and can be used for all inverters.

soulfire89, 2023-04-23:
Hey, 
i've the following documentation for RCT Inverters. Got it from the RCT Iobroker Adapter developer (https://github.com/aruttkamp/ioBroker.rct/blob/main/README.md).
Nearly every value can be read and logged. Unfortunatly I have no idea how it works.

https://rctclient.readthedocs.io/en/latest/inverter_registry.html#

**most Important values for car charging:**
- Power Values
 p_ac_load_sum -> houshold energy
- p_ac_grid_sum_lp -> grid load (positive) -> grid feed (negative)
- p_acc_lp -> battery charge (negative)/discharge (positive)

- Battery Values:
- soc -> actual SOC
- soc_target -> SOC Target

But there is a problem while using Teslas UMC. It's only one phase charging (max 32A). RCT PowerSensor has only 3 phase detection and shows wrong values while charging with one phase. I think its because of missing detection at ground phase.
Charging power is around double as high as shown inside Tesla app (32A*230V=7kW app compared to 14kW at RCT Power app)
Tested it for exact 60min and it charges exactly 7kW at my house energymeter, same like in tesla app.

If you need someone for testing, feel free to contact.

I would love to test Tesla SolarCharger, because I dont plan any wallbox. Teslas UMC is 100% ok for me (-> charging 70% at work for the last 25k km)

pkuehnel, 2023-04-25:
Sorry I missed your comment.

Why do you need that RCT power meter. Just use your smart Home meter.

The values you need is:

- grid Power (+ and -)
- battery power (+ and -)
- battery soc

that's it.

Can you maybe just use Iobroker or Homeassistant as Middleware, when you don't know how to connect TSC to your smart meter / home battery?

soulfire89, 2023-04-25:
Hey, no problem...
I've no possibility to read my smart home meter and battery values come from RCT inverter software.
But I'll try to set up a script at iobroker which sends the needed values to an XML Path like described in your documentation. I'm totally new to this topic, but it should work somehow.

pkuehnel, 2023-04-25:
Are you German? If yes search for Homeassistant in this thread: https://tff-forum.de/t/teslasolarcharger-laden-nach-pv-ueberschuss-mit-beliebiger-wallbox/170369?u=mane123

I have no idea if anyone is using iobroker.

solaranzeige.de would also work with TeslaSolarCharger (but no idea of it works with RTC solar).

soulfire89, 2023-04-25:
yes, I'm german. 
I've found something to read iobroker values with http request via RESTful API (e.g. http://ipaddress:8087/getPlainValue/soc  -> 80.7)

i just have to find out how two send it as list like you described for http://192.168.xxx.xxx/measurements.xml

Or is there a solution to define a single http-address for every needed value (grid power, battery power, soc) by its own?

pkuehnel, 2023-04-25:
Yes you can define URLs for every value by its own. You just could also use JSON or XML but that is optional.

soulfire89, 2023-04-25:
docker compose content is then without any plugin, just the stock one or?
I'll try my best tomorrow. Thanks for your help until now.

pkuehnel, 2023-04-25:
yep, without any Plugin...

soulfire89, 2023-04-25:
couldn't wait until tomorrow :) 
everything worked fine so far. input values were ok, now waiting for the sun and the first test tomorrow!

thanks a lot for your help!

pkuehnel, 2023-05-01:
Is everything working? If yes, could you share a link to the add-on you are using? Thanks.

soulfire89, 2023-05-01:
Yes it's working great. My Setup is a Synology 920+ with docker package installed.
1. Running ioBroker inside docker 
2.  Adapter RCT to collect Inverter Data https://github.com/aruttkamp/ioBroker.rct/
3.  RESTful API adapter to send datapoints via REST as an output of iobroker https://github.com/ioBroker/ioBroker.simple-api/
4. Use these links in your TSC BaseConfiguration as inputs (e.g. http://**ip-adresse:port**/getPlainValue/rct.0.g_sync.p_ac_grid_sum_lp

peterge1998, 2024-04-11:
I would be very interested in an RCT addon for teslasolarcharger too. We got a rct solar installation with a [RCT Power Storage DC10](https://www.rct-power.com/de/power-storage-dc.html) last year, I added it to homeassistant via the [hacs integration](https://github.com/weltenwort/home-assistant-rct-power-integration), which did work fine.

Is there anything I can help you with for developing this addon @pkuehnel ?

pkuehnel, 2024-04-11:
@peterge1998 How do you get the values from RCT? For everything like Modbus, REST, MQTT, there shouldn't be any issues. The plugins are only needed for inverters, batteries,... with very special APIs. E.g. Solax, see method in line 54 https://github.com/pkuehnel/TeslaSolarCharger/blob/master/Plugins.Solax/Services/SolaxApiService.cs

peterge1998, 2024-04-11:
@soulfire89 
Thats acutually a good idea, getting them from hassio could work. I need to research which entities I need to use, but it should be doable since the integration provides quite a few battery & grid values.

@pkuehnel 
The integration uses the inverter's API:
```
peterge@thinkpadt14s:~$ curl --http0.9 http://10.0.1.54:8899/ --output -
PGO��I�z˰!����U��4b�Dy�����IHk�l/�1{xD�贖�O��C�B5*�!�����S�*��A�34S�j��E�:�7��bf!Pn:���
                                                                                       H�ZH��aHF^	����&�F<,I��UזI�
```

Thank you guys, I will look into it when I order the M3, but I guess might need a little more help with it then 😬

pkuehnel, 2024-04-11:
Looks like a normal REST API call, what is the result? And why are so many not displayable chars in your message?

If you have no different you can use my referral ;-)

peterge1998, 2024-04-11:
Idk, its binary. The same shows up when opening http://10.0.1.54:8899/ in browser:
[image]

But I am able to use [rctclient](https://github.com/svalouch/python-rctclient/):
```
(venv) root@sshgateway2:~# rctclient read-value --host 10.0.1.54 --name battery.soc
0.876132607460022
```

I want to order a M3 2022 RWD from tesla used cars (eAP 🤗), not brand new, so I cant use your referral...

peterge1998, 2024-04-11:
Idk, its binary. The same shows up when opening http://10.0.1.54:8899/ in browser:
[image]

But I am able to use [rctclient](https://github.com/svalouch/python-rctclient/):
```
(venv) root@sshgateway2:~# rctclient read-value --host 10.0.1.54 --name battery.soc
0.876132607460022
```

I want to order a M3 2022 RWD from tesla used cars (EAP 🤗), not brand new, so I cant use your referral...

pkuehnel, 2024-04-12:
Ah, ok, I see — looks like it's not that easy to decode. The easiest way for you is if you have any Converter that can export the data via MQTT or REST. But best would be if you can decode the values in C# (or tell me exactly how to decode them and provide me access to your inverter, so I can implement it).

peterge1998, 2024-04-12:
Another idea - yesterday when I was setting up rctclient, I saw that the tool comes with a [simulator](https://rctclient.readthedocs.io/en/latest/simulator.html). I did not try it yet, but this paragraph in the docs just sounds like it could help out with the access problem:
> To aid in developing tools or this module itself, a simulator offers an easy way to safely test code without interfacing with real hardware and potentially causing problems. The simulator can be run by using the subcommand simulator of the CLI.

peterge1998, 2024-08-30:
I am unable to set up the inverter power, because the S0 is missing in the home assistant integration.
How are u solving this @soulfire89 ? Or do you have no power input on S0 in you installation?

This is what I did set up:

| TSC variable | RCT Power App | Home Assistant entity |
| ------------------ | ---------------------- | -------------------------------- |
| [image] Plus InverterPower | [image] |sensor.rct_power_storage_generator_a_power |
| [image] Plus InverterPower | [image] | sensor.rct_power_storage_generator_b_power |
| [image] Plus InverterPower | [image] | missing in home assistant :(|

I was able to find HomeBatterySoc, HomeBatteryPower and GridPower. I am just unable to set up my InverterPower. Any ideas?

[image]

soulfire89, 2024-08-30:
Hey, 
I'm not using home assistant. Taking the RCT Inverter Values with iobroker and sending them via RestAPI to TSC.

peterge1998, 2024-08-30:
The value that I need would be this one:
[image]

I guess I need to figure out how I can get this into home assistant first...

soulfire89, 2024-08-30:
Just write a request to the home assistant adapter developer ;)