pkuehnel/TeslaSolarCharger #144: Car power always 0W

MrBetatester, 2022-07-15:
We have a Model 3 and Model Y both with a Wallbox Gen 3. The UI show the current grid power and some information about the cars but not the current charging value. It also not change the ampere. TeslaMate  TeslaMateApi show the needed values correctly.

[image]

Thanks
Ronald

pkuehnel, 2022-07-15:
Probably you did not set a Home Geofence? You have a Geofence as environment variable (default is 'Home') and you need to set this Geofence in TeslaMate UI with the name 'Home'.

MrBetatester, 2022-07-16:
It was my fault. I checked TeslaMateApi response: "geofence":"Home " there was a space at the end of the name :-). Now it shows the current power but it doesn't control the loading process. 
I increased the SoC limit from 80 to 85 und more than the half of the needed power will be used from grid. I use the default update interval of 5 minutes but it doesn't stop loading after 20 minutes using grid power.

pkuehnel, 2022-07-16:
Could you please share your `docker-compose.yml`? Please don't forget to remove your passwords before you share.

MrBetatester, 2022-07-16:
I use port 4000 for teslamate, 4001 for teslamateapi and 4002 for teslaampsetter

```
version: "3"

services:
  teslamate:
    image: teslamate/teslamate:latest
    restart: always
    environment:
      - ENCRYPTION_KEY=##########################################################
      - DATABASE_USER=teslamate
      - DATABASE_PASS=##############################
      - DATABASE_NAME=teslamate
      - DATABASE_HOST=database
      - MQTT_HOST=mosquitto
    ports:
      - 4000:4000
    volumes:
      - ./import:/opt/app/import
    cap_drop:
      - all

  database:
    image: postgres:14
    restart: always
    environment:
      - POSTGRES_USER=teslamate
      - POSTGRES_PASSWORD=##############################
      - POSTGRES_DB=teslamate
    volumes:
      - teslamate-db:/var/lib/postgresql/data

  grafana:
    image: teslamate/grafana:latest
    restart: always
    environment:
      - DATABASE_USER=teslamate
      - DATABASE_PASS=##############################
      - DATABASE_NAME=teslamate
      - DATABASE_HOST=database
    ports:
      - 3000:3000
    volumes:
      - teslamate-grafana-data:/var/lib/grafana

  mosquitto:
    image: eclipse-mosquitto:2
    restart: always
    command: mosquitto -c /mosquitto-no-auth.conf
    # ports:
    #   - 1883:1883
    volumes:
      - mosquitto-conf:/mosquitto/config
      - mosquitto-data:/mosquitto/data

  teslamateapi:
    image: tobiasehlert/teslamateapi:latest
    logging:
        driver: "json-file"
        options:
            max-file: "5"
            max-size: "10m"
    restart: always
    depends_on:
      - database
    environment:
      - DATABASE_USER=teslamate
      - DATABASE_PASS=##############################
      - DATABASE_NAME=teslamate
      - DATABASE_HOST=database
      - MQTT_HOST=mosquitto
      - TZ=Europe/Berlin
      - ENABLE_COMMANDS=true
      - COMMANDS_ALL=true
      - API_TOKEN_DISABLE=true
    ports:
      - 4001:8080

  smartteslaampsetter:
    image: pkuehnel/smartteslaampsetter:latest
    logging:
        driver: "json-file"
        options:
            max-file: "5"
            max-size: "10m"
    restart: always
    depends_on:
      - teslamateapi
    environment:
      - CurrentPowerToGridUrl=http://192.168*.*/pv_tesla/GridPower.php
      #- CurrentInverterPowerUrl=http://192.168*.*/pv_tesla/InverterPower.php
      - TeslaMateApiBaseUrl=http://192.168*.*:4001
      - UpdateIntervalSeconds=30
      - CarPriorities=1|2
      - GeoFence=Home
      - MinutesUntilSwitchOn=5
      - MinutesUntilSwitchOff=5
      - PowerBuffer=0
      - TZ=Europe/Berlin
    ports:
      - 4002:80
    volumes:
      - teslaampsetter-configs:/app/configs

volumes:
  teslamate-db:
  teslamate-grafana-data:
  mosquitto-conf:
  mosquitto-data:
  teslaampsetter-configs:

```

pkuehnel, 2022-07-16:
You need to add `- ENCRYPTION_KEY=MySuperSecretEncryptionKey` (same key as TeslaMate) to your `teslamateapi` instance. I forgot to update the readme after yesterday's TeslaMate update.

MrBetatester, 2022-07-16:
I added the key but it doesn't help. How can I try TeslaMateAPI commands?

http://192.168.*.*:4001/api/v1/cars/1/command list all needed commands but if I call 
http://192.168.*.*:4001/api/v1/cars/1/command/wake_up or http://192.168.*.*:4001/api/v1/cars/1/wake_up I get the result 

{"code":"PAGE_NOT_FOUND","message":"Page not found"}

also for the other commands

MrBetatester, 2022-07-16:
Ok I send a POST request via PHP TeslaMateAPI works correctly :-)

pkuehnel, 2022-07-16:
And Tesla Solar Charger does still not work? Could you please recreate the container and after about one minute export the logs? In car settings the Auto Set Amp option is active?

MrBetatester, 2022-07-18:
I try it. As you can see on the attached log my car will be listen as irrelevant?

Log:
[log.txt](https://github.com/pkuehnel/TeslaSolarCharger/files/9129915/log.txt)

Current loading doesn't stop after 20 min:
[image]

pkuehnel, 2022-07-19:
Sorry, I missed your message. As I can see from your logs, car 1 was listed as irrelevant because of Chargelimit and SoC are the same 80% (but as I see in your screenshot you have 85% as charge limit) and Car 2 was irrelevant because Geofence was null.

At this point I have to say, that if your car is not charging and soc less than 4% lower than charge limit the car is listed as irrelevant as Teslas do not start charging if difference between SoC and Charge Limit is to low.

MrBetatester, 2022-07-19:
But it has charged the 5% in 1:10h :-) The cars also load 2% if I only drive to get some food.

pkuehnel, 2022-07-19:
Which car are you talking about? 1 or 2? The logs differ from your Screenshot.

MrBetatester, 2022-07-19:
Car 1 (Car 2 is not at home :-) )

pkuehnel, 2022-07-19:
Ahh and I saw another thing... Pilot current and Requested Current is null while the car is charging. Is your TeslaMate instance up to date (v1.27).

MrBetatester, 2022-07-19:
Ok it is 1.26.1 I will update it tomorrow

MrBetatester, 2022-07-20:
I updated all containers:

```
docker-compose logs smartteslaampsetter | tail -n 1000
....
smartteslaampsetter_1  | [07:11:53.837 VRB TeslaSolarCharger.Server.Services.ChargingService] Irrelevant cars: [{"Id": 1, "CarConfiguration": {"UpdatedSincLastWrite": false, "ChargeMode": "PvAndMinSoc", "MinimumSoC": 30, "LatestTimeToReachSoC": "2022-01-01T00:00:00.0000000", "MaximumAmpere": 14, "MinimumAmpere": 1, "UsableEnergy": 75, "ShouldBeManaged": true, "$type": "CarConfiguration"}, "CarState": {"Name": "RAZORBACK", "ShouldStartChargingSince": "9999-12-31T23:59:59.9999999", "ShouldStopChargingSince": "9999-12-31T23:59:59.9999999", "SoC": 80, "SocLimit": 80, "Geofence": "Home", "TimeUntilFullCharge": "00:00:00", "ReachingMinSocAtFullSpeedCharge": "2022-07-20T03:18:28.6835402+02:00", "AutoFullSpeedCharge": false, "LastSetAmp": 0, "ChargerPhases": 2, "ActualPhases": 3, "ChargerVoltage": 2, "ChargerActualCurrent": 0, "ChargerPilotCurrent": null, "ChargerRequestedCurrent": null, "PluggedIn": true, "ClimateOn": false, "ChargingPowerAtHome": 0, "ChargingPower": 0, "StateString": "asleep", "State": "Asleep", "Healthy": true, "ReducedChargeSpeedWarning": false, "$type": "CarState"}, "$type": "Car"}, {"Id": 2, "CarConfiguration": {"UpdatedSincLastWrite": false, "ChargeMode": "PvAndMinSoc", "MinimumSoC": 30, "LatestTimeToReachSoC": "2022-01-01T00:00:00.0000000", "MaximumAmpere": 14, "MinimumAmpere": 1, "UsableEnergy": 75, "ShouldBeManaged": true, "$type": "CarConfiguration"}, "CarState": {"Name": "Wecke das Teslum in Dir", "ShouldStartChargingSince": "9999-12-31T23:59:59.9999999", "ShouldStopChargingSince": "9999-12-31T23:59:59.9999999", "SoC": 80, "SocLimit": 80, "Geofence": "Home", "TimeUntilFullCharge": "00:00:00", "ReachingMinSocAtFullSpeedCharge": "2022-07-20T03:18:28.6835436+02:00", "AutoFullSpeedCharge": false, "LastSetAmp": 0, "ChargerPhases": 2, "ActualPhases": 3, "ChargerVoltage": 2, "ChargerActualCurrent": 0, "ChargerPilotCurrent": null, "ChargerRequestedCurrent": null, "PluggedIn": true, "ClimateOn": false, "ChargingPowerAtHome": 0, "ChargingPower": 0, "StateString": "asleep", "State": "Asleep", "Healthy": true, "ReducedChargeSpeedWarning": false, "$type": "CarState"}, "$type": "Car"}]
....
```

where TeslaMateApi shows the following output for car 1:
```

{
   "data":{
      "car":{
         "car_id":1,
         "car_name":"RAZORBACK"
      },
      "status":{
         "display_name":"RAZORBACK",
         "state":"charging",
         "state_since":"2022-07-20T07:02:30+02:00",
         "odometer":3011.67,
         "car_status":{
            "healthy":true,
            "locked":true,
            "sentry_mode":false,
            "windows_open":false,
            "doors_open":false,
            "trunk_open":false,
            "frunk_open":false,
            "is_user_present":false
         },
         "car_details":{
            "model":"Y",
            "trim_badging":"74D"
         },
         "car_exterior":{
            "exterior_color":"PearlWhite",
            "spoiler_type":"None",
            "wheel_type":"Apollo19"
         },
         "car_geodata":{
            "geofence":"Home",
            "latitude":****,
            "longitude":******
         },
         "car_versions":{
            "version":"2022.20.6",
            "update_available":false,
            "update_version":""
         },
         "driving_details":{
            "shift_state":"",
            "power":-3,
            "speed":0,
            "heading":271,
            "elevation":0
         },
         "climate_details":{
            "is_climate_on":false,
            "inside_temp":27.1,
            "outside_temp":25,
            "is_preconditioning":false
         },
         "battery_details":{
            "est_battery_range":319.04,
            "rated_battery_range":417.72,
            "ideal_battery_range":417.72,
            "battery_level":80,
            "usable_battery_level":80
         },
         "charging_details":{
            "plugged_in":true,
            "charge_energy_added":0.94,
            "charge_limit_soc":85,
            "charge_port_door_open":true,
            "charger_actual_current":5,
            "charger_phases":2,
            "charger_power":4,
            "charger_voltage":238,
            "charge_current_request":5,
            "charge_current_request_max":16,
            "scheduled_charging_start_time":"0001-01-01T00:53:28+00:53",
            "time_to_full_charge":1.25
         },
         "tpms_details":{
            "tpms_pressure_fl":0,
            "tpms_pressure_fr":0,
            "tpms_pressure_rl":0,
            "tpms_pressure_rr":0
         }
      },
      "units":{
         "unit_of_length":"km",
         "unit_of_temperature":"C"
      }
   }
}

```

It seems the limit will not be updated.

MrBetatester, 2022-07-20:
I restarted the container now the Limits will be updated. I try to load both cars to exceed the PV power but it is it much sun at the moment :-).

So I try to load only one car. At the logs I can see:

SendPostToTeslaMate(http://192.168.*.*:4001/api/v1/cars/1/command/set_charging_amps, {"charging_amps": "10"})

But the amps will not changed. Some time before the log shows System.Threading.Tasks.TaskCanceledException.

[log_full.txt](https://github.com/pkuehnel/TeslaSolarCharger/files/9149483/log_full.txt)

MrBetatester, 2022-07-20:
My short php test script works. I see the changes instantly at the Tesla App.

```
<?php

function commandTesla($cmd, $fields = null) {
    $url = "http://192.168.*.*:4001/api/v1/cars/1/command/$cmd";
    
    $curl = curl_init($url);
    curl_setopt($curl, CURLOPT_HEADER, false);
    curl_setopt($curl, CURLOPT_RETURNTRANSFER, true);
    curl_setopt($curl, CURLOPT_POST, true);
    curl_setopt($curl, CURLOPT_HTTPHEADER, array('Content-Type:application/json'));
    
    if ($fields != null) {
        $payload = json_encode($fields);
        curl_setopt($curl, CURLOPT_POSTFIELDS, $payload);
    }

    $json_response = curl_exec($curl);

    $status = curl_getinfo($curl, CURLINFO_HTTP_CODE);

    curl_close($curl);

    $response = json_decode($json_response, true);
    return $response;
}

//var_dump(commandTesla("wake_up"));
//var_dump(commandTesla("set_charge_limit", array("percent" => 75)));
var_dump(commandTesla("set_charging_amps", array("charging_amps" => 10)));

?>

Output:
array(1) { ["response"]=> array(2) { ["reason"]=> string(0) "" ["result"]=> bool(true) } }
```

pkuehnel, 2022-07-20:
Try not to use your url as teslamateBaseUrl but the URL from the readme: `http://teslamateapi:8080`. With this URL SmartTeslaAmpSetter is not routed over your Ethernet interface but can directly access the container inside the network which is created when you start all containers over `docker-compose up -d`. Note: In the URL: `teslamateapi` is the name of your service inside your `docker-compose.yml` and port 8080 is the default port of teslamateapi (The port you forword in your `docker-compose.yml` to access externally does not matter, as the containers communicate inside the network created by docker).

MrBetatester, 2022-07-20:
Thanks it seems to work now. I will check it the next few days how the power loading will be managed :-D. You can close the ticket if you do not need further information.

closed 2022-07-20 by pkuehnel