pkuehnel/TeslaSolarCharger #145: Feature Request: Home battery support.

MrBetatester, 2022-07-15:
Add more getter to collect information about SoC and power of the home Bettery. (Currently I manipulate the grid power to load the home battery to a desired SoC first)

pkuehnel, 2022-07-24:
One question regarding the home battery support: What would you wish as exact feature? As I am not having a home battery I only can assume things. In my opinion the feature should look like this:

New option to configure endpoint for battery SoC and charging/discharging power, then configure desired SoC (e.g. 90%) and desired minimum charging power until reaching this SoC (e.g. if your home battery is capable of charging with 5000W you set that value to 4000W so the battery (capable of adjusting charging power much quicker than the car) still has a buffer of 1000W e.g. if you switch off your microwave. As a result TeslaSolarCharger changes charging speed of the car to a point where your home battery is still charging with 4000W.

MrBetatester, 2022-07-25:
Yes this is exactly what i like to configure. Currently is do the following via the grid power endpoint. I subtract the discharging  power from the grid power.

```
<?php

    $ret = "";
    exec("/usr/bin/python /*/*/*/event_logger/pv/kostal_short.py", $ret);
    $ret = implode("\n", $ret);
    $data = json_decode($ret, true);

    $grid = intval($data["Grid"]);
    $bat = intval($data["Battery"]);

   if ($bat > 0) $grid -= $bat;

   echo $grid;
    
?>
```

The charging yesterday (9kWh-> home battery and 44kWh -> Car):
[image]
Our batterie(11kWh) charge/discharge with 2,5kW. Car charging started at 8 and ended 14. At 9 we started the coffee machine and the oven for making breakfast.

I think the updates a short enough to handle charging without keep space of 2,5KW for the home batterie. I only like to avoid loading the cars from the home batterie :-).

Maybe at autumn when we have not enough sun than I need the extra threshold for loading the home batterie first.

Twiederh, 2022-07-31:
The easiest way would be to have the battery fully charged first and then start the charge of the Tesla. This would work in most cases, except for situations, where there ist not enough sun during the day in winter (or the battery is too big for the solar installed). I tried previously with a script in Homeassistant that waited until the battery was charged to a certain percentage (I used 60%) and then used a buffer to charge the home battery slower and leave the rest for the car. This is more complicated and in most situations loading the home battery first is what most people would want. The option to load to a certain percentage and then continue to load only via Solar with the specified time and date by when the loading of the car needs to be finished leaves enough flexibility.

For me the only thing I want to avoid is that the power to load the Tesla is taken from the home battery since the efficiency would suffer severely.

MrBetatester, 2022-07-31:
Yes this is correct. At the Winter we set our home battery to 40%( Kostal recommend this to avoid loading the home bettery from grid) all PV power is used at the house itself. So we will also change the loading mode to full charge to get the minimum charge loss. At the summer days our home battery get a full circle.

pkuehnel, 2022-08-08:
Feature added with [Release 2.4.0](https://github.com/pkuehnel/TeslaSolarCharger/releases/tag/v2.4.0).

closed 2022-08-08 by pkuehnel