pkuehnel/TeslaSolarCharger #166: Feature Request: Plugin for Kostal Plenticore plus

VinciSC, 2022-07-21:
Kostal Pleticore Plus inverters do have a REST API, they require authentication. Found some projects accessing this API, most of them in pyton but also some in c#, e.g. marquisolivier/KostalApiClient. 
Sadly I have no clue how to integrate but I would love to test it as docker on my QNAP container station (Teslamate up and running).

pkuehnel, 2022-07-21:
As I can see [here](https://www.kostal-solar-electric.com/en-gb/products/accessories/smart-energy-meter/-/media/document-library-folder---kse/2020/12/15/14/22/ba_kostal_interface_ksem_de.pdf) Kostal also has a Modbus Interface. Modbus integration is currently in alpha release here: https://github.com/pkuehnel/TeslaSolarCharger/pull/165 but currently not documentation. In the coming days i will publish a documentation about this. Using Modbus is the prefered way as many inverters support it and it is without authentication.

VinciSC, 2022-07-22:
OK, thanks, wasn't aware of that interface and issue #165. Sounds good, I will test.

pkuehnel, 2022-07-23:
Modbus is now officially [released](https://github.com/pkuehnel/TeslaSolarCharger/releases/tag/v2.2.0). As I want to create templates for future users to make setup easier, I would be glad if you would send me your working configuration.

VinciSC, 2022-07-24:
Sure I will do so if I get it running. 
Used the docker-compose.yml from documentation section for a quick check but teslasolarcharger and grafana don't come up on my system (QNAP container station). Before that I had teslamate w/ teslamateapi running with no issues. Will check the log files in the next days, too busy today ...

pkuehnel, 2022-07-24:
Probably because of the volumes? In my docker-compose.yml I don't use docker volumes. The data of the containers are directly saved on the file system in directories next to your docker-compose.yml

VinciSC, 2022-07-25:
I just checked my logfiles, looks like the log in issue #186 reported by marteb.
`Unhandled exception. System.InvalidOperationException: Could not read BaseConfigurationJson file content.`
Will try the fix you proposed there and come back ...

pkuehnel, 2022-07-25:
You just have to update to the latest version.

pkuehnel, 2022-07-26:
Could you please provide logs of the modbus container? There are some debug messages of the originally read values and so on. And: can you please refer to the exact page on the documentation where you the register numbers from?

VinciSC, 2022-07-26:
The Kostal documentation I use can be downloaded [here](https://www.kostal-solar-electric.com/de-de/download/download/-/media/document-library-folder---kse/2020/12/15/13/38/ba_kostal-interface-description-modbus-tcp_sunspec_hybrid.pdf). The (imho) relevant register for feed power is described on page 18, first row in the table.

This is the logfile of my TeslaSolarCharger Container, grep'ed for "modbusplugin". Full log from same point in time is also available.
[logModBusPlugIn_20220726_2248.txt](https://github.com/pkuehnel/TeslaSolarCharger/files/9193260/logModBusPlugIn_20220726_2248.txt)

VinciSC, 2022-07-26:
BTW: I just checked the modbus settings of my inverter. It's configured on Big endian byte order as the documentation says on page 9, section 2.1.2.

pkuehnel, 2022-07-26:
Ok, I think I found the issue. Could you please check if at 17:25:30 (UTC time) the value should have been -1500.5?

In the logs you find the received byte array as hex value on every request. Just look for `Plugins.Modbus.Services.ModbusClient] Reversing Array` in your logs. In the logs you posted above, the last value (20:48:45.820) would be `406CCCCD` if you paste that value [here](https://gregstoll.com/~gregstoll/floattohex/) and press `convert to float` is the resulting value the correct value you are searching for? Maybe you have to check `Swap to use big-endian` option.

VinciSC, 2022-07-27:
The resolution in the Kostal app is very coarse but between 16:45 and 17:45 (UTC) it shows ~1.5 kW power feed, so -1500.5 would fit.
I did some analysis of the logfile this morning after recording (app screenshot every minute) the power feed for ~20 minutes. 
To me it seems the hex values in `Reversing Array` (columns D,F,H, J) fit to the power indicated by the app (column B) after FP2H conversion (columns E,G,I,K)  with [gregstoll.com](https://gregstoll.com/), see attached xls. 

[logfile_Messwerte.xlsx](https://github.com/pkuehnel/TeslaSolarCharger/files/9198730/logfile_Messwerte.xlsx)

VinciSC, 2022-07-27:
`Swap to use big-endian` was switched off.

pkuehnel, 2022-07-27:
The issue is very likley that the byte array is converted into a 32 bit integer value instead of a float value. As I have to restructure some things (factor multiplication should move to teslasolarcharger, away from modbusplugin) it will probably take until late evening tomorrow. It would help me for testing, if you could add a few values to your Excel file where the values in column B are negative (resulting in E, G, I, K being positive).

VinciSC, 2022-07-27:
I’ll check that this evening, have to work for some hours now 😊

VinciSC, 2022-07-27:
Btw: there are already some positive values in the table, at least F21 / G21, H21 / I21, J7 / K7. 

 

The reading from the Kostal App (column B) didn’t show these values as the scree update time is too long …

pkuehnel, 2022-07-27:
Ah sorry I missed them as I only checked B vor negative values. Should be enough. I will come back to you for testing an updated version, thanks.

VinciSC, 2022-07-27:
no worries and no hurry, I‘ll be glad to test.
Thank you, cool project 😎

pkuehnel, 2022-07-28:
New version is available. Just update your `docker-compose.yml` and use `:edge` instead of `:latest` tag for your modbusplugin container. Note: You have to use `GetFloatValue` instead of `GetInt32Value`. As I did not test compatibility between `:latest` teslasolarcharger and `:edge` modbusplugin I do not know if that is working.

VinciSC, 2022-07-28:
Did the update and a first quick test with calling `http://modbusplugin/api/Modbus/GetInt32Value?unitIdentifier=71&startingAddress=252&quantity=2&ipAddress=192.168.178.50&port=1502&factor=1&connectDelaySeconds=1&timeoutSeconds=10` several times.
Return value is` -211.2` all the time while inverter shows moving 1digit values around zero, positive and negative. Will check the logfiles and come back...

VinciSC, 2022-07-28:
Tried also with removed `&factor=1` in the call, same result.

pkuehnel, 2022-07-28:
ah sorry left a debug instruction in the code 😂 one moment...

pkuehnel, 2022-07-28:
Please update again, should work now...

VinciSC, 2022-07-29:
OK, now the return values of the call `http://modbusplugin/api/Modbus/GetFloatValue?unitIdentifier=71&startingAddress=252&quantity=2&ipAddress=192.168.178.50&port=1502&factor=1&connectDelaySeconds=1&timeoutSeconds=10` seem to be correct.
Let's wait for the sun and see if it starts charging ...

VinciSC, 2022-07-29:
Back again: Car was connected to the Wallbox in Home Zone over noon and there was varying sun and home energy consumption. The value of available power (red number in headline of overview) in TeslaSolarCharger looked good every time I checked. 
The car started charging with a very accurate charging power after it was plugged in.
After some time I noticed that charging process has been stopped. Over several hours the charging didn't restart although there where longer phases with available power. The car seems to be in sleeping mode (see attached pic). 

[image]

The settings for timing where 1digit at the beginning, finally I reduced to the following so this should not be the root cause:
[image]

I downloaded the logfile of my docker container and will analyze this evening but currently I have no idea what could be the issue. Any hints where to look? Maybe similar to issue #144 ... 
I suppose this is not a modbusplugin issue so maybe I should open a new issue and close this one?

pkuehnel, 2022-07-29:
Red means TeslaSolarCharger thinks you are sending 1716W to the grid. Just use the Invert Power tick (if you are on `:latest` with your teslasolarcharger).

pkuehnel, 2022-07-29:
[image]
Check this if you use `:latest`

This checkbox has been replaced by in (`:edge`):
[image]

There you would need to use `-1` as factor.

VinciSC, 2022-07-29:
> Power displayed in red means TeslaSolarCharger thinks you are sending 1716W to the grid. Just use the Invert Grid Power checkbox (if you are on `:latest` with your teslasolarcharger).

My understanding; the 1716W are exceeding house consumption and thus feed into the grid. TeslaSolarCharger "re-routes" this surplus power to my car if it holds for the configured "time with enough solar power...". Inverting sounds not straight forward to me but I'll try ...

I should be on `latest `for TeslaSolarCharger but container-station says image `build 2022/07/25 09:07:08` so I'll check and rebuild the docker ...
Do you recommend to switch to edge for TeslaSolarCharger?

pkuehnel, 2022-07-29:
> Inverting sounds not straight forward to me but I'll try ...

The issue is: It is not the same with every inverter what a value of 1716W means. Some return negative values when feeding into the grid, some return positive values when feeding into the grid. As a result my software needs to handle both ways. In your screenshot I see the number 1716 W displayed in read colour which means my software thinks you are consuming 1716W from the grid and that's why my software stopped charging and does not start it. If you check the Invert Value checkbox, my software internally multiplies the value from your inverter with -1 resulting in detecting 1716W as feeding into the grid which is correct with your inverter (but would be wrong e.g. with SMA). That results in your car starting to charge.

> Do you recommend to switch to edge for TeslaSolarCharger?

Simple answer: No, but if you have any issues opening the configuration page you have to upgrade (as `:latest` version does not support missing `&factor=1` in the GrodPowerUrl).

VinciSC, 2022-07-29:
OK, understood. You interpret pos. values as "available for PV charging". My (mis)understanding was inverted ... :-)
Will mark the checkbox and see what happens. Will take some hours, my wife is on the road and prefers the Tesla over the Golf diesel ...

VinciSC, 2022-07-29:
It works, posted my config in the disussion section.

Many thanks!

closed 2022-07-29 by VinciSC

pkuehnel, 2022-07-30:
@VinciSC I released a new version, you now can use `:latest` again for your modbus plugin to not get beta versions.
Note: like mentioned [here](https://github.com/pkuehnel/TeslaSolarCharger/issues/166#issuecomment-1199278961) Invert grid power value has been replaced by `Correction Factor` which should be updated to -1 automatically if you checked `Invert grid power` in the old version. Please leave a note if this is working for you after the upgrade.

VinciSC, 2022-07-30:
Did the update to `:latest` for both teslasolarcharger and modbus plugin. `Correction Factor` was already set as -1.0 when I opended teslasolarcharger UI (btw: all other settings where lost. If not a mistake on my side (?) it may be a future enhancement to persist the settings). 
Seems to work as currently available solar power is displayed as green value. Testing with car charging next week as my tesla is on the road again ...

pkuehnel, 2022-07-30:
Could you share your logs right after the first startup of the new version? The settings should not be lost (and were not on my tests and a few other users I already heard from). You can't do anything wrong. Just update and start (and do not delete the baseConfig.json, but as -1 was already set correctly the old values where there, default if no old values are there would be 1).

VinciSC, 2022-07-30:
Log since rebuilding with docker-compose see below.
To be more precise: in Base Configuration I had to reenter the Grid Power URL but the Correction Factor was already set to -1.0 .
Additionally I had to reenter my Power Buffer and both fileds Time with[out] enough solar power.
In Car Settings I had to reenter Max Amp and Usable kWh, where on 16 resp. 75, my setting is 15 and 55 (2019 SR+).

[logTeSoChMo_20220730_1908.txt](https://github.com/pkuehnel/TeslaSolarCharger/files/9226977/logTeSoChMo_20220730_1908.txt)

pkuehnel, 2022-07-30:
Hmm I just tried with a blank installation and everything was fine. Probably an issue with something that we messed up the last days.

VinciSC, 2022-08-01:
Maybe my usage of QNAPs container station and thus docker-compose to run the app is the reason. 

Or how I do the updates.

I will check baseConfig.json before and after next update …

pkuehnel, 2022-08-02:
`docker-compose pull teslasolarcharger` `docker-compose up -d` important is, that the new container has access to the old baseConfig.json