pkuehnel/TeslaSolarCharger #185: Restart-loop since 2.2.2

haexpo, 2022-07-25:
The Docker container is in an restart-loop since 2.2.2:

[09:06:51.455 VRB TeslaSolarCharger.Server.Helper.EnvironmentVariableConverter] ConvertAllValues()
[09:06:51.794 DBG TeslaSolarCharger.Shared.Wrappers.ConfigurationWrapper] Config value extracted: [ConfigFileLocation]: configs
[09:06:51.794 DBG TeslaSolarCharger.Shared.Wrappers.ConfigurationWrapper] Config value extracted: [ConfigFileLocation]: configs
[09:06:51.795 DBG TeslaSolarCharger.Shared.Wrappers.ConfigurationWrapper] Config value extracted: [BaseConfigFileName]: baseConfig.json
[09:06:51.795 DBG TeslaSolarCharger.Shared.Wrappers.ConfigurationWrapper] Config value extracted: [BaseConfigFileName]: baseConfig.json
Unhandled exception. System.InvalidOperationException: Could not read BaseConfigurationJson file content.
   at TeslaSolarCharger.Shared.Wrappers.ConfigurationWrapper.BaseConfigurationJsonFileContent() in /src/TeslaSolarCharger/Shared/Wrappers/ConfigurationWrapper.cs:line 370
   at TeslaSolarCharger.Shared.Wrappers.ConfigurationWrapper.IsBaseConfigurationJsonRelevant() in /src/TeslaSolarCharger/Shared/Wrappers/ConfigurationWrapper.cs:line 404
   at TeslaSolarCharger.Server.Helper.EnvironmentVariableConverter.ConvertAllValues() in /src/TeslaSolarCharger/Server/Helper/EnvironmentVariableConverter.cs:line 24
   at Program.<Main>$(String[] args) in /src/TeslaSolarCharger/Server/Program.cs:line 88
   at Program.<Main>(String[] args)

pkuehnel, 2022-07-25:
I will look into it.

pkuehnel, 2022-07-25:
Could you please provide me the TeslaSolarCharger part of your docker-compose.yml and your configs file directory? baseConfig.json would be enough.

haexpo, 2022-07-25:
docker-compose.yml:

smartteslaampsetter:
    image: pkuehnel/smartteslaampsetter:latest
    logging:
        driver: "json-file"
        options:
            max-file: "5"
            max-size: "10m"
    restart: always
    depends_on:
      - teslamateapi
    environment:
      - CurrentPowerToGridUrl=http://secret/TeslaPower.php
      - TeslaMateApiBaseUrl=http://192.168.178.195:8080
      - UpdateIntervalSeconds=30
      - CarPriorities=2|1
      - GeoFence=Home
      - MinutesUntilSwitchOn=2
      - MinutesUntilSwitchOff=2
      - PowerBuffer=0
      - TZ=Europe/Berlin
      - TeslaMateDbServer=database
      - TeslaMateDbPort=5432
      - TeslaMateDbDatabaseName=teslamate
      - TeslaMateDbUser=teslamate
      - TeslaMateDbPassword=secret
    ports:
      - 7190:80
    volumes:
      - teslaampsetter-configs:/app/configs

In the configs dir 
/var/lib/docker/volumes/pi_teslaampsetter-configs/_data

there is only one file: „carConfig.json"

haexpo, 2022-07-25:
The alpha Release works! Thanks a lot!

pkuehnel, 2022-07-25:
Please go back to `:latest` as `:alpha` is very often messed up. If you save your base configuration in the frontend one time `:latest` should work without issues. Note: After that all Environment variables (except `TZ`) are ignored, as only BaseConfiguration is taken into account.

closed 2022-07-25 by pkuehnel