pkuehnel/TeslaSolarCharger #186: TeslaSolarCharger container keeps restarting

marteb, 2022-07-25:
I added TeslaSolarCharger to an existing TeslaMate installation by adding the sections for `teslamateapi` and `teslasolarcharger` to my existing `docker-compose.yml` file and adjusting the values for `DATABASE_PASS` and `ENCRYPTION_KEY` (for teslamateapi). Then I ran the command `docker-compose up-d`.

But unfortunately there is no http service answering on port 7190. After a look in `docker ps` I get the following: `pkuehnel/teslasolarcharger:latest   "dotnet TeslaSolarCh…"   About an hour ago    Restarting (132) 5 seconds ago`

`docker logs` for the teslasolarcharger container gives:

`[09:11:36.845 VRB TeslaSolarCharger.Server.Helper.EnvironmentVariableConverter] ConvertAllValues()
[09:11:38.221 DBG TeslaSolarCharger.Shared.Wrappers.ConfigurationWrapper] Config value extracted: [ConfigFileLocation]: configs
[09:11:38.223 DBG TeslaSolarCharger.Shared.Wrappers.ConfigurationWrapper] Config value extracted: [ConfigFileLocation]: configs
[09:11:38.224 DBG TeslaSolarCharger.Shared.Wrappers.ConfigurationWrapper] Config value extracted: [BaseConfigFileName]: baseConfig.json
[09:11:38.225 DBG TeslaSolarCharger.Shared.Wrappers.ConfigurationWrapper] Config value extracted: [BaseConfigFileName]: baseConfig.json
Unhandled exception. System.InvalidOperationException: Could not read BaseConfigurationJson file content.
   at TeslaSolarCharger.Shared.Wrappers.ConfigurationWrapper.BaseConfigurationJsonFileContent() in /src/TeslaSolarCharger/Shared/Wrappers/ConfigurationWrapper.cs:line 370
   at TeslaSolarCharger.Shared.Wrappers.ConfigurationWrapper.IsBaseConfigurationJsonRelevant() in /src/TeslaSolarCharger/Shared/Wrappers/ConfigurationWrapper.cs:line 404
   at TeslaSolarCharger.Server.Helper.EnvironmentVariableConverter.ConvertAllValues() in /src/TeslaSolarCharger/Server/Helper/EnvironmentVariableConverter.cs:line 24
   at Program.<Main>$(String[] args) in /src/TeslaSolarCharger/Server/Program.cs:line 88
   at Program.<Main>(String[] args)`

What did I miss or got wrong?

pkuehnel, 2022-07-25:
Could you please send me the TeslaSolarCharger part of your docker-compose.yml

marteb, 2022-07-25:
Here it is:

`  
teslasolarcharger:
    image: pkuehnel/teslasolarcharger:latest
    logging:
        driver: "json-file"
        options:
            max-file: "10"
            max-size: "100m"
    restart: always
    depends_on:
      - teslamateapi
    environment:
      - TZ=Europe/Berlin ##You can change your Timezone here
    ports:
      - 7190:80
    volumes:
      - ./teslasolarcharger-configs:/app/configs
`

pkuehnel, 2022-07-25:
Could you please try again with `image: pkuehnel/teslasolarcharger:alpha`? I fixed a possible issue but I am not sure if this addresses your problem as I can not reproduce the issue.

marteb, 2022-07-25:
Yes, with the alpha release it works!

pkuehnel, 2022-07-25:
It crashes before the first setup ;-)
Thanks for your quick response.

closed 2022-07-25 by pkuehnel

ViktorPankraz, 2023-06-14:
```
  teslasolarcharger:
    image: pkuehnel/teslasolarcharger:latest
    container_name: teslasolarcharger
    logging:
        driver: "json-file"
        options:
            max-file: "10"
            max-size: "100m"
    restart: always
    depends_on:
      - teslamateapi
    environment:
      - Serilog__MinimumLevel__Default=Verbose #uncomment this line and recreate container with docker-compose up -d for more detailed logs
      - TZ=Europe/Berlin ##You can change your Timezone here
    ports:
      - 7190:80
    volumes:
      - teslasolarcharger-configs:/app/configs
```
For the log file I checked the path
LogPath: /var/lib/docker/containers/866a66c88bca57971a69a9726958ae0492f390affd806a6f69366b572305c717/866a66c88bca57971a69a9726958ae0492f390affd806a6f69366b572305c717-json.log

but that file is empty.

pkuehnel, 2023-06-14:
this was the wrong issue ;-)