pkuehnel/TeslaSolarCharger #188: LoadBalancing-Feature: MaxAmp for 2 cars

haexpo, 2022-07-25:
It would be helpful, to be able to set a MaxAmp (e. g. 16A) that can not be exceeded while charging both cars a the same time.

pkuehnel, 2022-07-25:
What's your main goal with this? Prevent charge stop of the seceond car? Or reducing combined load?

haexpo, 2022-07-25:
The main goal would be to reduce the combined load. I wouldn't have to use the load balancing feature that comes with the wall connectors V3 which may interferes with TeslaSolarCharger (haven't tested that).

pkuehnel, 2022-07-25:
You can use that feature without issues since [v1.8.0](https://github.com/pkuehnel/TeslaSolarCharger/releases/tag/v1.8.0)

pkuehnel, 2022-07-28:
Please reopen the issue if you have problems with Loadbalancing of your Wall Connector

closed 2022-07-28 by pkuehnel