pkuehnel/TeslaSolarCharger #269: Name or service not known (smaplugin:80)

fprucker, 2022-09-04:
Hi!

Under 'Overview' I can see my Tesla Wattage but not the Suppy/Overage of my SMA SHM 2.0.

Grid Power Url is set to http://smaplugin/api/CurrentPower/GetPower

According to 'docker logs teslasolarcharger' there is a problem with the smaplugin:

[15:18:30.167 VRB TeslaSolarCharger.Server.Services.PvValueService] GetHttpResponse(http://smaplugin/api/CurrentPower/GetPower, {})
System.Net.Http.HttpRequestException: Name or service not known (smaplugin:80)
 ---> System.Net.Http.HttpRequestException: Name or service not known (smaplugin:80)
   --- End of inner exception stack trace --- [See nested exception: System.Net.Http.HttpRequestException: Name or service not known (smaplugin:80)

According to 'docker logs teslasolarcharger_smaplugin' the plugin is running and providing the desired values:

[13:18:34.846 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] current supply: 0
[13:18:34.846 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] current overage: 6601.8
[13:18:34.846 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] Waiting for new values

Let me know if there is anything further I can provide.

Best regards,
Flo

pkuehnel, 2022-09-04:
Looks like a mistake in my documentation. Instead of 
```
http://smaplugin/api/CurrentPower/GetPower
```
use 
```
http://<ipToYourDockerHost>:7190/api/CurrentPower/GetPower
``` 
e.g.
```
http://192.168.1.50:7190/api/CurrentPower/GetPower
``` 
as `Grid Power Url` in your `Base Configuration`

pkuehnel, 2022-09-04:
Note: I edited the Port from `7190` to `7192` int my comment above.

fprucker, 2022-09-04:
wow, that was quick. you are amazing! works now. thank you!
(my setup: SHM2.0/STP25/Synology/Docker/TeslaSolarCharger/smaplugin/ModelY22)

how can I donate?

pkuehnel, 2022-09-04:
Just use the pink Donate button in the first line of the Readme or [this direct link](https://www.paypal.com/donate/?hosted_button_id=S3CK8Q9KV3JUL)
Thank you 👍

closed 2022-09-04 by pkuehnel