pkuehnel/TeslaSolarCharger #281: Issue "Could not access Database"

VinciSC, 2022-09-07:
Since update to 2.5 the issue "Could not access Database" is displayed on the overview page but TeslaSolarCHarger seems to work (Tesla is charged with overage PV power).

Checked the proposed possible solutions: 
- Database container is running 
- PW is set correctly in yml file
- regarding "database service name" I'm in doubt: I guess this refers to the entry "TeslaMate Database Server Name" on the base config page and to the section 
```
 database:
    image: postgres:[...]
```
in the yml file. These entries are identical so this seems to be OK.

As I'm running TeslaSolarCharger on a QNAP NAS within Container-Station I opened a terminal and checked the running services typing `docker ps`. Result:

```
CONTAINER ID   IMAGE           COMMAND                  CREATED       STATUS         PORTS             NAMES
[..]
e3f51caf95d0   postgres:14     "docker-entrypoint.s…"   2 days ago    Up 2 minutes   5432/tcp          tesochmo_database_1
[..]
```
Now I wondering about the stated Name `tesochmo_database_1`. `tesochmo` is the name i gave the container within QNAPs container-station, no ides why `_1` is added.
I guess this is the root cause but are unsure how to solve as the service name `database` is used at several different positions in the yml (3 `environment:` variables and one entry ` depends on:`).

Any suggestions before trying out (changing yml `env` and `depends on` to `tesochmo_database_1`? 
Any idea why TesloSolarCharger seems to work without database access?

pkuehnel, 2022-09-07:
That is because only since 2.5 there runs a check if everything is working. With 2.5 the database is not uses, yet. But from 2.6 on the Database will be used. That's why everything is still working. I just wanted to display the issue before it causes any problems.

You probably don't use the default database password, so you need to go to the `Base Configuration` page in your TeslaSolarCharger and scroll down to database. There you probably only have to insert the correct database username password and the issue should be gone.
[image]

VinciSC, 2022-09-07:
The PW in the .yml was set correctly, I already checked this before first post. 
However I set the PW in the base config again (copy&paste from the .yml :-) ), now the issue isn't displayed any more. Magic ...

Thanks for the hint!

closed 2022-09-07 by VinciSC

pkuehnel, 2022-09-07:
Since I migrated all settings from the .yml file (only TeslaSolarCharger part) to the configuration page (for better error handling) whatever you have set in the .yml (except Timezone) has no impact anymore.