pkuehnel/TeslaSolarCharger #283: Float values for correction factors

schneefloh, 2022-09-08:
My Victron system presents some modbus values like battery SOC with a factor of 10, e.g. 98% = 980. 
Correction factors only accept decimal values but I need 0.1 in this case.

pkuehnel, 2022-09-08:
`0.01` is working for me and calculated correctly:

[image]

Are you sure you pasted the value in the correct field?

schneefloh, 2022-09-08:
Yes, I think so. As soon as I hit enter the value reverts to 1. 
Even tried on different browsers and using the latest 2.5.0.
[image]

schneefloh, 2022-09-08:
addition: if I type 1.0 and hit enter it reverts to 10?

pkuehnel, 2022-09-08:
But it does not change when you leave the field? Only after submitting?

pkuehnel, 2022-09-08:
Which hardware and which OS do you use? What is the language of your OS? Could you please upload the logs of the `teslasolarcharger` container?

If you use the latest `docker-compose.yml` you can create a lofgile with
```
docker logs teslasolarcharger > teslasolarcharger.log
```

If you use an older version of the `docker-compose.yml` you probably have to use a different containername.

schneefloh, 2022-09-08:
It reverts to a value without dot even when I press tab or click somewhere else.
Docker runs on a Synology NAS with language set to German.
Logfile attached.
Thanks a lot for your help!
[teslasolarcharger.log](https://github.com/pkuehnel/TeslaSolarCharger/files/9530628/teslasolarcharger.log)

pkuehnel, 2022-09-08:
Ok, then this already happens in the browser and the logs don't help. please change your image to 
``` 
pkuehnel/teslasolarcharger:decimal
```
Does this help (version should be `2.5.1-alpha.0.4`)

schneefloh, 2022-09-09:
I can confirm everything works as expected in 2.5.1-alpha.0.4 without other changes.
Thanks!

pkuehnel, 2022-09-09:
You can stay on `:decimal` for now. The bugfix will be in the next release, then you can go to `:latest` again.

closed 2022-09-09 by pkuehnel

pkuehnel, 2022-09-21:
You knwo can go back to `:latest`.