pkuehnel/TeslaSolarCharger #362: using own mqtt with authorization

gaborfri, 2022-10-06:
i am not able to connect my own mqtt server. the server itself is configured in the settings, but only as URL (the port is added automatically). I seem to be missing the possibility to specify the user and password

pkuehnel, 2022-10-06:
What exactly do you mean with URL?
You can add your MQTT server + port (e.g. `192.168.1.50:1883`), your username + password for authentication in the MQTT server + Topics for grid power, inverter power (not needed just for displaying the value), and optional your home battery charging/discharging power and home batter soc:
[image]

gaborfri, 2022-10-06:
i mean not this setting. at the very end of the configuration part called "Mosquito servername". Here is only an input for the servername, but no configuration for the username/password.
At the end i get the following error message in my mosquitto log:
Client TeslaSolarCharger disconnected, not authorised

gabor

pkuehnel, 2022-10-06:
Does TeslaMate support the MQTT server to have authentication?

pkuehnel, 2022-10-30:
Seems to be irrelevant as no answer in 24 days.

closed 2022-10-30 by pkuehnel

idominiki, 2023-03-12:
Hi @pkuehnel 
I have an MQTT Server running on my home assistant which teslamate connects to. Is it possible to using this MQTT for TeslaSolarCharger? 
I don't have the settings to enter MQTT username and password in the base settings.
Thanks
Dominik

pkuehnel, 2023-03-16:
This is currently not possible.

pkuehnel, 2023-03-16:
https://github.com/pkuehnel/TeslaSolarCharger/issues/673

But this won't be possible in the nearer future.

BillGOH, 2023-08-27:
First of all, thanks for this great software.

Unfortunately I am running (a very busy) MQTT broker on my openHAB server with authentication (which Teslamate uses; and then openHAB uses the data). It would be nice if one could choose another MQTT server (with credentials), not the "built in" one.

pkuehnel, 2023-08-27:
You can change the MQTT server but (for now) credentials are not support.

BillGOH, 2023-08-28:
Thank you for your quick response. I added `allow_anonymous true` to the openHAB MQTT server config (= mosquitto) and it works fine now!