pkuehnel/TeslaSolarCharger #379: SoC limit of 100% is not being consumed

fprucker, 2022-10-17:
When I set the SoC limit in the Tesla app to 100%, TSC shows (and uses) 80%. If I set it to 99% or less, it correctly shows (and uses) the the set value. Can you reproduce this?

(use case: for a longer trip I want to use TSC to charge to 100% by a certain target date/time)

pkuehnel, 2022-10-17:
As my car is in Service at least until Friday I can't test right now. Are you talking about this value:
[image]

This should be the same as TeslaMate displays it. Does TeslaMate show 100% while TeslaSolarcharger displays 80%?

fprucker, 2022-10-17:
Exactly that value. Any value aside from 100% is adopted, but if you set it to 100% it shows 80% (or the last value which was different to 100% - have not thoroughly checked this). 

My workaround for now is setting it to 99% :-)

pkuehnel, 2022-10-22:
Got my car back and tried it out. 100% works for me Without any issues.
When you tried, was `StateString` displayed as `suspended`?
[image]

This would mean, that TeslaMate is trying to let the car fall asleep. While this is happening, No changes are deteced. If you still have this issue, please [share your logfiles](https://github.com/pkuehnel/TeslaSolarCharger#generate-logfiles).

fprucker, 2022-12-10:
Sorry for the late feedback. The problem was that it takes some time for your software /teslamate to pick up the car max SoC setting. After waiting for a minute or two the 100% max SoC appeared in your software and then I also was able to set it to 100% without an error message.

pkuehnel, 2022-12-11:
TeslaMate refreshes all values from TeslaMate every 10 seconds (if refreshing is not stopped due to TeslaMate letting the car fall asleep). My software refreshes the values immediately after TeslaMate gets new values. The values should be updated in the browser every five seconds, but it looks like this refresh in the browser only occurs for the charge limit if the whole page is refreshed. I will look into it. Note: Charging behavior is not affected by this issue, only the displayed value in the browser is not auto refreshed.

pkuehnel, 2022-12-25:
This issue will be fixed in the next version.

closed 2022-12-25 by pkuehnel