pkuehnel/TeslaSolarCharger #380: Exclamation mark in status

VinciSC, 2022-10-17:
Hi,
since updating from 2.7.5 to 2.7.7 I see an encircled exclamation mark below the current SoC and charging power (second line in picture below). 
[image]
No idea what is wrong. 
Any hint how to solve or what to check?
Thanks
VinciSC

VinciSC, 2022-10-17:
Checked TeslaMate, no red exclamation mark there. Scrolled down in TSC, the line "Car Healthy" was empty.
It looks like the car charged during afternoon (SoC from 56 to 63%) while the exclamation mark was present.
Restarted all containers now. The exclamation mark is gone and "Car Healthy" reports true.
Thanks for advice!
Regards
VinciSC

closed 2022-10-17 by VinciSC

pkuehnel, 2022-10-17:
Issue is probably that the bool value defaults to false. Will be fixed with https://github.com/pkuehnel/TeslaSolarCharger/issues/381