pkuehnel/TeslaSolarCharger #445: Collaborate with evcc?

andig, 2022-11-24:
Hallo pkuehnel, hättest Du Lust, mal mit https://evcc.io zu sprechen um zu schauen, ob/wie wir unsere beiden Lösungen vielleicht integrieren können? Wenn Du magst melde Dich gerne hier oder direkt bei https://evcc.io/slack

pkuehnel, 2022-11-24:
Wir haben ja schon mal über das Thema gesprochen, [damals hast du mir leider nicht weiterhelfen können](https://github.com/evcc-io/evcc/discussions/2309), worauffolgend ich ja erst mit diesem Projekt angefangen habe. Wie hättest du dir die Integration vorgestellt?

andig, 2022-11-25:
Argh- der Thread ging ja noch ne ganze Weile weiter und schien mir mit API-Doku erledigt. Was brauchst Du denn konkret? Oder wenn wirs umdrehen wollen: hast Du ein Api an dem ich mich orientieren könnte?

pkuehnel, 2022-11-25:
Mein Ziel wäre damals eigentlich gewesen, dass ich eine Middleware schreibe, die sich für EVCC als Auto (zum Abfragen des SOC) und gleichzeitig als Charger (zum setzen der Ampere) ausgibt, was ich auch nach wie vor als die einfachste Variante für dich sehen würde. Aber nachdem ich jetzt eine funktionierende Lösung habe, bin ich eigentlich nicht wirklich gewillt von vorne anzufangen...

Das Problem ist, das Ganze war ja ursprünglich nur für mich alleine gedacht und da ich [TeslaMate](https://github.com/adriankumpf/teslamate) im Einsatz habe und durch meine Software keine Schlafprobleme vom Auto bekommen wollte, habe ich meine Software sehr stark mit TeslaMate verzahnt => aktuell funktioniert meine Software ausschließlich zusammen mit TeslaMate. Was bei mir aktuell das größte Problem ist: Die Anbindung an die x verschiedenen SmartMeter/Akkus/Wechselrichter, also was für mich erst mal am einfachsten wäre: Du stellt mir als REST Endpunkt den aktuellen Überschuss in Watt zur Verfügung (als wärst du das SmartMeter) und ich melde dir in Watt zurück, was das Auto gerade lädt (ggf. mit SoC, Ampere, Volt), das würde mehr oder Weniger schon out of the box funktionieren von meiner Seite. Ist halt für den Anwender sehr viel Overhead zum konfigurieren mit TeslaMate usw. Du hast ja selbst auch direkten Zugriff auf die API, oder?

fprucker, 2022-11-29:
Kann leider nichts dazu beitragen - aber eine Integration in evcc wäre wirklich toll. Macht ja keinen Sinn, wenn die Anbindung an die x verschiedenen SmartMeter/Akkus/Wechselrichter doppelt gepflegt werden muss. Es geht nicht nur darum, smarte Teslas an 'dummen' Wallboxes zu steuern, sondern auch darum, für all die Leute evcc zugänglich zu machen, die gar keine Wallbox haben (UMC, etc.)

pkuehnel, 2022-11-29:
Ich wäre offen für alles, was ich allerdings nicht möchte, ist das abfragen des Zustands des Fahrzeugs zu übernehmen, aufgrund der Schlaf Thematik. Gerne SOC usw. per MQTT an mich + ich kann auch die direkte Steuerung des Tesla übernehmen, also mich insgesamt unabhängig von TeslaMate machen, aber die Zustandsabfragen möchte ich nicht übernehmen. @andig Wäre das möglich?

andig, 2022-11-29:
Den SoC schickt evcc raus- allerdings für alle Fahrzeuge. Wir müssten sicher was machen um einen spezifischen Tesla (?) als solchen zu erkennen. Aufwecken tun wir das Auto aktuell nicht (HTTP 408)- wenns schläft dann ist das so. Wäre das notwendig? Bevor wir da ins Detail gehen müssten wir erstmal das Zusammenspiel verstehen.

pkuehnel, 2022-11-29:
Ok, bevor ich eine Liste aller Werte zusammensuche, die ich aktuell verarbeite, mehrere Fragen an dich:
1. Wie sorgst du aktuell dafür, dass ein Tesla zu schlafen beginnt? Er bleibt ja wach, wenn permanent Daten von der API abgerufen werden.
2. Wie hast du dir die Regelung vorgestellt? Macht ja vermutlich eher wenig Sinn, wenn du einige Geräte hast und ich die Teslas und dann regeln wir beide, das wird sich vermutlich aufschaukeln. Ich kann aktuell folgendes abbilden:
   * Laden erst starten, wenn x Minuten genug Überschuss für Mindestladeleistung vorhanden
   * Laden erst stoppen, wenn y Minuten Netzbezug, auch bei geringster Ladeleistung
   * Laden erst starten, wenn Hausakku zu x % voll
   * Wenn Hausakku mit mindestens x Watt lädt und dennoch Überschuss da ist Tesla laden starten
Ich denke das kannst du auch alles abbilden, oder? Dann Würde es Sinn machen, wenn du mir nur die Leistung gibst, mit der Tatsächlich geladen werden soll und ich das sofort versuche umzusetzen.
3. UI: Wie hast du dir die Konfiguration vorgestellt? Zum Zeitpunkt des Threads bei evcc war die Konfiguration ja über die YAML Datei. Ich habe bei mir mittlerweile alles im Web, weil YAML nicht jedermanns Sache ist. Meine präferierte Variante wäre: Alles bei dir konfigurieren und du schickst mir dann die Konfiguration (bzw alles was ich davon brauche) über einen REST Endpunkt.
4. Können wir uns auf einen kleinen Donate Button irgendwo in der Software bei dir einigen? Oder wo auch immer du deinen hast.

andig, 2022-11-29:
1. Gar nicht aktiv. Es wird einfach nur relativ selten und nur während des ladens abgefragt.
2. Die ist in evcc schon komplett drin und wird verwendet.
3. Yaml. Erstmal gehts ja nur um Funktion.
4. Klar, da finden wir eine Lösung.

pkuehnel, 2022-11-29:
1. Ok, dann würde ich die komplette Steuerung und das Auslesen der Daten von Tesla übernehmen, dadurch dass es ja keine Wallbox gibt, kann ich ja nur über eine Abfrage am Auto rausfinden, ob das Auto im Geofence und angesteckt ist, sprich einfach relativ selten Abfragen is nicht...
2. Dann würde ich sagen du sagst mir die Leistung und ich lade mit der Leistung. Wie funktioniert MQTT bei dir? Hast du einen separaten Broker, oder bist du selbst der Broker?
3. Ok passt, Tesla Token usw. würde ich dann auch von dir nehmen, ist ja in deiner yaml konfigurierbar, weil was ich vermeiden will ist, dass ein Teil bei dir in der yaml konfiguriert ist und ein Teil bei mir irgendwo anders... Würde auch das komplette UI bei dir lassen.
4. Perfekt.

Wie wollen wir weitermachen und in welchem zeitlichen Horizont hast du das Ganze geplant? Wo du dir noch was überlegen musst ist denke ich die Starre Wallbox, die es bei dir gibt... Die Wallbox Konfig müsste dynamisch sein, Phasen können sich ändern, maximal mögliche Ampere können sich ändern usw.

andig, 2022-11-30:
> Ok, dann würde ich die komplette Steuerung und das Auslesen der Daten von Tesla übernehmen

Welchen Teil macht denn der TWC dabei? Ans Auto komme ich ja prinzipiell ran, einschl. Geoposition.

> Dann würde ich sagen du sagst mir die Leistung und ich lade mit der Leistung

Das passiert wieder über das Auto? Freigegeben wird der Strom- oder ist das bei Tesla anders?

> Hast du einen separaten Broker, oder bist du selbst der Broker?

Separat.

> Die Wallbox Konfig müsste dynamisch sein, Phasen können sich ändern, maximal mögliche Ampere können sich ändern usw.

Was meinst Du damit? Apis gibts ja- nur wer steuert da was zu welchem Zweck?

pkuehnel, 2022-11-30:
> Welchen Teil macht denn der TWC dabei?

Gar nichts, genau das ist was meine Software macht. Zugriff aufs Auto und was eine Wallbox / Juice Booster / UMC oder was auch immer macht ist völlig egal.

> Das passiert wieder über das Auto? Freigegeben wird der Strom- oder ist das bei Tesla anders?

Alles passiert über das Auto, es wird auch der Strom freigegeben, aber um den Strom zu berechnen muss ich Volt (ist ja nicht gleich weiltweit) und Anzahl der Phasen wissen, kann ja sein, dass ich heute an der dreiphasigen Wallbox lade und morgen mit Schuko.

> Was meinst Du damit? Apis gibts ja- nur wer steuert da was zu welchem Zweck?

So weit ich weiß ist ja bei dir in der Konfig fest definiert, wie viel Ampere eine Wallbox kann, nachdem ich aber direkt das Auto unabhängig der Wallbox steuere, ist das dynamisch.

andig, 2022-11-30:
> Gar nichts, genau das ist was meine Software macht. 

Ok- und was *macht* sie dann (also mit dem TWC)?

> aber um den Strom zu berechnen muss ich Volt (ist ja nicht gleich weiltweit) und Anzahl der Phasen wissen, kann ja sein, dass ich heute an der dreiphasigen Wallbox lade und morgen mit Schuko.

Auch das macht evcc schon alles.

> So weit ich weiß ist ja bei dir in der Konfig fest definiert, wie viel Ampere eine Wallbox kann, nachdem ich aber direkt das Auto unabhängig der Wallbox steuere, ist das dynamisch.

Spielt m.E. keine Rolle. Das physikalische Limit wird von der Verdrahtung definiert und ist in Hardware abgesichert. Ansonsten -> Api aufrufen.

pkuehnel, 2022-11-30:
> Ok- und was macht sie dann (also mit dem TWC)?

Gar nichts... Software läuft voellig autark von irgendeiner Wallbox... ob TWC oder goe oder was auch immer... Ich fasse die Wallbox nicht an.

> Spielt m.E. keine Rolle. Das physikalische Limit wird von der Verdrahtung definiert und ist in Hardware abgesichert.

Für die Regelung spielt es schon eine Rolle... Wenn ich 2000 Watt mehr Überschuss habe, muss ich wissen, ob ich den Tesla noch 2000 Watt schneller laden kann, oder, nur um 690 Watt und den Rest gebe ich an Auto Nr. 2

Um ehrlich zu sein sehe ich (für dich) den Sinn der Collaboration nicht ganz. Das ganze verkompliziert nur alles. Wenn du eh schon Zugriff auf die API hast, musst du einfach nur die Werte im Tesla statt in der Wallbox setzen.

andig, 2022-12-01:
> Für die Regelung spielt es schon eine Rolle... Wenn ich 2000 Watt mehr Überschuss habe, muss ich wissen, ob ich den Tesla noch 2000 Watt schneller laden kann, oder, nur um 690 Watt und den Rest gebe ich an Auto Nr. 2

Das macht alles evcc. Und ziemlich gut ;)

> Gar nichts... Software läuft voellig autark von irgendeiner Wallbox... ob TWC oder goe oder was auch immer... Ich fasse die Wallbox nicht an.

Mhhm. Aber was tut die Software dann denn die Fahrzeugdaten von evcc kommen und die Überschusssteuerung auch da erfolgt?

fprucker, 2022-12-01:
… den Tesla steuern statt die Wallbox. EVCC ohne Wallbox. Ich z.B. lade mit einer nicht smarten Delta Wallbox und mit UMC.

andig, 2022-12-01:
Dann geht es doch nur noch drum `set_charging_amps` beim Tesla zu setzen? Da evcc die Fahrzeugverbindung eh hat kann ich das dann natürlich auch direkt tun?

pkuehnel, 2022-12-01:
> Um ehrlich zu sein sehe ich (für dich) den Sinn der Collaboration nicht ganz

Meine Rede

closed 2022-12-01 by pkuehnel