pkuehnel/TeslaSolarCharger #466: Global Max Ampere

fprucker, 2022-12-20:
With two cars and some other loads I sometimes run into the limits of my grid connection. In addition to the existing per-car current limits, would it be possible to introduce
1. a maximum current on the grid connection (and dynamically limit the cars accordingly by priority) and/or
2. a maximum current on the sum of the cars (instead of just per car) and (and dynamically limit the cars accordingly by priority)?

pkuehnel, 2022-12-20:
I am experiencing the same issue. Option 1 would be the better solution, but with many drawbacks:
1. Not all smart meters return current per phase
2. Much more configuration effort (new API call for each phase)
3. If charging on only one phase, TSC needs to know from which smart meter phase the car is charging
4. Much more development effort for me

With option 2, it is not easy to use the maximum possible current as current used by other loads is unknown. But I will implement option 2 for now due to the above drawbacks. It might take ~two weeks.

Can I come back to you when the feature is finished for beta testing?

fprucker, 2022-12-20:
thank you and sure, happy to join the testing!

regarding option 1: a simplified version could cap power instead of current per phase?

pkuehnel, 2022-12-20:
I will test both options; I will probably implement both: A power cap based on smart meter values and a current cap based on TSC-managed cars.

pkuehnel, 2023-01-02:
I tried a couple of things but in the end, I chose to only implement a current cap based on all managed cars. I have the same issue as you do but implementing an option to get the current from each phase of your smart meter is way too much effort (regarding development and configuration for the users). If you want you can try it with the tag `:edge`.
You can set the desired current in the base configuration page:
[image]

What I can offer is adding a REST API endpoint where you can dynamically update this value, so if you have something like IOBroker you can write a script setting the max combined current value in TeslaSolarCharger every ~30 seconds based on the maximum current of your smart meter.

pkuehnel, 2023-01-02:
As implementing this was very easy, I just did it. If you want to change the value, call a Get Request:
`http://192.168.1.50:7190/api/BaseConfiguration/UpdateMaxCombinedCurrent?maxCombinedCurrent=23`

This would update the value to 23 Ampere. After that change no restart whatsoever is needed, so you can change this value whenever you want and it is used immediately.

fprucker, 2023-01-05:
thanks a lot; I have only tried with 1 car and so far it works. will try later with two cars.
unfortunately I do not have ioBroker, I only have an SMA inverter and SHM2.0.

Regarding the hint I suggest to change this to: "Set a value if you want to reduce the max combined used current PER PHASE of all cars. E.g. if you have two cars each set to max 16A but your INSTALLATION can only handle 20A PER PHASE you can set 20A here. So if one car uses 16A  PER PHASE the other car can only use 4A PER PHASE. Note: Power is distributed based on the set car priorities.

fprucker, 2023-01-05:
tried with 2 cars; I can confirm that the feature works as expected (caps global current and adheres to priority by first maxing out on max (individual) amps on higher priority car and distributing the remaining amps (if any) to the lower priority car. perfect!

pkuehnel, 2023-01-06:
max combined Current Help text will be updated in the next version, thanks for your input.

closed 2023-01-06 by pkuehnel

fprucker, 2023-01-11:
do you also experience that initially the cars "overshoot" the global max current before they coming down to global max within a minute or so?

pkuehnel, 2023-01-12:
Yes, sadly I can not avoid this (without restructuring the whole software). The Delay should be ~12.5 seconds from Tesla API (this is unavoidable) + the intervall the car values are set (minimum 25 seconds (twice the 12.5 seconds) as otherwise it would permanently overshoot in both directions). So worst case would be 25 + 12.5 seconds overshoot (+ another 12.5 seconds until you see the power reduction).