pkuehnel/TeslaSolarCharger #507: Feature Request: Prevent Home Battery Discharging for reaching MinSoC

VinciSC, 2023-01-05:
Hi,
in current season with low PV earnings there is hardly enough energy to charge my home battery to a level that holds until next day. So I use TSC in MinSoC + PV mode to always have a minimum range available. After a ride this depletes the home battery with high discharging power in a very short time, afterwards car charges by grid. 
A mode where a MinSoC is charged over night after the home battery has been completely discharged by the regular house consumption (german: Grundlast) would be helpfull. It would reduce high discharging loads to the battery and improves overall efficiency as the energy is not charged and discharged twice.

pkuehnel, 2023-01-06:
Are you sure that it is more efficient to discharge the home battery with low power? I don't have a home battery but most people I talk to (including my solar installer) say that home batteries are more efficient with high loads (as the inverter needs power while being on, and is switching off when the battery is empty). There is also no double discharge. The home battery is charged once and discharged once, the same as the car battery. If the power for the car comes from the grid or from your home battery does not make a difference at that point, there is no additional charge/discharge when the car is charged from your home battery, there is always one charge of the home battery and one charge for your car battery. But I understand the psychological advantage if you can power your home the whole night vs. the car for a few minutes.

What you can do for now: Do not use the Min Soc + PV mode but the PV only mode. There you can set a date, time, and min soc. This would result in the car starting to charge at the latest possible moment to reach that SoC in time (Note: The battery capacity in your car settings page needs to be set correctly as this is used to calculate charging time). For now, you will need to update the date every day but I am working on a feature to use the time every day independent of the set date.
Also note: This feature is highly dependent on the wakeup of the car. But as you probably already noticed: Sometimes (a few times a year) the app is not capable of waking up the car. If this happens, the car won't be charged as TeslaSolarCharger uses the same API as the app does and you may have an empty battery in the morning.

VinciSC, 2023-01-06:
Hm, seems you are completely right, it's more a psychological thing. Charging/discharging the home battery and charging/discharging the cars battery each has it's efficiency factor, independent of the energy source. 
From cost efficiency point of view it seems completely inverse than I thought first: Direct usage of grid power w/o efficiency factor of charging/discharging and double charging/discharging efficiency losses by charging the car from the home battery is the cost minimum. Obviously it will mak only a little difference but in current times ...
Thanks for discussing and giving the right thought-provoking impulse!

closed 2023-01-06 by VinciSC