pkuehnel/TeslaSolarCharger #521: Grid Power Buffer in Baseconfig: only positive inputs posdible

MalteSchm, 2023-01-07:
When setting the Grid Power Buffet on my android Tablet using the Samsung keyboard no minus sign is shown on the keyboard,  I.e. negative value cannot be set

pkuehnel, 2023-01-07:
I just tried it with my Google Pixel phone (using [SwiftKey](https://play.google.com/store/apps/details?id=com.touchtype.swiftkey&hl=de&gl=US) as keyboard) and it works fine.

On my desktop, I can even use the down key (as no minimum value is set):
[image]

Could you please try the first input field on [this page](https://thepaciellogroup.github.io/AT-browser-tests/test-files/input-number.html): 
[image]

I don't have a Samsung device to debugging is not possible for me.

pkuehnel, 2023-01-07:
Looks like this is a Samsung specific issue on JavaScript: Look at the first answer [here](https://stackoverflow.com/questions/21677695/input-type-number-not-possible-to-insert-negative-numbers-on-mobile-phone)

MalteSchm, 2023-01-08:
Agreed this is my finding.
I'll update these values with another device then. 

Btw:
While testing I discovered that the correction factor seems not to use the JS number type and the full keyboard is shown

pkuehnel, 2023-01-08:
Or you just install Switft Key and use that for these changes. Changing keyboard are three clicks (at least on my device).

Yes this is because these fields are normal text fields (as this is my first project with a web UI I didn't know how to do it better that time, especially with country based `.` or `,` as decimal separator).

pkuehnel, 2023-01-08:
One last question: Is there a `-` on your keyboard e.g. in the Min SoC field on the main page (is not shown if you are in maximum power mode):
[image]

MalteSchm, 2023-01-08:
No, same keyboard is used here I.e. no minus sign

closed 2023-01-09 by pkuehnel