pkuehnel/TeslaSolarCharger #581: Add option to always charge in Spot price mode if under specified spotprice

pkuehnel, 2023-01-27:
This should only happen if there is no date set in the future or if the selected date is too far to plan charging slots.

pkuehnel, 2023-01-27:
Also charge if below solar price

andreasfaerber, 2024-02-25:
A first step into that direction (As C# is not for me):

https://github.com/andreasfaerber/tsc-spotpricecharger

Needs some beta testers, in particular with solar systems as these are not taken into account so far (i have none) to get these respected if anyone is interested in testing.

pkuehnel, 2025-07-28:
With the next update you can set a charging target e.g. tomorrow at 1PM you want 60% in you car. TSC then uses the grid prices as well as weather predictions to charge in the cheapest times. Update will be available tomorrow.