pkuehnel/TeslaSolarCharger #634: Howto find out internal Unit Identifier for Modbus Setup

Madaraxa, 2023-02-20:
Hi,
I got a RCT Inverter and while it should support Modbus I cant get any Data. I got the StartingAdress, the Type of Data (int16 - this should mean integer? which leads to "2"?) and the factor (10W - so "0.1").
The port should be 502 as it is standard for modbus. I also got the IP. So the only missing thing is the Unit Identifier.
Sadly in "discussions" someone wrote 71 in one of the examples, so i guess I have to check every number from 0 to XYZ?
Thanks in Advance

VinciSC, 2023-02-21:
Hi,
no try and error in the mentioned example, the UnitID 71 is simply taken from the interface description document that Kostal provides for my inverter (same w/ port 1502 instead of the standard 502). So you should search for a corresponding interface documentation for your RCT device.
BTW: are you sure RCT supports Modbus TCP? If not and they use Modbus RTU instead it will not work that way...

Madaraxa, 2023-02-21:
Hmm. Ok.
The documentation is not clear i think. But maybe it is rtu. As there is no Information for a Port or identifier...
https://rct-power.zendesk.com/attachments/token/WNO8HEqEUI8qQUI9TmVR62Sbm/?name=RCT_Power_RS485_Communication_20200429_v.1.2.pdf
There is an iobroker and home assist Modul afaik. Maybe i have to try and get the value from there.

VinciSC, 2023-02-21:
Page 2, line 2 of the referenced document says "Inverter using MODBUS V1.1 RTU standard protocol" so that sounds clear to me.
Just googeled for "rct modbus tcp" and e.g. the third hit (solaranzeige.de) indicates that there are solutions available to access RCT data via Ethernet. 
Good luck!

pkuehnel, 2023-06-04:
no response

closed 2023-06-04 by pkuehnel