pkuehnel/TeslaSolarCharger #686: SMA problem

merlinfive, 2023-03-22:
Hi I setup teslasolarcharger with smaplugin (using homemanager 2.0 which is working fine in iobroker) , getting this error in teslasolarcharger. What do I miss ?

[image]
[image]

 
from log
[09:29:20.054 INF Microsoft.Hosting.Lifetime] Application is shutting down...
[09:29:22.294 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] StartLogging()
[09:29:22.313 DBG Plugins.SmaEnergymeter.Services.EnergyMeterService] Use energymeterport 9522
[09:29:22.333 INF Microsoft.Hosting.Lifetime] Now listening on: http://[::]:7192
[09:29:22.335 INF Microsoft.Hosting.Lifetime] Application started. Press Ctrl+C to shut down.
[09:29:22.341 INF Microsoft.Hosting.Lifetime] Hosting environment: Production
[09:29:22.341 INF Microsoft.Hosting.Lifetime] Content root path: /app
[09:29:28.423 INF Microsoft.AspNetCore.Hosting.Diagnostics] Request starting HTTP/1.1 GET http://192.168.168.78:7192/api/CurrentPower/GetPower - -
[09:29:28.462 WRN Microsoft.AspNetCore.HttpsPolicy.HttpsRedirectionMiddleware] Failed to determine the https port for redirect.
[09:29:28.468 INF Microsoft.AspNetCore.Hosting.Diagnostics] Request finished HTTP/1.1 GET http://192.168.168.78:7192/api/CurrentPower/GetPower - - - 404 0 - 48.1201ms
[09:29:43.363 INF Microsoft.AspNetCore.Hosting.Diagnostics] Request starting HTTP/1.1 GET http://192.168.168.78:7192/api/CurrentPower/GetPower - -
[09:29:43.368 INF Microsoft.AspNetCore.Hosting.Diagnostics] Request finished HTTP/1.1 GET http://192.168.168.78:7192/api/CurrentPower/GetPower - - - 404 0 - 4.7490ms
[09:29:58.361 INF Microsoft.AspNetCore.Hosting.Diagnostics] Request starting HTTP/1.1 GET http://192.168.168.78:7192/api/CurrentPower/GetPower - -
[09:29:58.362 INF Microsoft.AspNetCore.Hosting.Diagnostics] Request finished HTTP/1.1 GET http://192.168.168.78:7192/api/CurrentPower/GetPower - - - 404 0 - 1.2602ms

merlinfive, 2023-03-22:
ok moving forward .. after stopping the iobroker Adapter for sma which is running as a container on same synology I get  values in smaplugin.log and the error in iobroker log if I try to start the adapter again

iobroker error
sma-em.0 | 2023-03-22 20:26:54.098 | error | UDP Socket error: Error: bind EADDRINUSE 0.0.0.0:9522

but still in Teslasolarcharger Overview

Grid Power is not available
Possible solutions:
If you do not use solar values set grid source to none.
Are all settings related to grid power (url, extraction patterns, headers,...) correct?
Are there any firewall related issues preventing reading the grid power value?

maybe because it is "dark now" ?

from smaplugin.log

[18:44:18.168 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] StartLogging()
[18:44:18.337 DBG Plugins.SmaEnergymeter.Services.EnergyMeterService] Use energymeterport 9522
[18:44:18.358 DBG Plugins.SmaEnergymeter.Services.EnergyMeterService] Use IP Address 239.12.255.254
[18:44:18.359 DBG Plugins.SmaEnergymeter.Services.EnergyMeterService] Parsed Ip Adress: 239.12.255.254
[18:44:18.359 DBG Plugins.SmaEnergymeter.Services.EnergyMeterService] Joining Multicast group
[18:44:18.359 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] Waiting for new values
[18:44:18.381 INF Microsoft.Hosting.Lifetime] Now listening on: http://[::]:7192
[18:44:18.382 INF Microsoft.Hosting.Lifetime] Application started. Press Ctrl+C to shut down.
[18:44:18.382 INF Microsoft.Hosting.Lifetime] Hosting environment: Production
[18:44:18.382 INF Microsoft.Hosting.Lifetime] Content root path: /app
[18:44:18.897 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] New values received
[18:44:18.902 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] current supply: 1079.1
[18:44:18.902 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] current overage: 0
[18:44:18.902 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] Waiting for new values
[18:44:19.891 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] New values received

merlinfive, 2023-03-22:
I used the syntax from your link 

[image]

and from the smaplugin.log I can see "numbers" but still error in Overview

[18:44:18.168 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] StartLogging()
[18:44:18.337 DBG Plugins.SmaEnergymeter.Services.EnergyMeterService] Use energymeterport 9522
[18:44:18.358 DBG Plugins.SmaEnergymeter.Services.EnergyMeterService] Use IP Address 239.12.255.254
[18:44:18.359 DBG Plugins.SmaEnergymeter.Services.EnergyMeterService] Parsed Ip Adress: 239.12.255.254
[18:44:18.359 DBG Plugins.SmaEnergymeter.Services.EnergyMeterService] Joining Multicast group
[18:44:18.359 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] Waiting for new values
[18:44:18.381 INF Microsoft.Hosting.Lifetime] Now listening on: http://[::]:7192
[18:44:18.382 INF Microsoft.Hosting.Lifetime] Application started. Press Ctrl+C to shut down.
[18:44:18.382 INF Microsoft.Hosting.Lifetime] Hosting environment: Production
[18:44:18.382 INF Microsoft.Hosting.Lifetime] Content root path: /app
[18:44:18.897 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] New values received
[18:44:18.902 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] current supply: 1079.1
[18:44:18.902 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] current overage: 0
[18:44:18.902 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] Waiting for new values
[18:44:19.891 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] New values received
[18:44:19.892 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] current supply: 1080.8
[18:44:19.892 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] current overage: 0

so from my pov smaplugin is collecting data

but teslasolarcharger didnt get the value

**[22-Mar-2023 21:24:59.150 ERR TeslaSolarCharger.Server.Services.PvValueService] Could not get value. NotFound, Not Found**
[22-Mar-2023 21:24:59.150 INF TeslaSolarCharger.Server.Services.TelegramService] Can not send Telegram Message because channel is empty.
[22-Mar-2023 21:25:14.141 ERR TeslaSolarCharger.Server.Services.PvValueService] Could not get value. NotFound, Not Found
[22-Mar-2023 21:25:14.141 INF TeslaSolarCharger.Server.Services.TelegramService] Can not send Telegram Message because channel is empty.

pkuehnel, 2023-03-22:
Are you 100% sure the IP Address is correct? This needs to be the IP Address of the SMA Plugin, not your Home Manager. SMA Plugin log looks like it never gets called.

If URL is definitely correct, please [enable detailed logging](https://github.com/pkuehnel/TeslaSolarCharger#generate-logfiles) in your TeslaSolarCharger container (container recreation with `docker-compose up -d` needed) and upload the log [here](https://nextcloud.pkuehnel.de/s/86TwFmH5f8MCQrD), as detailed logs may contain private data like your home geo-fence location.

merlinfive, 2023-03-22:
the 192.168.168.78 is the ip of my synology , home manager is a different IP

merlinfive, 2023-03-22:
[22-Mar-2023 22:03:26.929 VRB TeslaSolarCharger.Server.Services.PvValueService] Request grid power.
[22-Mar-2023 22:03:26.929 VRB TeslaSolarCharger.Server.Services.PvValueService] GetHttpResponse(Method: GET, RequestUri: 'http://192.168.168.78:7192/api/CurrentPower/GetPower', Version: 1.1, Content: <null>, Headers:
{
  Accept: */*
}) [called by Start]
[22-Mar-2023 22:03:26.938 ERR TeslaSolarCharger.Server.Services.PvValueService] Could not get value. NotFound, Not Found
[22-Mar-2023 22:03:26.938 VRB TeslaSolarCharger.Server.Services.TelegramService] SendMessage(Getting value did result in statuscode NotFound with reason Not Found)

pkuehnel, 2023-03-23:
`http://192.168.168.78:7192/swagger`

Do you see a UI Page then? If yes, can you make a screenshot?

merlinfive, 2023-03-23:
port seems to be open Discovered open port 7192/tcp on 192.168.168.78 (checked by Zenmap) and if I shutdown solarcharger the port gets closed

swagger also not reachable
[image]

merlinfive, 2023-03-23:
from the smaplugin logs we can see the collected numbers but there seems no webservice running in smaplugin container ?!

pkuehnel, 2023-03-23:
> [09:29:22.333 INF Microsoft.Hosting.Lifetime] Now listening on: http://[::]:7192

Means the web server in the SmaPlugin is running. Are there any logs in the SMA plugin when you enter the URL in your browser? What happens if you change the port in your `docker-compose.yml` to a random other port, e.g., 13487 and try it with that port?

merlinfive, 2023-03-23:
no difference , changed to 13487

from sma log
[12:56:09.732 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] New values received
[12:56:09.732 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] current supply: 0
[12:56:09.732 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] current overage: 935.2
[12:56:09.732 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] Waiting for new values
[12:56:10.713 INF Microsoft.AspNetCore.Hosting.Diagnostics] Request starting HTTP/1.1 GET http://192.168.168.78:13487/api/CurrentPower/GetPower - -
[12:56:10.713 INF Microsoft.AspNetCore.Hosting.Diagnostics] Request finished HTTP/1.1 GET http://192.168.168.78:13487/api/CurrentPower/GetPower - - - 404 0 - 0.3644ms
[12:56:10.726 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] New values received
[12:56:10.727 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] current supply: 0
[12:56:10.727 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] current overage: 925.8
[12:56:10.727 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] Waiting for new values
[12:56:11.731 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] New values received
[12:56:11.732 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] current supply: 0
[12:56:11.732 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] current overage: 917
[12:56:11.732 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] Waiting for new values

from teslacharger log

--- End of inner exception stack trace --- [See nested exception: System.Net.Http.HttpRequestException: Connection refused (192.168.168.78:7192) 

post change of URL

[23-Mar-2023 13:54:10.713 VRB TeslaSolarCharger.Server.Services.PvValueService] GetHttpResponse(Method: GET, RequestUri: 'http://192.168.168.78:13487/api/CurrentPower/GetPower', Version: 1.1, Content: <null>, Headers:
{
  Accept: */*
}) [called by Start]

[23-Mar-2023 13:54:11.108 ERR TeslaSolarCharger.Server.Services.PvValueService] Could not get value. NotFound, Not Found

merlinfive, 2023-03-23:
is this the correct setting (direct) for homemanger or do we have to change ? , there was nothing in documentation

[image]

merlinfive, 2023-03-23:
the plugin collect correct data from the homemanager , visible in log 

[12:55:56.728 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] New values received
[12:55:56.728 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] current supply: 0
[12:55:56.728 VRB Plugins.SmaEnergymeter.Services.EnergyMeterService] current overage: 1042.2

pkuehnel, 2023-03-23:
Opening URL in your browser is still the same? Getting 404?

> he plugin collect correct data from the homemanager , visible in log

Yep, the only issue is, you can not access the URL. Firewall or something is no issue? Very likeley not, as SMA Plugin gets the requests:

> [12:56:10.713 INF Microsoft.AspNetCore.Hosting.Diagnostics] Request starting HTTP/1.1 GET http://192.168.168.78:13487/api/CurrentPower/GetPower - -
[12:56:10.713 INF Microsoft.AspNetCore.Hosting.Diagnostics] Request finished HTTP/1.1 GET http://192.168.168.78:13487/api/CurrentPower/GetPower - - - 404 0 - 0.3644ms

I will run tests at home in about 5 hours.

pkuehnel, 2023-03-23:
Ah, I think if found the issue... Please use:
http://192.168.168.78:13487/api/CurrentPower/GetCurrentPower

as workaround for now... will be fixed in the next version.

merlinfive, 2023-03-23:
I moved back to the original port amd the the new URL (old port) it works now ! great

mabe the is another typo in documentation 
[image]
 (PORT 7091)

if I use Port7190/swagger I get the swagger site

pkuehnel, 2023-03-24:
Interesting, I did not change anything in that region.. Could you please restart TeslaMate? Or switch between charge modes?

merlinfive, 2023-03-29:
Hi , it's me again ;-)

yesterday I had to restart my synology , post restart I did not receive any numbers from smaplugin , always "0" .
I thought somthing crashed and because I had not much data inside I deleted everything (container , volumes Images, network)
and used the existing docker-compse.yml and started from scratch.
same result (2.14.2) , I also restarted the homemanger to make sure the UDP port is not blocked.

The Homemanger is also capable of using single cast , would it make sense to include this as a feature ?

[image]

pkuehnel, 2023-03-29:
Single cast is not implemented (and I did not it exists, will look into it). Maybe it is just because of the URL? As mentioned in the [2.14.1 Release Notes](https://github.com/pkuehnel/TeslaSolarCharger/releases/tag/v2.14.1) you need to update the URL to that one in the documentation:
Should be http://192.168.168.78:13487/api/CurrentPower/GetPower in your case.

If that does not help, please send me the logs of your SMA Plugin again.

closed 2023-03-29 by pkuehnel

merlinfive, 2023-03-29:
Hi , 

 

I used correct URL ( <http://192.168.168.78:7192/api/CurrentPower/GetPower> http://192.168.168.78:7192/api/CurrentPower/GetPower) and I’m able to open the website but number is always “0” , logfiles are attached the thread

pkuehnel, 2023-03-29:
Could you please add me as "Anlagenadministrator" in your SunnyPortal? https://www.sunnyportal.com/Templates/UserManagementConfiguration.aspx

My E-Mail would be patrick.kuehnel96@gmail.com

I won't change anything, but as I also have a SMA Home Manager, I could check what's different in configuration. It is no network issue as according to your logs you get the “no correct energymeter datagram” values every 30 seconds which is normal.

merlinfive, 2023-03-29:
added

pkuehnel, 2023-03-29:
Ich denke wir können auf deutsch wechseln. Der Einzige unterschied, den ich sehe ist, dass bei mir beim Home Manager in den Eigenschaften die Erweiterte Konfiguration aktiv ist, bei dir die einfache. Das macht dann den unterschied, dass unten der Bereich Zählerkonfiguration nicht angezeigt wird. Eventuell führt das auch dazu, dass dann keine Zählerwerte geschickt werden?
Soesieht es bei mir aus:
[image]

[image]

merlinfive, 2023-03-29:
Hi , 

 

kein Problem , gerne auch deutsch , die erweitere Konfiguration kann ich über bearbeiten aktivieren und dann sind die Einträge mit Deinen identisch , ich habe ja auch eine Zeitlang (2.14.0/2.14.1/2.14.2) Daten bekommen.

Das Problem trat erst seit dem reboot auf , deshalb dachte ich mach mal alles from scratch aber es tritt wieder auf , seltsam

 

 <https://user-images.githubusercontent.com/35361981/228525519-7cf882ee-48b5-4eb9-bc9b-d3050ecf6c37.png> 

 <https://user-images.githubusercontent.com/35361981/228525416-05af668c-26d7-47d8-9386-47c8ae0774c0.png>

pkuehnel, 2023-03-29:
Ich werde mich um Unicast bemühen.

pkuehnel, 2023-03-29:
@merlinfive Aber es funktioniert nicht zufällig einfach, wenn du da die IP einträgst? Ich kann es bei mir nur sehr schlecht testen, weil wenn ich da was eintrage, funktioniert mein WR nicht mehr, weil ich 70 % Abregelung habe.

merlinfive, 2023-03-30:
Denke nicht das das funktioniert da in der .yml ja die multicast Adresse steht, werde aber nachher mal ein bisschen testen/spielen .Dauert ja noch etwas bis mein neuer Tesla kommt von daher sind die Daten bei mir noch nicht so wichtig .

merlinfive, 2023-03-30:
warte erstmal falls ich der einzige bin , ich habe heute morgen mal die Kommunikation des WR auf IGMPv2 umgestellt , so wie es aussieht geht der Homemanager dann auch auf v2 zurück was durch meinen Switch durchgeht 
In der SMA Doku steht das Unicast bei eingetragener IP zusätzlich gesendet wird , aber leider nix über die Ports , denke wir können die Prio dafür auf low setzen .

[image]

pkuehnel, 2023-04-15:
Ich habe im SunnyPortal jetzt die IP Adressen des Wechselrichters und des TSC eingetragen und hat sofort geklappt. Ich komme damit sogar über das WireGuard VPN drüber und musste im SMA Plugin nichts anpassen, sollte also einfach funktionieren:

[image]