pkuehnel/TeslaSolarCharger #689: Double charge Session for one planned session in a Mode Spot+PV

donatasnicequestion, 2023-03-23:
Some minor issue noticed with a Mode Spot+PV: one charging session (as planned ) and another one very short session following the first one:

[image]

pkuehnel, 2023-03-24:
Could you please provide log files of the TeslaSolarCharger container for that time? [How to generate logfiles](https://github.com/pkuehnel/TeslaSolarCharger#generate-logfiles)

donatasnicequestion, 2023-03-25:
[teslasolarcharger.log](https://github.com/pkuehnel/TeslaSolarCharger/files/11069246/teslasolarcharger.log)

please see a log attached (time period acording to screenshot)

pkuehnel, 2023-03-25:
The car reported `90%` SoC at `23-Mar-2023 04:34:40.754`. After stopping the charging process the car reported `89%` at `23-Mar-2023 04:35:10.756`

I will add a threshold that charging only starts if difference between SoC and configured MinSoc is higher than 2%. Thanks for your logfiles.

pkuehnel, 2023-03-25:
@donatasnicequestion For now, you could set the charge limit in the car to 90% as well. Charging does not start if charge limit is less than 3% higher than SoC.

donatasnicequestion, 2023-03-26:
Thank you for your effort :) The threshold (1-2 % difference to desired SoC, in this example 90%) to start  would eliminate the behavior described (a  second charging session starting directly after the first one).

pkuehnel, 2023-03-26:
Do you know which version you were on when you had the issue? Looks like this is already implemented 🤔 Need to deeper inspect the issue.

donatasnicequestion, 2023-03-26:
For my understanding there are three values for Soc:

1. car.CarState.SocLimit - the limit set by Tesla (App or Car) - provided by Tesla API, for Example: 94 %
2. car.CarState.Soc - current (or actual) Soc - provided by Tesla API, for example 80%
3. car. CarConfiguration.MinimumSoC  - as configured in TeslaSolarCharger. for example 90%

The threshold is currently implemented for my understanding to handle the case at reaching car.CarState.SocLimit.

I think what is missing is to handle the case reaching car. CarConfiguration.MinimumSoC the same way and not to start charging again if car.CarState.SoC >=  car. CarConfiguration.MinimumSoC - 2 too.
 ?

pkuehnel, 2023-03-26:
I thought so, too in the first place but this is also already implemented, see line 39 in https://github.com/pkuehnel/TeslaSolarCharger/blob/master/TeslaSolarCharger/Server/Services/ChargeTimeCalculationService.cs

Seems to be an issue I am not aware of.

donatasnicequestion, 2023-03-28:
Ok, it is a  behavior of Tesla API and a car which isn't obvious.

pkuehnel, 2023-03-28:
Will be fixed in 2.15.0

closed 2023-03-28 by pkuehnel