pkuehnel/TeslaSolarCharger #747: request for single phase charging

merlinfive, 2023-04-10:
Hi ,
eine Option für einphasiges laden wäre schön .
Hintergrund , bei kleinen PV Anlagen könnte die ins Netz eingespeiste Leistung deutlich reduziert werden da man dann für den nächsten "Step" in Ampere nur 235W und nicht  705W (bei 235V) benötigt und so granularer nachführen könnte.
Ich nutze 2 Schütze für die 1/3 Phasen Umschaltung bei mir , manuell remote über Shellys.  

Hier könnte man bei einphasigem laden z.B. noch etwas rausholen.
[image]

Ich könnte mir das als Checkbox in der GUI 1/3 phasig vorstellen auf der dann die Berechnungsgrundlage angepasst wird

pkuehnel, 2023-04-10:
Die Berechnungsgrundlage basiert auf dem was der Tesla per API zurückgibt, also wenn du einphasig lädst, wird auch einphasig berechnet. Was erkannt wird siehst du, wenn du beim Auto die Details aufklappst, da ist ein `Actual Phases` dabei. Funktioniert das nicht, wie es soll?

FYI Umschaltung zwischen ein-und dreiphasig nur während gestopptem Ladevorgang machen, sonst kann es zu Schäden am Ladegerät (im Auto) kommen.

merlinfive, 2023-04-10:
Hi , 

 

ich hatte es bisher noch nicht ausprobiert weil ich dachte das es nur für 3-phasiges Laden gedacht ist , werde es nachher dann gleich mal testen .

 

Die Umschaltung mache ich 2-stufig , ich trenne erst alle 3 Phasen zwischen Wallbox und Netz mit Schütz 1 , dann nehme ich über 2 Schütze L2+L3 weg , dann wieder Schütz 1 Aktiv ,

jetzt liegt nur L1 an der Wallbox an und das laden kann nur 1-phasig erfolgen , wenn ich nur L2+L3 wegnehme geht die WB auf Störung daher 2-stufig

pkuehnel, 2023-04-10:
Falls es nicht klappt bitte wieder öffnen.

closed 2023-04-10 by pkuehnel

merlinfive, 2023-04-10:
Funktioniert , TOP !

merlinfive, 2023-04-10:
evtl. hab ich noch was gefunden , ich nutze ein 2party dashboard , bei 3phasen wird keine Ladeleistun g im Graphen angezeigt , bei 1 Phase schon , Anfang 3-phasig dort wo die orange Kurve anfängt 1-phasig
[image]

pkuehnel, 2023-04-10:
Evtl ein Bug in dem third party tool?

merlinfive, 2023-04-10:
Ist ja nur ein Dashboard ?!

pkuehnel, 2023-04-10:
Wo holt sich das Dashboard die Daten her? Vermutlich aus der TeslaMate Datenbank?

merlinfive, 2023-04-10:
Ja genau

pkuehnel, 2023-04-10:
Was zeigt das TeslaMate eigene Dashboard an? Also bei mir funktioniert das, daher die Vermutung, dass es ein Bug im Dashboard ist.

merlinfive, 2023-04-10:
Im Overview bekomme ich die Leistung angezeigt , aber eben nicht als Graphen oder ich hab es in den Teslamate dashboards übersehen ,

werde ich die Tage nochmal checken , Mittwoch geht mein Tesla weg und dann muß ich bis ~24/25 ohne auskommen ☹

pkuehnel, 2023-04-10:
Gelber Graph Wenn du auf Charges gehst und dann einen auswählst:
[image]

merlinfive, 2023-04-10:
Ok da passt es dann muß es am dashboard liegen

 

 <https://user-images.githubusercontent.com/35361981/230901144-13418f71-d602-43bc-944f-7a1721bde5af.png>

merlinfive, 2023-04-10:
Hab was entdeckt , er zeigt beim Teslamate dashboard zwar die Leistung an (gelb) aber da wo dreiphasig geladen wurde zeigt er null phasen  , erst wenn ich auf einphasig gehe zeigt er auch eine phase an

 

Markiert Wechsel von 3 auf 1 Phase , davor 2KW bei null Phasen , bei Dir passt es , seltsam

 

 <https://user-images.githubusercontent.com/35361981/230901144-13418f71-d602-43bc-944f-7a1721bde5af.png>

pkuehnel, 2023-04-10:
Ich denke TeslaMate kommt mit dem Wechsel der Phasen nicht klar, evtl den Ladevorgang länger pausieren, sodass TeslaMate einen neuen beginnt? Tesla Meldet nämlich 1 Phase bei einphasigem Laden, bei dreiphasigem Laden allerdings nicht 3 sondern 2. Nach jedem Ladevorgang korrigiert TeslaMate die 2 zu einer 3, evtl funktioniert das nicht richtig, wenn es zwischen 1 und dreiphasig wechselt? Kannst mal ins Log von TeslaMate schauen.

merlinfive, 2023-04-10:
Wird ich nochmal testen , im Log sehe ich nichts auffälliges

 

teslasolarcharger-teslamate-1  | 2023-04-10 11:40:05.398 car_id=1 [info] Derived efficiency factor: 185.0 Wh/km (5x confirmed)

teslasolarcharger-teslamate-1  | 2023-04-10 11:40:05.398 car_id=1 [info] Charging / NoPower / 2.54 kWh – 126 min

teslasolarcharger-teslamate-1  | 2023-04-10 11:40:05.473 car_id=1 [info] Start / :online

teslasolarcharger-teslamate-1  | 2023-04-10 11:40:05.648 car_id=1 [info] Connecting ...

teslasolarcharger-teslamate-1  | 2023-04-10 11:43:08.014 car_id=1 [info] Suspending logging

teslasolarcharger-teslamate-1  | 2023-04-10 11:43:34.320 [info] PUT /api/car/1/logging/resume

teslasolarcharger-teslamate-1  | 2023-04-10 11:43:34.322 car_id=1 [info] Resuming logging

teslasolarcharger-teslamate-1  | 2023-04-10 11:43:34.322 [info] Sent 204 in 1ms

teslasolarcharger-teslamate-1  | 2023-04-10 11:44:06.771 [info] GET  <https://nominatim.openstreetmap.org/reverse> https://nominatim.openstreetmap.org/reverse -> 200 (170.001 ms)

teslasolarcharger-teslamate-1  | 2023-04-10 11:44:06.848 car_id=1 [info] Charging / SOC: 36% / Home

teslasolarcharger-teslamate-1  | 2023-04-10 11:44:06.848 car_id=1 [info] Disconnecting ...

teslasolarcharger-teslamate-1  | 2023-04-10 11:51:05.443 [info] Refreshing access token ...

teslasolarcharger-teslamate-1  | 2023-04-10 11:51:05.796 [info] POST  <https://auth.tesla.com/oauth2/v3/token> https://auth.tesla.com/oauth2/v3/token -> 200 (352.214 ms)

teslasolarcharger-teslamate-1  | 2023-04-10 11:51:05.827 [info] Scheduling token refresh in 6 h

 

Hier habe ich Ladung unterbrochen und von 3 auf eine Phase umgestellt

 

teslasolarcharger-teslamate-1  | 2023-04-10 12:15:23.049 car_id=1 [info] Derived efficiency factor: 185.0 Wh/km (6x confirmed)

teslasolarcharger-teslamate-1  | 2023-04-10 12:15:23.049 car_id=1 [info] Charging / Disconnected / 1.06 kWh – 31 min

teslasolarcharger-teslamate-1  | 2023-04-10 12:15:23.103 car_id=1 [info] Start / :online

teslasolarcharger-teslamate-1  | 2023-04-10 12:15:23.260 car_id=1 [info] Connecting ...

teslasolarcharger-teslamate-1  | 2023-04-10 12:17:25.379 [info] GET  <https://nominatim.openstreetmap.org/reverse> https://nominatim.openstreetmap.org/reverse -> 200 (165.453 ms)

teslasolarcharger-teslamate-1  | 2023-04-10 12:17:25.448 car_id=1 [info] Charging / SOC: 37% / Home

teslasolarcharger-teslamate-1  | 2023-04-10 12:17:25.448 car_id=1 [info] Disconnecting ...

teslasolarcharger-teslamate-1  | 2023-04-10 12:34:56.739 [info] GET /

teslasolarcharger-teslamate-1  | 2023-04-10 12:34:56.744 [info] Sent 200 in 4ms

teslasolarcharger-teslamate-1  | 2023-04-10 12:34:59.755 [info] GET /geo-fences

teslasolarcharger-teslamate-1  | 2023-04-10 12:34:59.776 [info] Sent 200 in 20ms

teslasolarcharger-teslamate-1  | 2023-04-10 12:37:09.305 [info] GET /

teslasolarcharger-teslamate-1  | 2023-04-10 12:37:09.309 [info] Sent 200 in 3ms

teslasolarcharger-teslamate-1  | 2023-04-10 14:48:22.133 [info] GET /

teslasolarcharger-teslamate-1  | 2023-04-10 14:48:22.138 [info] Sent 200 in 4ms

teslasolarcharger-teslamate-1  | 2023-04-10 14:48:26.196 [info] GET /settings

teslasolarcharger-teslamate-1  | 2023-04-10 14:48:26.202 [info] Sent 200 in 6ms