pkuehnel/TeslaSolarCharger #792: How to Docker Volume mount with docker-compose.yml?

soulfire89, 2023-04-27:
The listed folders below volumes were not copied to TeslaSolarCharger docker folder. Container works correct, but how can I backup the data?

Every folder was created manually like described inside docker-compose.yml
[image]

or is it necessary to mention these folder at the and of docker-compose.yml below the point volumes
volumes:
  teslamate-db:
  teslamate-grafana-data:
  mosquitto-conf:
  mosquitto-data:
  teslasolarcharger-configs:

pkuehnel, 2023-04-27:
Very likely an issue in my README. These folders are not used anymore as there were to many issues regarding permissions. In the current version a docker volume is created for each service. The location of these files depend on your docker setup. Just Google for default docker volume location on windows

soulfire89, 2023-04-27:
The TSC folder included no other folders then import and the .yml. No other folders were created. 
And there is no other path on my PC with docker data.

pkuehnel, 2023-04-27:
Yes there is one. If you are one Windows 11 and on the latest Docker Version, it is in

`\\wsl$\docker-desktop-data\data\docker\volumes`

It could be on various locations depending on your set up, just ask Google for `Docker Volume location Windows xx`

One result would https://stackoverflow.com/questions/43181654/locating-data-volumes-in-docker-desktop-windows

but like I said there are various possible locations based on your Docker setup.

soulfire89, 2023-04-27:
I'm not using docker on windows. It's docker for synology and the folder is on /volume1/docker but as I said, there are no folders for each part.
I tried to create volume mounts with docker HUD but it didn't work.
And i dont want to try it with SSH, because I'm not that familiar with it.

pkuehnel, 2023-04-28:
Well, this is not an TeslaSolarCharger related issue. I had it running using Docker mounts and Docker volume both is working.  I have no idea how Docker is working on Synlogy. On Linux/Windows you can use folder mounts instead of volumes.

E.G. if you want to mount the data in your home directory you can use

```
  teslasolarcharger:
    image: pkuehnel/teslasolarcharger:latest
    container_name: teslasolarcharger
    logging:
        driver: "json-file"
        options:
            max-file: "10"
            max-size: "100m"
    restart: always
    environment:
#      - Serilog__MinimumLevel__Default=Verbose #uncomment this line and recreate container with docker-compose up -d for more detailed logs
      - TZ=Europe/Berlin ##You can change your Timezone here
    ports:
      - 7190:80
    volumes:
      -/home/user/tscdata:/app/configs
```

Note: if you change that: all your data in TSC/TeslaMate/... will be lost

pkuehnel, 2023-05-01:
I did not find any section in the README where it says the directories would be created.

closed 2023-05-01 by pkuehnel