pkuehnel/TeslaSolarCharger #793: Telegram sending post issue to TeslaMateApi due to change of standard Port

soulfire89, 2023-04-28:
Having this issue with the Telegram Service:
I had to change the Port of TeslaMateApi from standard 8080:8080 into **8090**:8080, because **8080** was allready in use.
Somehow the TSC Telegram Service doesn't recognize the new port and trys to send data over the standard port configuration.

[image]

pkuehnel, 2023-04-28:
This is as intended. Everything is working fine, as TSC uses the internal port, which can not be changed. The issue says that the vehicle is not available. This sometimes occurs (once every few hours, or when the car does not wake up within ~30 seconds) and is a TeslaAPI related issue. We can not do anything about it.

pkuehnel, 2023-04-28:
FYI: You can remove the ports section of your teslamateapi completly it is only needed, if you want to access it from external. This is why the ports section is commented out in the README. It is not needed for TSC.

closed 2023-04-28 by pkuehnel

soulfire89, 2023-04-28:
Thanks for the info, but it's installed, and I have some of the features to use it for.