pkuehnel/TeslaSolarCharger #804: Add support for certificate connection via mqtt

hisabimbola, 2023-05-07:
I'm using victron and to connect to public mqtt one needs to pass in a certificate. This is currently not possible.

Is it possible to add this functionality?

Thanks for the work

pkuehnel, 2023-05-07:
It should be possible, as I am using MqttNET for MQTT connections. (Note for me: https://github.com/dotnet/MQTTnet/issues/115#issuecomment-722328813).

But as I am currently pretty busy and this is a lot of work to do (adding file upload for certificate upload), this is nothing that will be implemented in the near future. I recommend you use Modbus TCP (should work on Victron) or any other Middleware for that, e.g., HomeAssistant or something like that.

If you could create a PR for that, the Solar MQTT Authentication is [here](https://github.com/pkuehnel/TeslaSolarCharger/blob/master/TeslaSolarCharger/Server/Services/SolarMqttService.cs) from line 49 to 54.

hisabimbola, 2023-05-07:
Yeah, I have a very dirty implementation for that [here](https://github.com/hisabimbola/TeslaSolarCharger/commit/1022ec7986b0bec7e754607f36dc111448469c72) but it works for me.

I think a quick implementation could be to store the cert in a folder, same as the docker-compose and then maybe add a checkbox in the UI that selects `tls` if true, then it'll search for the cert name and if found, will use that to connect.

What you think about that basic implementation

pkuehnel, 2023-05-08:
Little dirty, yes, but easily fixable: Add the file to the docker volume and implement the file name pattern like baseConfig.json. To do that follow these steps:
- Add the filename of the cert to the `appsettings.json`
- Do not use a fixed path for your cert but implement a new Method for getting the path in `IConfigurationWrapper` (just do the same what `ConfigurationWrapper.BaseConfigFileFullName()` does)
- Use the newly created `IConfigurationWrapper.SolarMqttCertFileFullName()` in `SolarMqttService` 
- Add `.WithTls` only if that file exists

When done, you can create a pull request, and I will do the UI and file upload things when I have time to do it. But you can already use that feature when you manually copy your certificate in your Docker Volume and get all future updates.

hisabimbola, 2023-05-25:
I'll close this issue as I ended using local mqtt and that works for me.

closed 2023-05-25 by hisabimbola