pkuehnel/TeslaSolarCharger #831: Changing Chargemode through Telegram Bot..?

TheKingNL, 2023-06-01:
Hi!

I have been using the Charger for some time and it works great. One question. Would it be possible to make the Telegram bot bi-directional? Like changing the "ChargeMode" using Telegram Bot commands? And maybe request like a quick overview of the charging status? Or is this not possible/very difficult?

Thanks in advance!

pkuehnel, 2023-06-02:
For now it is not planned but I would be glad to see a PullRequest for that.

pkuehnel, 2023-08-31:
Everything TSC does has an external link. Just add `/swagger` to the URL and you see all REST API endpoints. You can also open Chrome Dev Tools and click on the Network tab, there you see all requests how the TSC UI does them. Note: You alway have to send the whole objects and can not change only one value.