pkuehnel/TeslaSolarCharger #841: TeslaSolarCharger container keeps restarting

ViktorPankraz, 2023-06-14:
I'm trying to install TeslaSolarCharger on my Raspberry Pi 4 Model B Rev 1.1 
Using the installing instruction with the default docker-compose.yml did not work. Teslamate, Database and TeslaSolarCharger container did not run (endless restarting loops). After playing around with some different versions all containers are running now except the TeslaSolarCharger.
It keeps restarting even for other versions ( I have tested :alpha :2.12.0 :2.16.0 and :latest )

Here the verbose output from "docker-compose --verbose up" command:

_compose.cli.verbose_proxy.proxy_callable: docker attach <- ('579f82f547ffe56de13556c8f30e53d58b2e4cbb465348f3f06eedd8af92221e', stdout=True, stderr=True, stream=True)
compose.cli.verbose_proxy.proxy_callable: docker attach -> <docker.types.daemon.CancellableStream object at 0xb4dc4990>
compose.cli.verbose_proxy.proxy_callable: docker inspect_container <- ('579f82f547ffe56de13556c8f30e53d58b2e4cbb465348f3f06eedd8af92221e')
compose.cli.verbose_proxy.proxy_callable: docker inspect_container -> {'AppArmorProfile': '',
 'Args': ['TeslaSolarCharger.Server.dll'],
 'Config': {'AttachStderr': False,
            'AttachStdin': False,
            'AttachStdout': False,
            'Cmd': None,
            'Domainname': '',
            'Entrypoint': ['dotnet', 'TeslaSolarCharger.Server.dll'],
            'Env': ['Serilog__MinimumLevel__Default=Verbose',
                    'TZ=Europe/Berlin',
...
compose.cli.verbose_proxy.proxy_callable: docker wait <- ('579f82f547ffe56de13556c8f30e53d58b2e4cbb465348f3f06eedd8af92221e')
compose.cli.verbose_proxy.proxy_callable: docker wait -> {'StatusCode': 132}
teslasolarcharger exited with code 132_

This is my current config:
teslamate/teslamate:1.25.0
postgres:latest
teslamate/grafana:latest
eclipse-mosquitto:2
tobiasehlert/teslamateapi:latest
pkuehnel/teslasolarcharger:latest

I found the issue #186 with the same exit code, but I guess here its for a different reason. 
Any help would be appreciated

pkuehnel, 2023-06-14:
If you have nothing configured, yet in TSC I would recommend to remove the docker volume for TSC. If not, I would need the container logs

ViktorPankraz, 2023-06-14:
I renoved the volume already several times. Did also a complete new installation. The Volumen of TSC is still empty. There are no logs available for this container, I checked the log file, but it's empty. It looks like it already crashes during startup.
Any advise how to get more detailed logs?

pkuehnel, 2023-06-14:
Hmmm... How did you try to get the logs and what was the error?

pkuehnel, 2023-06-14:
How does the TSC part of your `docker-compose.yml` look like?

pkuehnel, 2023-06-14:
Generating a Logfile like described [here](https://github.com/pkuehnel/TeslaSolarCharger#generate-logfiles) also produces an empty file?

What happens if you remove the volumes section completely an recreate the container with `docker compose up -d`?

ViktorPankraz, 2023-06-14:
Yes, Generating a Logfile like described [here](https://github.com/pkuehnel/TeslaSolarCharger#generate-logfiles) also produces an empty file. It should be the same file, that I have looked up before.
I removed the volume section completely, but still no success.
[image]

pkuehnel, 2023-06-14:
I have no idea to be honest. Very likely an issue with your infrastructure as logging `logger created` is pretty much the first thing TSC does. Dou you get an output when you use `docker-compose up teslasolarcharger` does not log anything (without the `-d` you get the logs directly in your console)

ViktorPankraz, 2023-06-14:
Thats what I get on the console:

root@raspberry:/home/homeserver/teslacharger-config# docker-compose up teslasolarcharger
teslacharger-config_database_1 is up-to-date
teslacharger-config_teslamateapi_1 is up-to-date
Starting teslasolarcharger ... done
Attaching to teslasolarcharger
teslasolarcharger exited with code 132
teslasolarcharger exited with code 132
teslasolarcharger exited with code 132

Nothing more :(

pkuehnel, 2023-06-14:
https://medium.com/@nprch_12/docker-exited-132-e38f9dd2cd0d#:~:text=Error%20code%20132%20means%20that,May%20not%20support%20SSE4.2

Note: I did not dive into it, just the first thing I found.

ViktorPankraz, 2023-06-15:
I got the Container to run on a different environment. So it should be an issue of the docker environment on my PI. Let's close the issue here. Many thanks for your fast support.

closed 2023-06-15 by ViktorPankraz