pkuehnel/TeslaSolarCharger #850: Documentation: Missing hint that "import" directory needs to be created alongside docker-compose.yml

realdadfish, 2023-07-07:
Otherwise, the setup fails with

```
ERROR: for teslasolarcharger_teslamate_1  Cannot start service teslamate: Bind mount failed: '/local/path/to/import' doCreating teslasolarcharger_mosquitto_1    ... done
Creating teslasolarcharger_teslamateapi_1 ... done
Creating teslasolarcharger                ... done

ERROR: for teslamate  Cannot start service teslamate: Bind mount failed: '/local/path/to/import' does not exists
```

pkuehnel, 2023-07-31:
I just double checked the error: Where does `/local/path/to/import` come from in your error message? The default should be `./import:/opt/app/import` and with that, the import directory is auto-created with the first `docker compose up -d`

pkuehnel, 2023-10-13:
Closing as got no response.

closed 2023-10-13 by pkuehnel