pkuehnel/TeslaSolarCharger #851: Time of use Tarif

AusSandman, 2023-07-07:
Allow time of use pricing in charge price by either integrating with TeslaMate Agile or allowing user to enter 4 timed Tarif rates with option to select days.

pkuehnel, 2023-07-07:
Can you please explain this with an example?

AusSandman, 2023-07-08:
> Can you please explain this with an example?

When user does not have a fixed price per kWh of electricity, but price varies based on time of day. Ideally would like to prevent charging during Peak periods. But is useful to report charging costs when charging needs are not met by solar.

Example of pricing:
[image]

pkuehnel, 2023-07-08:
> either integrating with TeslaMate Agile

What do you mean with that?

AusSandman, 2023-07-08:
This add-on to TeslaMate https://github.com/MattJeanes/TeslaMateAgile allows time of use rates to feed into the TeslaMate to calculate charging costs. I am using this already with my TeslaMate. It doesn't however have the ability to set feed in rates.
This may be an easier way of getting the purchase price per kWh? Then just set a feed in rate in TeslaSolarCharger?
I would be nice to get accurate price of charging in TeslaSolarCharger.

AusSandman, 2023-07-08:
Would also be great to be able to set a "Do not Charge" period. This would be useful to prevent vehicle being charged during "Peak" power periods. This could just be a check box from the UI perhaps to not allow charging during "Peak? power.

pkuehnel, 2023-10-12:
I will try to integrate the `IPriceDataService` from [TeslaMateAgile](https://github.com/MattJeanes/TeslaMateAgile)

pkuehnel, 2023-11-01:
Forked to https://github.com/pkuehnel/ElectricityPriceCalculator

AusSandman, 2023-11-06:
> @AusSandman <https://github.com/AusSandman> Are you still interested in
> this feature? You can test it hopefully tomorrow.

Yes, look forward to testing

pkuehnel, 2023-11-07:
Sorry did not have the time today. Will take a few more days.

pkuehnel, 2023-11-10:
@AusSandman Can you give me your time based pricings and also your timezone? For now the backend is Ready but the UI is not, so I need to generate a JSON for you which you can send to TeslaSolarCharger so it calculates the correct prices for you and you can test already.

AusSandman, 2023-11-13:
- FixedPrice__TimeZone=Australia/Perth
      - FixedPrice__Prices__0=00:00-06:00=0.1845
      - FixedPrice__Prices__1=06:00-09:00=0.2255
      - FixedPrice__Prices__2=09:00-15:00=0.082
      - FixedPrice__Prices__3=15:00-21:00=0.5125
      - FixedPrice__Prices__4=21:00-23:00=0.2255
      - FixedPrice__Prices__5=23:00-24:00=0.1845

pkuehnel, 2023-11-13:
In the meantime UI got ready as well. Just change the Tag of TeslaSolarCharger in your `docker-compose.yml` from `:latest` to `:edge` and update everything with
```
docker compose pull && docker compose up -d
```

pkuehnel, 2023-11-13:
Please provide me feedback if everything is working and calculated correctly. Note If you update a Charge Price with a valid since date in the past all prices get updated based on that date, in TSC and in TeslaMate as well.

AusSandman, 2023-11-14:
Having issues getting TSC to work at all since my car updated to 2023.38.6. So hard to tell at this point. Managed to get prcing data in but car seems to charge itself and TSC doesn't stop it anymore like it used to.

pkuehnel, 2023-11-14:
But it should work. Did you update TeslaMate as well? You need the latest TeslaMate version. Can you take a screenshot of TSC when it should stop charging but does not?

AusSandman, 2023-11-14:
I already have updated teslamate. TSC (and teslmate) both show the car as not charging when it is. This is rectified if I wake the car up, but there is an issue I am guessing with the car software version?
I will try get a screenshot of this tomorrow.

pkuehnel, 2023-11-14:
Hmm. This should be fixed with the latest version of TeslaMate (1.27.4 should be displayed at the bottom of TeslaMate --> Settings). I am also on 2023.38.6 with my car, and everything is still working.

AusSandman, 2023-11-14:
I am running TeslaMate 1.27.4. Might try a two button reset on the car tomorrow.

pkuehnel, 2023-11-14:
And additionally a reboot of your Docker host.

AusSandman, 2023-11-16:
So, good news is after rebooting my car TCS is working again. However the calculated costs don't seem correct. Grid power cost during time of charginf was 0.082
[image]

pkuehnel, 2023-11-16:
Charge Cost should display the complete cost of the charge, Price per kWh are the average cost per kWh based on used grid kWh (including when these kWh were used) and solar kWh. I think best would be if you could check when no solar power was used.

AusSandman, 2023-11-21:
I have checked charging at night, but as some battery power was used it has calculated that as solar (which is fine).
So calculations apear to be correct.
[image]

closed 2026-10-17 by pkuehnel