pkuehnel/TeslaSolarCharger #894: Problems setting charge mode via HTTP PUT

BillGOH, 2023-09-02:
Again, thanks for this great software and your efforts.

I am trying to set charge mode via the following command (via openHAB):

`curl -X 'PUT' 'http://192.168.202.4:7190/api/Config/UpdateCarConfiguration?carId=1' -H 'accept: */*' -H 'Content-Type: application/json' -d '{"chargeMode": 2}'`

It works - charge mode is changed -, however all values under car settings are set to 0. Any idea to fix this? Thanks.

pkuehnel, 2023-09-02:
Currently it is not supported to only send part ov the object (as the UI does not need it). You need to get the whole object via the get request first, change the values you need and send the PUT including all properties.

BillGOH, 2023-09-02:
OK. Thanks for the explanation.

pkuehnel, 2024-03-03:
Duplicate of https://github.com/pkuehnel/TeslaSolarCharger/issues/840

closed 2024-03-03 by pkuehnel