pkuehnel/TeslaSolarCharger #943: New ChargeMode to charge at min. power when limited solar infeed

pdb01, 2023-10-25:
First of all thank you for providing this great piece of software!

I limit charging of my Tesla to 5A (as this is the minimum in the Tesla app), which means that with limited solar power, the car stops charging if there is less than ca. 1150W (5A * 230V, 1 phase) available.

I believe it would be interesting to still continue charging at 5A with limited grid power and use the available solar power (lower than the minimum to charg at 5A) to a maximum, as it's fed back to the grid otherwise.

So for example, charging at 5A (= 1150W), but 'supported' with some 500 to 800W from grid. So as long as there is a positive solar infeed (with buffer to compensate for stable power draw from the house, e.g. 200W), to continue to charge at the minimum specified value of 5A.
If this would be caused by temporary clouding, the benefit would be that current could increase also above the 5A when sun comes back.

Today I have a workaround to save car settings to max. 5A and put a higher SoC value so it's forced to charge at max. 5A, but then it keeps at 5A even if solar conditions approve.

Thanks, and if too difficult, well, then I continue as it is ;-)

In case this is already possible, thanks for guiding me through it... I'm using solar infeed value (only available every 5 min.) and Homewizard P1 grid value.

Br
Pieter

pkuehnel, 2023-10-25:
There are two options:

1. Just go to the base configuration in TSC and set the Power Buffer to `-1150 W` (or any Value you want). Note: this results in always pulling 1150 W from the grid (probably not what you want). I set this value to -690W (or 230W if charging on 1 phase) in Winter so all power generated is used to charge the car, and nothing is fed into the grid.
2. Go to Base Configuration, scroll down to the bottom, and open the advanced setting. Then click on `Expand` and update the `Time without enough solar power until charging stops`.

I think both options are not exactly what you want but with a combination of both settings, I think you can at least somewhat automate it.

What you really want would be a power buffer which is only used for charge stops, correct? So you still would have the regular power buffer as it exists now but another one, so the charge stop timer only starts if the grid infeed is below the second power buffer.

pdb01, 2023-10-25:
Indeed, something like that :-)

As I tried to explain, in 2 clicks I can work with TSC today already to do what it should as follows:
- I put MaximumAmpere Current to 5A in "Car Settings" and save
- I put min SOC to a higher value than the actual state and save, so it's forced to start charging immediately at the 5A
--> this does the trick to take the lowest possible charging and consuming all available solar power. 
But it's not smart to completely stop charging in case solar is gone completely (in the evening for example) or if solar goes beyond the 5A (in morning or after clouds, ...) as it's then still blocked at 5A.

I used to play ideed with the power buffer, but it's also not adapting to the situation as described above with increasing or decreasing sun (and if I'm unable to play with it constantly :-p). 
A second timer might indeed do the trick, if not complicates too much the whole software :-)

Another idea is to allow charging below 5A so the impact is less, but I'm not sure about the impact of lower than 5A charging on the battery life etc. (I ask myself the same question on updating charging current every 30 seconds during regular TSC operation, but guess this should be fine, right? :-p)

pkuehnel, 2023-10-25:
The slower you charge, the better it is for the battery. Changing the current does not matter as well. The only problem is that the car consumes about 250W while being awake, and the charger is less efficient the slower you charge, so with 2A on one phase, about 70% of charging power is gone due to the car being awake and the charger being inefficient. Charging below 1000 Watt does not make sense, in my opinion.

What you also need to consider is the wear while the car is awake and for each charge stop. While the car is awake the computer is running, all the cooling stuff for the computer as well,... On each charge start and stop a few switches in the car and in your wallbox/UMC/..., which also wear down,...

In your case I would go down to 4A minimum and use the default settings. You probably could save a few Euros when you maximize everything, but I don't think it's worth the effort. Just to give you an example: In December and January I don't use the TSC at all. But If I would have used ever single kWh in that time for the car, I would have saved 22 Euros last year. And without TSC the car just charges to 80% when coming home and that's it. No worries, no hassle to find the optimum,... April to September is TSC time ;-)