pkuehnel/TeslaSolarCharger #988: BUG? unable to start and control charging - Response status code does not indicate success: 403 (Forbidden)

JoKo1714, 2023-11-25:
app is working normally but cannot send control commands via teslamate to the car.
Model Y - Car Software Version 2023.32.200
Teslamate Version 1.27.4
TeslaSolarCHarger Version 2.22.0

teslasolarcharger.log
...

[25-Nov-2023 11:09:38.033 INF TeslaSolarCharger.Server.Services.LatestTimeToReachSocUpdateService] Charge date is not updated as car 1 is currently charging
[25-Nov-2023 11:09:38.066 DBG TeslaSolarCharger.Server.Services.ChargingService] Current overage is -256 Watt.
[25-Nov-2023 11:09:38.067 DBG TeslaSolarCharger.Server.Services.ChargingService] Relevant Geofence: Home
[25-Nov-2023 11:09:38.067 DBG TeslaSolarCharger.Server.Services.ChargingService] Relevant car ids: [1]
[25-Nov-2023 11:09:38.067 DBG TeslaSolarCharger.Server.Services.ChargingService] Irrelevant car ids: []
[25-Nov-2023 11:09:38.067 DBG TeslaSolarCharger.Server.Services.ChargingService] Relevant cars: [{"Id": 1, "CarConfiguration": {"ChargeMode": "PvOnly", "MinimumSoC": 20, "LatestTimeToReachSoC": "2023-11-26T00:00:00.0000000", "IgnoreLatestTimeToReachSocDate": true, "MaximumAmpere": 13, "MinimumAmpere": 1, "UsableEnergy": 60, "ShouldBeManaged": true, "ShouldSetChargeStartTimes": false, "ChargingPriority": 1, "$type": "CarConfiguration"}, "CarState": {"Name": null, "ShouldStartChargingSince": null, "EarliestSwitchOn": null, "ShouldStopChargingSince": "2023-11-25T10:54:08.0219306+01:00", "EarliestSwitchOff": "2023-11-25T10:56:08.0219306+01:00", "ScheduledChargingStartTime": null, "SoC": 45, "SocLimit": 100, "IsHomeGeofence": true, "TimeUntilFullCharge": "2.20:15:00", "ReachingMinSocAtFullSpeedCharge": "2023-11-25T11:09:38.0211439+01:00", "AutoFullSpeedCharge": false, "LastSetAmp": 0, "ChargerPhases": 1, "ActualPhases": 1, "ChargerVoltage": 230, "ChargerActualCurrent": 5, "ChargerPilotCurrent": 13, "ChargerRequestedCurrent": 5, "PluggedIn": true, "ClimateOn": false, "Latitude": 53.454794, "Longitude": 9.930881, "DistanceToHomeGeofence": -41, "ChargingPowerAtHome": 1150, "StateString": "charging", "State": "Charging", "Healthy": true, "ReducedChargeSpeedWarning": false, "PlannedChargingSlots": [], "$type": "CarState"}, "$type": "Car"}]
[25-Nov-2023 11:09:38.068 DBG TeslaSolarCharger.Server.Services.ChargingService] Irrelevant cars: []
[25-Nov-2023 11:09:38.068 DBG TeslaSolarCharger.Server.Services.ChargingService] Adding powerbuffer 0
[25-Nov-2023 11:09:38.068 DBG TeslaSolarCharger.Server.Services.ChargingService] Averaged overage -241
[25-Nov-2023 11:09:38.069 DBG TeslaSolarCharger.Server.Services.ChargingService] Overage after subtracting power buffer (0): -241
[25-Nov-2023 11:09:38.069 DBG TeslaSolarCharger.Server.Services.ChargingService] Home battery min soc: null
[25-Nov-2023 11:09:38.069 DBG TeslaSolarCharger.Server.Services.ChargingService] Home battery should charging power: null
[25-Nov-2023 11:09:38.069 DBG TeslaSolarCharger.Server.Services.ChargingService] At least one car is charging.
[25-Nov-2023 11:09:38.069 DBG TeslaSolarCharger.Server.Services.ChargingService] Power to control: -241
[25-Nov-2023 11:09:38.070 DBG TeslaSolarCharger.Server.Services.ChargingService] Amp to control: -2
[25-Nov-2023 11:09:38.070 DBG TeslaSolarCharger.Server.Services.ChargingService] Update Car amp for car null
[25-Nov-2023 11:09:38.070 DBG TeslaSolarCharger.Server.Services.ChargingService] Amps to set: 3
[25-Nov-2023 11:09:38.070 DBG TeslaSolarCharger.Server.Services.ChargingService] Min amp for car: 1
[25-Nov-2023 11:09:38.070 DBG TeslaSolarCharger.Server.Services.ChargingService] Max amp for car: 13
[25-Nov-2023 11:09:38.070 DBG TeslaSolarCharger.Server.Services.ChargingService] Normal amp set
[25-Nov-2023 11:09:38.221 ERR TeslaSolarCharger.Server.Services.TeslamateApiService] Error while sending post to TeslaMate. Response: {"error":"Tesla Vehicle Command Protocol required, please refer to the documentation here: https://developer.tesla.com/docs/fleet-api#2023-10-09-rest-api-vehicle-commands-endpoint-deprecation-warning","error_description":"","response":null}
[25-Nov-2023 11:09:38.350 ERR Quartz.Core.JobRunShell] Job DEFAULT.9902f7fa-6e50-4229-9297-89365b5935c8 threw an unhandled Exception:
System.Net.Http.HttpRequestException: Response status code does not indicate success: 403 (Forbidden).
   at System.Net.Http.HttpResponseMessage.EnsureSuccessStatusCode()
   at TeslaSolarCharger.Server.Services.TeslamateApiService.SendPostToTeslaMate(String url, Dictionary`2 parameters) in /src/TeslaSolarCharger/Server/Services/TeslamateApiService.cs:line 258
   at TeslaSolarCharger.Server.Services.TeslamateApiService.SetAmp(Int32 carId, Int32 amps) in /src/TeslaSolarCharger/Server/Services/TeslamateApiService.cs:line 121
   at TeslaSolarCharger.Server.Services.ChargingService.ChangeCarAmp(Car car, Int32 ampToChange, DtoValue`1 maxAmpIncrease) in /src/TeslaSolarCharger/Server/Services/ChargingService.cs:line 448
   at TeslaSolarCharger.Server.Services.ChargingService.SetNewChargingValues() in /src/TeslaSolarCharger/Server/Services/ChargingService.cs:line 137
   at TeslaSolarCharger.Server.Scheduling.Jobs.ChargingValueJob.Execute(IJobExecutionContext context) in /src/TeslaSolarCharger/Server/Scheduling/Jobs/ChargingValueJob.cs:line 22
   at Quartz.Core.JobRunShell.Run(CancellationToken cancellationToken)
[25-Nov-2023 11:09:38.353 ERR Quartz.Core.ErrorLogger] Job DEFAULT.9902f7fa-6e50-4229-9297-89365b5935c8 threw an exception.
Quartz.SchedulerException: Job threw an unhandled exception.
 ---> System.Net.Http.HttpRequestException: Response status code does not indicate success: 403 (Forbidden).
   at System.Net.Http.HttpResponseMessage.EnsureSuccessStatusCode()
   at TeslaSolarCharger.Server.Services.TeslamateApiService.SendPostToTeslaMate(String url, Dictionary`2 parameters) in /src/TeslaSolarCharger/Server/Services/TeslamateApiService.cs:line 258
   at TeslaSolarCharger.Server.Services.TeslamateApiService.SetAmp(Int32 carId, Int32 amps) in /src/TeslaSolarCharger/Server/Services/TeslamateApiService.cs:line 121
   at TeslaSolarCharger.Server.Services.ChargingService.ChangeCarAmp(Car car, Int32 ampToChange, DtoValue`1 maxAmpIncrease) in /src/TeslaSolarCharger/Server/Services/ChargingService.cs:line 448
   at TeslaSolarCharger.Server.Services.ChargingService.SetNewChargingValues() in /src/TeslaSolarCharger/Server/Services/ChargingService.cs:line 137
   at TeslaSolarCharger.Server.Scheduling.Jobs.ChargingValueJob.Execute(IJobExecutionContext context) in /src/TeslaSolarCharger/Server/Scheduling/Jobs/ChargingValueJob.cs:line 22
   at Quartz.Core.JobRunShell.Run(CancellationToken cancellationToken)
   --- End of inner exception stack trace --- [See nested exception: System.Net.Http.HttpRequestException: Response status code does not indicate success: 403 (Forbidden).
   at System.Net.Http.HttpResponseMessage.EnsureSuccessStatusCode()
   at TeslaSolarCharger.Server.Services.TeslamateApiService.SendPostToTeslaMate(String url, Dictionary`2 parameters) in /src/TeslaSolarCharger/Server/Services/TeslamateApiService.cs:line 258
   at TeslaSolarCharger.Server.Services.TeslamateApiService.SetAmp(Int32 carId, Int32 amps) in /src/TeslaSolarCharger/Server/Services/TeslamateApiService.cs:line 121
   at TeslaSolarCharger.Server.Services.ChargingService.ChangeCarAmp(Car car, Int32 ampToChange, DtoValue`1 maxAmpIncrease) in /src/TeslaSolarCharger/Server/Services/ChargingService.cs:line 448
   at TeslaSolarCharger.Server.Services.ChargingService.SetNewChargingValues() in /src/TeslaSolarCharger/Server/Services/ChargingService.cs:line 137
   at TeslaSolarCharger.Server.Scheduling.Jobs.ChargingValueJob.Execute(IJobExecutionContext context) in /src/TeslaSolarCharger/Server/Scheduling/Jobs/ChargingValueJob.cs:line 22
   at Quartz.Core.JobRunShell.Run(CancellationToken cancellationToken)]

pkuehnel, 2023-11-25:
Do you have a new car? Have you used TeslaSolarCharger with this car before?

JoKo1714, 2023-11-25:
It's a new car. It has not worked before. Model Y SR from Berlin. Built in October

JoKo1714, 2023-11-25:
I think it's related to https://github.com/evcc-io/evcc/issues/10460

pkuehnel, 2023-11-25:
Correct, all new cars are currently not working and from January on no car will: 

https://developer.tesla.com/docs/fleet-api#2023-11-17-rest-api-vehicle-commands-endpoint-deprecation-timeline-action-required

But I am already on it but will take some time, but there will be a solution.

pkuehnel, 2023-11-28:
The backend part is fixed and working in the current beta release. I hope to get everything else done until Sunday next week.

closed 2023-12-06 by pkuehnel