pkuehnel/TeslaSolarCharger #1040: Home Battery State of Charges doesn't update properly

martin-eberle, 2023-12-30:
TSC Version 2.23.1
Using Home Assistant to provide home battery SoC via REST: http://192.168.x.x:xxxx/api/states/sensor.battery_state_of_charge

Quite often the SoC in TSC is wrong/old, it just doesn't update properly. All other values like Inverter, Home Battery Power, and Grid Power are also requested from the same REST API from Home Assistant and updated properly.

When I look it up in the terminal via curl, the REST API delivers the correct value at any time.

martin-eberle, 2023-12-31:
I think I just found the cause for this behaviour but it still looks like a bug in TSC for me which should be fixed in proper error handling.

It seems, that the problem occurs when one of the values (in my case it was the value "Inverter" (solar power)) is unavailable. After sunset my inverter doesn't deliver 0 as power but the sensor just gets "unavailable". So I fixed this via Home Assistant that the solar power defaults to 0 if there is no sensor data.

In my opinion this error should also be handled by TSC. So if one value is unavailable it should not cause vor all other values to be not updated.

pkuehnel, 2023-12-31:
Wil be fixed but could take some time as Tesla Fleet API is prior 1 for now and there will be a redesign for getting PV Values to make it more flexible.

pkuehnel, 2024-03-25:
Is fixed in the latest beta and will be fixed in the next release