pkuehnel/TeslaSolarCharger #1088: Problem with Webview on Android Tablet

BillGOH, 2024-01-22:
To begin with, thanks again for this great software. I have an odd issue (and I would understand if it gets low priority, but maybe others are affected as well).

I am using an Amazon Fire HD 10 Tablet from 2021 with HABPanel and openHAB. HABPanel is displayed by Fully Kiosk Browser. Since 2.24 TeslaSolarCharger cannot be displayed within HABPanel any more. I see only the percentage circle and then the browser states that an unhandled error occurred.

I checked with Fully Kiosk Browser and they [said](https://www.fully-kiosk.com/en/#faqs), that I am using an outdated version of Amazon Webview (ver. 90). However, it is - unfortunately - not possible to update this module on this tablet (which has been "cracked" by [Fire Toolbox](https://xdaforums.com/t/windows-linux-tool-fire-toolbox-v33-1.3889604/)). It is also not possible to install the original Android webview.

For me, the website of TeslaSolarCharger looks quite simple so I am not sure what changed with 2.24 that requires a more complex browser rendering.

Since many smarthomers use Fully Kiosk Browser on Amazon tablets it might be worth to simplify the site (again)?

Thanks for help!

pkuehnel, 2024-01-22:
I added MudBlazor as third party component and updated everything to the latest .NET version. Do you have more details about the issue? A more detailed error should be visible if you can open the browser dev tools.

BillGOH, 2024-01-27:
Thank you for your quick reply. 

Unfortunately, I cannot use debug tools with Fully Kiosk Browser, because those are not implemented. 

If I install another browser it does not use system webview, but it's own engine (where TeslaSolarCharger works).

DomiSanNo1, 2024-01-27:
I have the same issue on Apple iPad Safari...

pkuehnel, 2024-01-27:
@DomiSanNo1 Could you please update to the latest version? I updated a UI component to the latest version. It is doubtful that the issue is fixed, but maybe it is resolved. If not, I would need some logs from the browser or at least a screenshot of the error.

BillGOH, 2024-01-27:
Thanks for your efforts. No change here, though. Unfortunately the browser only says that an unhandled error occurred.

DomiSanNo1, 2024-01-27:
Hey, it doesnt work. Here is the Screenshot [image]

pkuehnel, 2024-01-28:
What happens if you load a page that does not exist, e.g. `http://192.168.1.50:7190/asdf` (use your correct IP address)?

What happens if you load `CarSettings` `http://192.168.1.50:7190/CarSettings`

What happens if you load Swagger: `http://192.168.1.50:7190/swagger/index.html`

BillGOH, 2024-01-28:
Swagger works, the other sites not.

pkuehnel, 2024-01-28:
@BillGOH is it the same for you?

pkuehnel, 2024-01-28:
@BillGOH Try updating TSC with the tag `nomudblazor` instead of `latest`. Note: Some UI features are not available then.

BillGOH, 2024-01-29:
Unfortunately not working. Same error. Thank you for your patience.

pkuehnel, 2024-01-29:
Ok then I very likely can not help as the only difference then is the latest version of Microsoft .NET. It is for sure not any caching issue?

BillGOH, 2024-01-29:
No. I checked it and did restart the app and the device. 
Thanks - nevertheless - for your help and quick feedback.

DomiSanNo1, 2024-02-02:
Yes, the same for me.

pkuehnel, 2024-02-19:
As I am not able to reproduce this issue and it seems like the update to .NET8 caused the issue I probably can not do much about it. Sorry.