pkuehnel/TeslaSolarCharger #1098: Home Battery min SoC not reached but car is charged with PV power

gjuecat, 2024-01-27:
Home Battery min SoC set to 70%
actual SoC of Home Battery = 20%
car was chrged the whole day with solar power

as I can remeber this topic worked for me all the time but not since approx. 1 week

pkuehnel, 2024-01-27:
As far as I know I did not change anything in that region for a few months. Please send me a screenshot of your Home Battery settings and a screenshot of the Solar values in the homepage of TSC including all car details while the car is charging but should not. Note: The car details contain your car location, which is irrelevant for me.

gjuecat, 2024-01-27:
Car details from Teslamate or somewhere else?

pkuehnel, 2024-01-27:
At the homepage of TSC open the "more car details" section:
[image]

gjuecat, 2024-01-31:
[image]
[image]
[image]
[image]
[image]

pkuehnel, 2024-01-31:
What's your configured home battery goal charging power?
[image]

martin-eberle, 2024-02-01:
I'm having the same issue right now. Worked fine until about yesterday, when I updated to TSC v2.24.5.

pkuehnel, 2024-02-01:
@martin-eberle which version did you use before updating?

martin-eberle, 2024-02-01:
> @martin-eberle which version did you use before updating?

It pretty sure was at least v2.24.2, maybe it already was v2.24.3 or v2.24.4.

gjuecat, 2024-02-01:
my Home Battery Goal charging power was set to 3000 W
now I changed to 7000W

pkuehnel, 2024-02-01:
@gjuecat With 3000 Watt set, the TSC did exactly what it should do - reserved 3000 Watt for charging your home battery.

closed 2024-02-01 by pkuehnel

gjuecat, 2024-02-02:
@pkuehnel thanks for clarification, didn't look at the Battery Goal charging power