pkuehnel/TeslaSolarCharger #1108: Epex Spot AT (Awattar)

outrun0506, 2024-02-04:
Integration of Epex Spot AT Data would be nice. API can be found here: https://api.awattar.at/v1/marketdata

Should nearly be the same as for Epex Spot DE.

pkuehnel, 2024-02-04:
That's already possible if you are on v 2.23.1 or higher, see [here](https://github.com/pkuehnel/TeslaSolarCharger/releases/tag/v2.23.1)
Sorry for this being not documented, but I want to create an UI for that, the Environment variable thing is just a workaround for now. Note: As TSC caches data in its SQLite database, the new values will only be there 2 days later.

The TSC part of your `docker compose.yml` should then look like this:

```
teslasolarcharger:
    image: pkuehnel/teslasolarcharger:latest
    container_name: teslasolarcharger
    logging:
        driver: "json-file"
        options:
            max-file: "10"
            max-size: "100m"
    restart: always
    depends_on:
      - teslamateapi
    environment:
#      - Serilog__MinimumLevel__Default=Verbose #uncomment this line and recreate container with docker compose up -d for more detailed logs
      - TZ=Europe/Berlin ##You can change your Timezone here
      - AwattarBaseUrl=https://api.awattar.at/v1/marketdata
    ports:
      - 7190:80
    volumes:
      - teslasolarcharger-configs:/app/configs
```

Do not forget a `docker compose up -d` after updating your yml, as changes in Environment variables only take affect after rebuilding (not only restarting) the container.

outrun0506, 2024-02-04:
Ah 😅 yes, well sorry. Should habe read the Release Notes.... Sorry for that.

pkuehnel, 2024-02-04:
Our are not the first one who asks for that ;-)

closed 2024-02-04 by pkuehnel