pkuehnel/TeslaSolarCharger #1125: Unauthorized Tesla token every ~2 days

gjuecat, 2024-02-18:
[image]

Hi,
I get this message approx. every 2-3 days. Is there a solution that the token is authorized for a longer time? Am I the only one with this problem?

Thx
Jürgen

pkuehnel, 2024-02-18:
No, I have the same issue and it looks like this is currently an issue on Tesla's API. Please update to the latest version, I improved logging on that front. When being on the latest version you should find a log `Your token or refresh token is invalid. Very likely you have changed your Tesla password.` And after that a reason from Tesla.

pkuehnel, 2024-02-18:
@gjuecat Try updating to the `:edge` instead of the `:latest` release. I added a workaround: Until now, if Tesla returned an "Unauthorized" Status code one time, TSC immediately deleted the token (as it is unauthorized). Now, the token is not deleted, and it retries forever using the token. Note: The error notification you posted will still be displayed but will be autodeleted if any further Tesla API call returns a success status code.

pkuehnel, 2024-02-18:
But still not sure if it fixes the issue.

pkuehnel, 2024-02-21:
I think I found the cause of the issue; In about 10 minutes there is a new edge release available. Please try with that one again.

pkuehnel, 2024-02-21:
I created a new bug - please update again. Now everything should be fixed.

gjuecat, 2024-02-21:
upgrade edge version again?

pkuehnel, 2024-02-21:
Yes

pkuehnel, 2024-02-22:
As I had this issue a couple of times each day and now it's gone, I release it. This means you can go back to `:latest` in about 15 minutes which will be v 2.24.7. If you still have that issue, feel free to reopen the issue here.

closed 2024-02-22 by pkuehnel