pkuehnel/TeslaSolarCharger #1126: Option to set charge speed for Min SoC + PV Mode

idominiki, 2024-02-18:
Would be great if a charge speed can be set while in Min SoC + PV Mode

pkuehnel, 2024-02-18:
Can you give me an example when this would make sense?

idominiki, 2024-02-18:
Network-friendly charging 😊

pkuehnel, 2024-02-18:
You can change the max charging speed in the car configuration page but does not make sense imo, as charging is more efficient the faster you charge (at least with Tesla's as they have 300W standby consumption).

pdb01, 2024-02-19:
I have the exact same wish as I live in flanders (Belgium), where the grid operator has imposed a fee based on the quarterly peak power you ask from grid (in a way to limit peak usage and reduce investment needs). Concretely: we have to pay 40 EUR for each kW above 2,5 kW on the yearly invoice, which is measured on quarterly basis and the "worst" quarter of the year (averaged) is taken into account for this fee.

**Therefore, I'm looking indeed as well to have a way that TSC looks at the max combined current (and in case of Spot Price + PV also calculates the charging slots based on the reduced max combined current...)?**

Even better for my use case would be that TSC looks to the current grid power, which I put for the moment at e.g. -2500W (as the min. peak value) in order that TSC always charges at max. 2500W, compensates in case another device (oven, heater, ...) powers up and allows higher car charging when solar power is active.
This works great for Min SoC + PV (with SoC at 0%), but does not work for automated slot planning with Spot price.... 

thanks anyway, the tool is great already, and especially with Swagger integration I'm able to automate parts of it :-)

pkuehnel, 2024-02-19:
So @idominiki what @pdb01 wants is basically the same as what you want? Reducing the load on the grid?
@pdb01In your case it is based as a sum of kW on all phases. How is this measured? If you are 2 seconds above this threshold you have to pay? And FYI: Do not use the UpdateBaseConfiguration endpoint frequently as calling this results in many things going on under the hood. That's why there is a separate `UpdatePowerBuffer` endpoint. This just updates the value in memory and you can use it as often as you want.

idominiki, 2024-02-19:
@pkuehnel  of course this is the main reason. If I want to charge at full power to battery level XX, I use the full power mode.

pdb01, 2024-02-19:
No, it's an averaged measurement. The average power demand of the current 15-minutes interval is remembered for each month. In the end, an average of the 12 months peak power values is then the one used for invoicing.
Anyways, I'm only at 1 phase.

I intend to use UpdateCarConfiguration to limit the car current to a "safe" value when I use a home assistant switch for night time charging with low spot prices. But it is a challenge... a maximum charge speed would make things easier :-)

tobias-nolz, 2026-11-01:
For Austria, the use case as @pdb01 described in https://github.com/pkuehnel/TeslaSolarCharger/issues/1126#issuecomment-1952652632 will become increasingly relevant as this will soon apply to everyone here.

E-Control has adopted a reform of the Austrian grid tariffs that takes effect from 1 January 2027. Grid operators have until mid-2028 to switch over. Part of the grid fee will then be a capacity charge (Leistungspreis), based on the highest 15-minute peak the smart meter measures. 

So separating the grid limit and solar excess limit would make a lot of sense:

- Solar excess: up to the car's/connector's maximum (16A in my case). The capacity charge only applies to power taken from the grid, so high power from PV costs nothing extra and I want to make use of all my excess.
- Grid: a separate, lower maximum current (e.g. 5A).

Proposal: an optional "max current when charging from grid" setting. The planner would use it to size grid slots, so it plans longer, slower slots, and the regulation would use it to cap the current whenever the car isn't running on solar excess.