pkuehnel/TeslaSolarCharger #1146: mode set to "PV only" but car charges with spot price

gjuecat, 2024-03-03:
I set the mode to PV only 2 or 3 days before. On Sunday morning the car charged to min SoC like spot price setting.

This was the first time that this happend.

pkuehnel, 2024-03-03:
When you set the mode to PV Only and set a date and time in the future TSC starts to charge at the latest possible point in time to reach the MinSoC in time. This is needed e.g. if you want your car at 80% at 1 PM, solar power is used as long as reaching 80% on time. As soon as the 80% can only be reached with charging on maximum power, the car is charged on maximum power until min SOC is reached.

gjuecat, 2024-03-03:
OK, I was already wondering why there's a date and time. I thought PV only means that only PV is used for charging. => compared to "Min SoC + PV" which I used in former to times to charge to min SoC with max. Power (without PV) and all above min SoC with PV power??

If I switch between spot price + PV and PV only => I always have to change the min SoC that each charging mode makes sense for me. Am I wrong with that?

pkuehnel, 2024-11-21:
Looks like I missed the last comment. Is this issue still relevant? If not pleas close it.