pkuehnel/TeslaSolarCharger #1177: Language Settings DE-DE display English text

ralphsiegert, 2024-03-29:
I found a - for me very minor - issue: 
According to display, my language settings are DE-DE, but all texts are displayed in EN. Cant remember if or when I ever switched language but is looks wrong. 

[image]

pkuehnel, 2024-03-29:
There is no German translation for now, the displayed setting is only used for the decimal separator (`,` instead of `.`) and date format.

ralphsiegert, 2024-03-29:
Haha. So I am sorry. Is this something where a Non-Dev guy with tech understanding can work on? Is there a language file or something?

closed 2026-10-17 by pkuehnel