pkuehnel/TeslaSolarCharger #1279: Charge prices no longer updating

pdb01, 2024-05-31:
Hi,

Once again thank you for the great piece of software!

Since your last update, however, prices are no longer updated. 
Nor in Tesla Solar Charger view, nor in TeslaMate they appear.

If relevant, I use spot prices (as a small side comment: I'm in Belgium so the German prices are not always in line with mine, but I'm already happy enough with the rough estimation).

In the TSC log, I find following (just one of many):

[31-May-2024 14:40:06.298 ERR TeslaSolarCharger.Server.Services.TscOnlyChargingCostService] Error while updating charge prices of charging process with ID 461.
System.InvalidOperationException: The LINQ expression 'DbSet<SpotPrice>()
    .Where(s => (DateTimeOffset)s.EndDate >= __from_0 && (DateTimeOffset)s.StartDate <= __to_1)' could not be translated. Either rewrite the query in a form that can be translated, or switch to client evaluation explicitly by inserting a call to 'AsEnumerable', 'AsAsyncEnumerable', 'ToList', or 'ToListAsync'. See https://go.microsoft.com/fwlink/?linkid=2101038 for more information.
   at Microsoft.EntityFrameworkCore.Query.QueryableMethodTranslatingExpressionVisitor.Translate(Expression expression)
   at Microsoft.EntityFrameworkCore.Query.QueryCompilationContext.CreateQueryExecutor[TResult](Expression query)
   at Microsoft.EntityFrameworkCore.Query.Internal.QueryCompiler.<>c__DisplayClass12_0`1.<ExecuteAsync>b__0()
   at Microsoft.EntityFrameworkCore.Query.Internal.CompiledQueryCache.GetOrAddQuery[TResult](Object cacheKey, Func`1 compiler)
   at Microsoft.EntityFrameworkCore.Query.Internal.QueryCompiler.ExecuteAsync[TResult](Expression query, CancellationToken cancellationToken)
   at Microsoft.EntityFrameworkCore.Query.Internal.EntityQueryable`1.GetAsyncEnumerator(CancellationToken cancellationToken)
   at System.Runtime.CompilerServices.ConfiguredCancelableAsyncEnumerable`1.GetAsyncEnumerator()
   at Microsoft.EntityFrameworkCore.EntityFrameworkQueryableExtensions.ToListAsync[TSource](IQueryable`1 source, CancellationToken cancellationToken)
   at TeslaSolarCharger.Server.Services.GridPrice.OldTscConfigPriceService.GetPriceData(DateTimeOffset from, DateTimeOffset to, String configString) in /src/TeslaSolarCharger/Server/Services/GridPrice/OldTscConfigPriceService.cs:line 32
   at TeslaSolarCharger.Server.Services.TscOnlyChargingCostService.GetPricesInTimeSpan(DateTime from, DateTime to) in /src/TeslaSolarCharger/Server/Services/TscOnlyChargingCostService.cs:line 219
   at TeslaSolarCharger.Server.Services.TscOnlyChargingCostService.FinalizeChargingProcess(ChargingProcess chargingProcess) in /src/TeslaSolarCharger/Server/Services/TscOnlyChargingCostService.cs:line 158
   at TeslaSolarCharger.Server.Services.TscOnlyChargingCostService.UpdateChargePricesOfAllChargingProcesses() in /src/TeslaSolarCharger/Server/Services/TscOnlyChargingCostService.cs:line 70

I hope this makes sense to you.

Thanks!

Best regards,
Pieter

pkuehnel, 2024-05-31:
Which version is are you currently running? How many charge prices did you configure, and could you send me a screenshot of the detail view for each charge price configuration?

pkuehnel, 2024-05-31:
Could you please add a chargeprice with a date 01-01-2020 (just to make sure there is no charging process before the first configured charge price)?

pdb01, 2024-05-31:
Done, with no change (but no active changes happened since).
However, still several errors for older charges:

[31-May-2024 19:00:22.511 ERR TeslaSolarCharger.Server.Services.TscOnlyChargingCostService] Error while finalizing charging process with ID 499.
System.InvalidOperationException: The LINQ expression 'DbSet<SpotPrice>()
    .Where(s => (DateTimeOffset)s.EndDate >= __from_0 && (DateTimeOffset)s.StartDate <= __to_1)' could not be translated. Either rewrite the query in a form that can be translated, or switch to client evaluation explicitly by inserting a call to 'AsEnumerable', 'AsAsyncEnumerable', 'ToList', or 'ToListAsync'. See https://go.microsoft.com/fwlink/?linkid=2101038 for more information.
   at Microsoft.EntityFrameworkCore.Query.QueryableMethodTranslatingExpressionVisitor.Translate(Expression expression)
   at Microsoft.EntityFrameworkCore.Query.QueryCompilationContext.CreateQueryExecutor[TResult](Expression query)
   at Microsoft.EntityFrameworkCore.Query.Internal.QueryCompiler.<>c__DisplayClass12_0`1.<ExecuteAsync>b__0()
   at Microsoft.EntityFrameworkCore.Query.Internal.CompiledQueryCache.GetOrAddQuery[TResult](Object cacheKey, Func`1 compiler)
   at Microsoft.EntityFrameworkCore.Query.Internal.QueryCompiler.ExecuteAsync[TResult](Expression query, CancellationToken cancellationToken)
   at Microsoft.EntityFrameworkCore.Query.Internal.EntityQueryable`1.GetAsyncEnumerator(CancellationToken cancellationToken)
   at System.Runtime.CompilerServices.ConfiguredCancelableAsyncEnumerable`1.GetAsyncEnumerator()
   at Microsoft.EntityFrameworkCore.EntityFrameworkQueryableExtensions.ToListAsync[TSource](IQueryable`1 source, CancellationToken cancellationToken)
   at TeslaSolarCharger.Server.Services.GridPrice.OldTscConfigPriceService.GetPriceData(DateTimeOffset from, DateTimeOffset to, String configString) in /src/TeslaSolarCharger/Server/Services/GridPrice/OldTscConfigPriceService.cs:line 32
   at TeslaSolarCharger.Server.Services.TscOnlyChargingCostService.GetPricesInTimeSpan(DateTime from, DateTime to) in /src/TeslaSolarCharger/Server/Services/TscOnlyChargingCostService.cs:line 219
   at TeslaSolarCharger.Server.Services.TscOnlyChargingCostService.FinalizeChargingProcess(ChargingProcess chargingProcess) in /src/TeslaSolarCharger/Server/Services/TscOnlyChargingCostService.cs:line 158
   at TeslaSolarCharger.Server.Services.TscOnlyChargingCostService.FinalizeFinishedChargingProcesses() in /src/TeslaSolarCharger/Server/Services/TscOnlyChargingCostService.cs:line 50

pkuehnel, 2024-05-31:
Please upload a backup of your database to [my nextcloud](https://nextcloud.th-deg.de/s/m4ZRpzt9kideqsy)

pkuehnel, 2024-05-31:
One moment. I think I found the issue.

pkuehnel, 2024-05-31:
@pdb01 Please change your tag of the TSC in your `docker-compose.yml` from `:latest` to `:bug1279` and update with `docker compose pull && docker compose up -d`

pkuehnel, 2024-06-02:
As I did not get an answer but could reproduce the issue, I will release the fix.

pdb01, 2024-06-05:
Correct. Now seems all is correctly transferred and updated.

closed 2024-06-05 by pdb01