pkuehnel/TeslaSolarCharger #1282: Issue with Tesla Token? TSC not able to initiate charge.

TheKingNL, 2024-06-01:
I just did a complete refresh of TSC on Docker. (Because I was getting below issue after an upgrade.) I keep getting the following error in TSC;

_You did not request a Tesla Token, yet.
Possible solutions:
Open the Base Configuration and request a new token. Important: You need to allow access to all selectable scopes._

I did provide a Token after the install and TeslaMate graphs are working. TSC also found my car and it can read SoC. It does not start to charge however.

When I open the base configuration page in TSC I have no option to request new tokens. 

Does this has to do with the owner API vs Fleet API?

Or did I miss something else..?

pkuehnel, 2024-06-01:
There should be a button to request a token.

[image]

TheKingNL, 2024-06-01:
Thanks @pkuehnel Turns out that Edge was the issue. Somehow it was not displaying the new version of the pages but the old version which did not have this button. It did provide the correct SoC etc so it was a bit odd. Tried it in Chrome now and I was able to add the token.

closed 2024-06-01 by TheKingNL