pkuehnel/TeslaSolarCharger #1289: Mobile UI ugly since latest update

tkoenig89, 2024-06-03:
I just updated my docker stack today to the latest version and sadly i found my mobile view to look pretty bad. I would not say its broken, but its bad in comparison to how well it worked before. I hope there might be a simple css fix, as the UI does not use the full width of the screen. Version before update was from February.

[image]

pkuehnel, 2024-06-04:
Hmm, does not look like that for me. Which phone do you have and which browser are you using?

tkoenig89, 2024-06-04:
Thanks for the quick reply. I have a **Moto G54 5G** (Android 14) running the latest Version of Edge (the Chromium based one). Until the update everything looked pretty organized on mobile, which was quite nice because i mostly use TSC from my phone.

pkuehnel, 2024-06-04:
Hmm, I can only reproduce the issue with a width below 193 pixels, and then I do not have more space, so makes sense that it is that way. Can you reproduce the issue in Chrome also?

pkuehnel, 2024-06-04:
Do you know which version you had before and which version do you have now, so I can check on changes.

tkoenig89, 2024-06-04:
It's now at 2.25.5 and I'm pretty sure before i had 2.24.7. First time installed TSC in February and never got prompted for an update until a few days ago. Then updated the complete docker stack. Now showing 2.25.5.

pkuehnel, 2024-06-04:
I can reproduce the issue but only on Edge on Android.

tkoenig89, 2024-06-04:
Thanks for looking into this. But this has very low priority, as there is already a simple fix (Unless someone depends on increased font size - and even then the UI works in general). And thank you in general for providing this very nice and valuable software!

pkuehnel, 2026-10-17:
Fixed, probably a few versions ago.

closed 2026-10-17 by pkuehnel