pkuehnel/TeslaSolarCharger #1296: TSC isn't managing Current since this morning

int2001, 2024-06-07:
Using Version [2.27.6](https://github.com/pkuehnel/TeslaSolarCharger/releases/tag/v2.27.6)

Behaviour:
- Start/Stop seems to work, but no regulation of power is missing. It's loading with the preselected value within the Tesla-App.
- Checked: CarSettings (MaxPower, etc.) as well as MQTT-Data i'm providing for Grid/Solar/etc. - can't find an issue there

Log shows following exceptions:
```
[07-Jun-2024 16:23:38.691 ERR Quartz.Core.ErrorLogger] Job DEFAULT.bc961fac-9da7-4997-a082-8f7836351d41 threw an exception.
Quartz.SchedulerException: Job threw an unhandled exception.
 ---> Newtonsoft.Json.JsonReaderException: Unexpected character encountered while parsing value: R. Path '', line 0, position 0.
   at Newtonsoft.Json.JsonTextReader.ParseValue()
   at Newtonsoft.Json.JsonReader.ReadForType(JsonContract contract, Boolean hasConverter)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.Deserialize(JsonReader reader, Type objectType, Boolean checkAdditionalContent)
   at Newtonsoft.Json.JsonSerializer.DeserializeInternal(JsonReader reader, Type objectType)
   at Newtonsoft.Json.JsonConvert.DeserializeObject(String value, Type type, JsonSerializerSettings settings)
   at Newtonsoft.Json.JsonConvert.DeserializeObject[T](String value, JsonSerializerSettings settings)
   at TeslaSolarCharger.Server.Services.TeslaFleetApiService.SendCommandToTeslaApi[T](String vin, DtoFleetApiRequest fleetApiRequest, HttpMethod httpMethod, String contentData) in /src/TeslaSolarCharger/Server/Services/TeslaFleetApiService.cs:line 490
   at TeslaSolarCharger.Server.Services.TeslaFleetApiService.SetAmp(Int32 carId, Int32 amps) in /src/TeslaSolarCharger/Server/Services/TeslaFleetApiService.cs:line 131
   at TeslaSolarCharger.Server.Services.ChargingService.ChangeCarAmp(DtoCar dtoCar, Int32 ampToChange, DtoValue`1 maxAmpIncrease) in /src/TeslaSolarCharger/Server/Services/ChargingService.cs:line 461
   at TeslaSolarCharger.Server.Services.ChargingService.SetNewChargingValues() in /src/TeslaSolarCharger/Server/Services/ChargingService.cs:line 153
   at TeslaSolarCharger.Server.Scheduling.Jobs.ChargingValueJob.Execute(IJobExecutionContext context) in /src/TeslaSolarCharger/Server/Scheduling/Jobs/ChargingValueJob.cs:line 12
   at Quartz.Core.JobRunShell.Run(CancellationToken cancellationToken)
   --- End of inner exception stack trace --- [See nested exception: Newtonsoft.Json.JsonReaderException: Unexpected character encountered while parsing value: R. Path '', line 0, position 0.
   at Newtonsoft.Json.JsonTextReader.ParseValue()
   at Newtonsoft.Json.JsonReader.ReadForType(JsonContract contract, Boolean hasConverter)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.Deserialize(JsonReader reader, Type objectType, Boolean checkAdditionalContent)
   at Newtonsoft.Json.JsonSerializer.DeserializeInternal(JsonReader reader, Type objectType)
   at Newtonsoft.Json.JsonConvert.DeserializeObject(String value, Type type, JsonSerializerSettings settings)
   at Newtonsoft.Json.JsonConvert.DeserializeObject[T](String value, JsonSerializerSettings settings)
   at TeslaSolarCharger.Server.Services.TeslaFleetApiService.SendCommandToTeslaApi[T](String vin, DtoFleetApiRequest fleetApiRequest, HttpMethod httpMethod, String contentData) in /src/TeslaSolarCharger/Server/Services/TeslaFleetApiService.cs:line 490
   at TeslaSolarCharger.Server.Services.TeslaFleetApiService.SetAmp(Int32 carId, Int32 amps) in /src/TeslaSolarCharger/Server/Services/TeslaFleetApiService.cs:line 131
   at TeslaSolarCharger.Server.Services.ChargingService.ChangeCarAmp(DtoCar dtoCar, Int32 ampToChange, DtoValue`1 maxAmpIncrease) in /src/TeslaSolarCharger/Server/Services/ChargingService.cs:line 461
   at TeslaSolarCharger.Server.Services.ChargingService.SetNewChargingValues() in /src/TeslaSolarCharger/Server/Services/ChargingService.cs:line 153
   at TeslaSolarCharger.Server.Scheduling.Jobs.ChargingValueJob.Execute(IJobExecutionContext context) in /src/TeslaSolarCharger/Server/Scheduling/Jobs/ChargingValueJob.cs:line 12
   at Quartz.Core.JobRunShell.Run(CancellationToken cancellationToken)]
```

simonchristoph, 2024-06-07:
Same here, TSC pull is from today but it also didn't work before I installed the update
Model Y 2024.14.9
I can add the information, that the last used current is set, not the from from the app. I m loading with 1A all day long even though I have power for about 10A.
It seems as if the Tesla doesn't get the information on how fast he should load.

int2001, 2024-06-07:
Good to know, that i'm not alone ;)
PV producing enough energy to feed with >10A here. What a pitty...

pkuehnel, 2024-06-07:
Tesla implemented rate limits (max 50 requests per car per day https://developer.tesla.com/docs/fleet-api#membership-levels)
There is currently no way to go around this. About 50% of cars are affected by this issue since today.
I will develop a way to send the requests via BLE - I hope to finish it this weekend, but not sure how much I have to do. Like I said, for now there is no solution and as long as Tesla does not implement different membership levels than the free one, there won't be a solution.

For the BLE way, a Bluetooth device is needed near the car with an internet connection. The Raspberry Pi should be able to handle it, but like I said, I will start with the development today, so no details known for now.

pkuehnel, 2024-06-07:
As I am not affected, I would appreciate it if anyone having the issue could upload a Verbose log [here](https://nextcloud.th-deg.de/s/a6wMiDb93Ny9jKz). [How to create a verbose log](https://github.com/pkuehnel/TeslaSolarCharger?tab=readme-ov-file#generate-logfiles).

simonchristoph, 2024-06-07:
I hope an other solution (intelligent wallbox) won't be necessary.
I mean this open connection to the car for developers like @pkuehnel is one of many reasons I really love teslas. It's so great for the whole community around Teslas that this company enabled things like this.

I think Tesla should overthink the limits.

simonchristoph, 2024-06-07:
Following the link @pkuehnel posted above I see even more problems arriving.
50 commands is not enough for e.g. TSC but also 200 requests for data will be not enough for applications like Teslamate too.
Oh dear :-(

simonchristoph, 2024-06-08:
> As I am not affected, I would appreciate it if anyone having the issue could upload a Verbose log [here](https://nextcloud.th-deg.de/s/a6wMiDb93Ny9jKz). [How to create a verbose log](https://github.com/pkuehnel/TeslaSolarCharger?tab=readme-ov-file#generate-logfiles).

@pkuehnel do you have a idea, why only some users are affected? Keeping this in mind an other question comes up:
Are you sure about the reason for the failure of communication as explained above?

pkuehnel, 2024-06-08:
@simonchristoph It has alway been this way for any changes, at first only a few users are affected (last week) and after some time huge waves of users are affected. Yes 100% sure. Tesla Returns Too many Request as status code and adds `Retry in x seconds` as response message.

pdb01, 2024-06-10:
Not sure where to write/ask this, but as it's connected with the main issue, I go here:
Great that you have developed the BLE way to communicate already! 

As my 'main' Raspberry Pi is too far away, I will need a separate one, as you've explained.
--> Can this be a more simple Raspberry Pi ZERO W (1 or 2), or you need something more complex?
--> Can this even be placed inside the car (WiFi goes up to the car) and powered by the onboard USB (as long as it's then powered through Sentry mode... might not be energy efficient but at least maybe it works as well? (in this way I avoid having to place this raspberry just for that reason in a garage with all the cold and moist etc.)
--> or some kind of BLE extender, .... ?

Thanks for any insights :-)

int2001, 2024-06-10:
> --> Can this be a more simple Raspberry Pi ZERO W (1 or 2), or you need something more complex?

Testing RPi-Zero 2W during this week (i hpe it'll arrive today/tomorrow).
Currently it works with an old RPi3+ (And of course 4 or 5)

> --> Can this even be placed inside the car (WiFi goes up to the car) and powered by the onboard USB (as long as it's then powered through Sentry mode... might not be energy efficient but at least [...]

it should be okay to place it within the car. For the Pi there are only a few prerequisites:
- Needs Network ((W)LAN) while charging (as well for start/stop)
- Needs to be reachable via DNS or same IP within your (W)LAN while charging (as well for start/stop)
- Needs to be capable of using BLE (don't get confuse with Bluetooth. BLE and Bluetooth are different protocols)

pkuehnel, 2024-06-10:
Pi Zero W1 won't work as armV7 is required.

pkuehnel, 2024-06-10:
Just tested a Raspberry Pi Zero 2W. Is working, but you need to install Raspbian OS without desktop environment. At least my Pi Zero 2W crashed already when trying to update with `sudo apt-get upgrade`.

Closing this as documentation is online and everything is working.

closed 2024-06-10 by pkuehnel

simonchristoph, 2024-06-10:
You are just genius!
Thanks for this great workaround

pdb01, 2024-06-14:
Re-opening this one, as I believe still connected... ?

TSC and new BLE perfectly working, but seems the connection to Teslamate is lost and therefore settings from TSC are not pulled through BLE and the car, so it's doing crazy things I don't request :-p 

TeslaMate gives following errors:
2024-06-14 09:07:38.503 [info] GET https://owner-api.teslamotors.com/api/1/vehicles/xxxxxxxx/vehicle_data -> 429 (80.138 ms)
2024-06-14 09:07:38.503 [warning] TeslaApi.Error / "Retry in 60742 seconds\n"
2024-06-14 09:07:38.503 car_id=1 [error] Error / :unknown

I see some notes on teslamate page about it as well... ?