pkuehnel/TeslaSolarCharger #1299: wrong numbers in handled charges

mqth76, 2024-06-07:
I charged via my wallbox about 13kwh today but TeslasolarCharger only shows this:

Start Time | Calculated Price | Price Per Kwh | Used Grid Energy | Used Home Battery Energy | Used Solar Energy

07.06.2024 12:37:55 | 0,01 | 0,071 | 0,00 | 0,12 | 0,02
07.06.2024 12:28:05 | 0,02 | 0,095 | 0,00 | 0,12 | 0,09
07.06.2024 12:06:27 | 0,03 | 0,086 | 0,00 | 0,06 | 0,29
07.06.2024 10:34:01 | 0,07 | 0,073 | 0,00 | 0,13 | 0,83
07.06.2024 10:11:24 | 0,07 | 0,078 | 0,00 | 0,10 | 0,80

pkuehnel, 2024-06-09:
Which version are you on? I have the correct numbers, Will investigate in a few days.

mqth76, 2024-06-09:
Presently in 2.27.7. Problems startet with .5 or .6.....

pkuehnel, 2024-06-09:
You do not have a home batter, correct?

mqth76, 2024-06-09:
Yes

mqth76, 2024-06-09:
Sorry got ist wrong, I Do have a battery

mqth76, 2024-06-10:
is it possible, that you changed the columns from kwh to price? thats what
I might have figured out.....

pkuehnel, 2024-06-10:
No, did not change it.

mqth76, 2024-06-25:
Hi,
additionally one charging event ist splitted into several events...

[image: image.png]

THX

pkuehnel, 2024-06-30:
Images are not transferred to the issue, please comment again on GitHub.

mqth76, 2024-07-01:
[image]

[image]

charged 12,96kwh, but logged in various events with incomprehensible numbers...

pkuehnel, 2024-07-01:
Please upload a Backup of your TSC [here](https://nextcloud.th-deg.de/s/a6wMiDb93Ny9jKz)

pkuehnel, 2024-07-01:
If possible, add some screenshots of your TeslaMate charging data

mqth76, 2024-07-01:
done.

THX